eclipse-milo/milo #471: When KingIOServer(OPCUA Server) abnormal shutdown, Milo OPCUAClient sometimes occur deadlock

ligang2495, 2019-05-16:
Hello，I use Milo sdk-client(v0.3.0-M1) as the OPCUA Client and use KingIOServer(v3.7) as  the OPCUA Server .
When the connection disconnect between client and server , client will reconnect.
code follows:

```
public void reConnect(){
	try{
		cleanOpcUaClient();
		doConnectAndSubscription();
	}catch (Exception ex){
		logger.error("reconnect failure ,"+ExceptionUtils.getStackTrace(ex));
		reConnect();
	}
}

private void cleanOpcUaClient() throws Exception{
	if (client != null){
		try{
			client.disconnect().get(reconnectInterval,TimeUnit.MILLISECONDS);
		}catch (Exception e){
			logger.error("cleanOpcUaClient : client disconnect timeout,maybe cause by OPCUA Server program crashed");
			Stack.releaseSharedResources();
		}
		client = null;
	}
}

private void doConnectAndSubscription(){
	try{
		client = createOpcClient();
		doSubscription();
	}catch (Exception e){
		logger.error("connect failure , delay {}ms to reconnect... cause by {}",reconnectInterval,e.getMessage());
		try {
			TimeUnit.MILLISECONDS.sleep(reconnectInterval);
		} catch (InterruptedException e1) {
			logger.error("sleep faulure:"+ExceptionUtils.getStackTrace(e));
		}
		reConnect();
	}
}
```
when the opcua server shutdown, client sometimes occur deadlock,and the following code will throwing an  TimeoutException.

`client.disconnect().get(reconnectInterval,TimeUnit.MILLISECONDS);`

If do not set timeout,it will always block.

The jstack logs follows：

```
Found one Java-level deadlock:
=============================
"milo-shared-thread-pool-3":
  waiting for ownable synchronizer 0x0000000772aa8198, (a java.util.concurrent.locks.ReentrantReadWriteLock$FairSync),
  which is held by "milo-shared-thread-pool-1"
"milo-shared-thread-pool-1":
  waiting for ownable synchronizer 0x0000000772a95280, (a java.util.concurrent.locks.ReentrantReadWriteLock$FairSync),
  which is held by "milo-shared-thread-pool-3"

Java stack information for the threads listed above:
===================================================
"milo-shared-thread-pool-3":
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x0000000772aa8198> (a java.util.concurrent.locks.ReentrantReadWriteLock$FairSync)
        at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer.parkAndCheckInterrupt(AbstractQueuedSynchronizer.java:836)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer.acquireQueued(AbstractQueuedSynchronizer.java:870)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer.acquire(AbstractQueuedSynchronizer.java:1199)
        at java.util.concurrent.locks.ReentrantReadWriteLock$WriteLock.lock(ReentrantReadWriteLock.java:943)
        at com.digitalpetri.strictmachine.StrictMachine.getFromContext(StrictMachine.java:110)
        at com.digitalpetri.netty.fsm.ChannelFsm.getChannel(ChannelFsm.java:84)
        at org.eclipse.milo.opcua.stack.client.transport.tcp.OpcTcpTransport.channel(OpcTcpTransport.java:55)
        at org.eclipse.milo.opcua.stack.client.transport.AbstractTransport.sendRequest(AbstractTransport.java:66)
        at org.eclipse.milo.opcua.stack.client.UaStackClient.sendRequest(UaStackClient.java:220)
        at org.eclipse.milo.opcua.sdk.client.session.SessionFsmFactory.closeSession(SessionFsmFactory.java:725)
        at org.eclipse.milo.opcua.sdk.client.session.SessionFsmFactory.lambda$configureClosingState$36(SessionFsmFactory.java:642)
        at org.eclipse.milo.opcua.sdk.client.session.SessionFsmFactory$$Lambda$366/1381499796.execute(Unknown Source)
        at com.digitalpetri.strictmachine.dsl.ActionBuilder$$Lambda$218/707914145.accept(Unknown Source)
        at com.digitalpetri.strictmachine.dsl.ActionBuilder$PredicatedTransitionAction.execute(ActionBuilder.java:76)
        at com.digitalpetri.strictmachine.StrictMachine$PollAndEvaluate.lambda$run$0(StrictMachine.java:207)
        at com.digitalpetri.strictmachine.StrictMachine$PollAndEvaluate$$Lambda$270/439585081.accept(Unknown Source)
        at java.util.ArrayList.forEach(ArrayList.java:1257)
        at com.digitalpetri.strictmachine.StrictMachine$PollAndEvaluate.run(StrictMachine.java:198)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
"milo-shared-thread-pool-1":
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x0000000772a95280> (a java.util.concurrent.locks.ReentrantReadWriteLock$FairSync)
        at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer.parkAndCheckInterrupt(AbstractQueuedSynchronizer.java:836)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer.acquireQueued(AbstractQueuedSynchronizer.java:870)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer.acquire(AbstractQueuedSynchronizer.java:1199)
        at java.util.concurrent.locks.ReentrantReadWriteLock$WriteLock.lock(ReentrantReadWriteLock.java:943)
        at com.digitalpetri.strictmachine.StrictMachine.getFromContext(StrictMachine.java:110)
        at org.eclipse.milo.opcua.sdk.client.session.SessionFsm.getSession(SessionFsm.java:66)
        at org.eclipse.milo.opcua.sdk.client.OpcUaClient.getSession(OpcUaClient.java:708)
        at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaSubscriptionManager.maybeSendPublishRequests(OpcUaSubscriptionManager.java:423)
        at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaSubscriptionManager.lambda$sendPublishRequest$20(OpcUaSubscriptionManager.java:499)
        at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaSubscriptionManager$$Lambda$599/1764447247.accept(Unknown Source)
        at java.util.concurrent.CompletableFuture.uniWhenComplete(CompletableFuture.java:760)
        at java.util.concurrent.CompletableFuture$UniWhenComplete.tryFire(CompletableFuture.java:736)
        at java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:474)
        at java.util.concurrent.CompletableFuture.completeExceptionally(CompletableFuture.java:1977)
        at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$sendRequest$4(UaStackClient.java:226)
        at org.eclipse.milo.opcua.stack.client.UaStackClient$$Lambda$301/1708478837.accept(Unknown Source)
        at java.util.concurrent.CompletableFuture.uniWhenComplete(CompletableFuture.java:760)
        at java.util.concurrent.CompletableFuture$UniWhenComplete.tryFire(CompletableFuture.java:736)
        at java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:474)
        at java.util.concurrent.CompletableFuture.completeExceptionally(CompletableFuture.java:1977)
        at com.digitalpetri.netty.fsm.util.CompletionBuilders$CompletionBuilder.lambda$with$0(CompletionBuilders.java:75)
        at com.digitalpetri.netty.fsm.util.CompletionBuilders$CompletionBuilder$$Lambda$275/934230292.accept(Unknown Source)
        at java.util.concurrent.CompletableFuture.uniWhenComplete(CompletableFuture.java:760)
        at java.util.concurrent.CompletableFuture$UniWhenComplete.tryFire(CompletableFuture.java:736)
        at java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:474)
        at java.util.concurrent.CompletableFuture.completeExceptionally(CompletableFuture.java:1977)
        at com.digitalpetri.netty.fsm.ChannelFsmFactory.handleConnectFailureEvent(ChannelFsmFactory.java:545)
        at com.digitalpetri.netty.fsm.ChannelFsmFactory$$Lambda$252/1935659603.execute(Unknown Source)
        at com.digitalpetri.strictmachine.dsl.ActionBuilder$$Lambda$218/707914145.accept(Unknown Source)
        at com.digitalpetri.strictmachine.dsl.ActionBuilder$PredicatedTransitionAction.execute(ActionBuilder.java:76)
        at com.digitalpetri.strictmachine.StrictMachine$PollAndEvaluate.lambda$run$0(StrictMachine.java:207)
        at com.digitalpetri.strictmachine.StrictMachine$PollAndEvaluate$$Lambda$270/439585081.accept(Unknown Source)
        at java.util.ArrayList.forEach(ArrayList.java:1257)
        at com.digitalpetri.strictmachine.StrictMachine$PollAndEvaluate.run(StrictMachine.java:198)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)

Found 1 deadlock.
```
Looking forward to your reply，thanks very much!

kevinherron, 2019-05-16:
Thank you, I will fix this in 0.3.1.

kevinherron, 2019-05-16:
In the meantime, I believe you can work around it by doing something like this:

```
        <dependency>
            <groupId>org.eclipse.milo</groupId>
            <artifactId>sdk-client</artifactId>
            <version>0.3.0</version>
            <exclusions>
                <exclusion>
                    <groupId>com.digitalpetri.netty</groupId>
                    <artifactId>netty-channel-fsm</artifactId>
                </exclusion>
            </exclusions>
        </dependency>

        <dependency>
            <groupId>com.digitalpetri.netty</groupId>
            <artifactId>netty-channel-fsm</artifactId>
            <version>0.3</version>
        </dependency>
```

This excludes the dependency on `netty-channel-fsm` version 0.2 from Milo's client and then replaces it with a dependency on version 0.3, which includes changes that should prevent this deadlock.

ligang2495, 2019-05-17:
Thank you very much!

closed 2019-05-17 by ligang2495