eclipse-milo/milo #482: Cannot read Value's Attribute from a Node of NodeClass Variable

mastersteen, 2019-05-24:
Hi Kevin,

I call well known read method: org.eclipse.milo.opcua.sdk.client.OpcUaClient.read(0, TimestampsToReturn.Both, readValueIds); where a pass a predefined Node (Identifier: "Demo.BoilerDemo.Boiler1.TemperatureSensor.Temperature") pointing to a NodeClass Variable Node and got

Value: Variant{value=null}
 StatusCode: StatusCode{name=Bad_AttributeIdInvalid, value=0x80350000, quality=bad}

If use another approach with other method readValue:
org.eclipse.milo.opcua.sdk.client.OpcUaClient.readValue(0, TimestampsToReturn.Both, UaUtil.getNodeId("Demo.BoilerDemo.Boiler1.TemperatureSensor.Temperature"));
I got:

DataValue: Variant{value=0.0}
StatusCode: StatusCode{name=Good, value=0x00000000, quality=good}.

So please have a look and let me know if I do something wrong.

kevinherron, 2019-05-24:
When you call the first method what are you passing as the `readValueIds` parameter?

mastersteen, 2019-05-24:
For example:

ReadValueId readValueId = new ReadValueId(
                                                  UaUtil.getNodeId("Demo.BoilerDemo.Boiler1.TemperatureSensor.Temperature"),
                                                  AttributeId.EventNotifier.uid(),
                                                  null,
                                                  QualifiedName.NULL_VALUE
                                              );
 List<ReadValueId> readValueIds = new ArrayList<>();
        readValueIds.add(readValueId1);

kevinherron, 2019-05-24:
The EventNotifier attribute is not a valid attribute for a VariableNode.

In the second call it's using `AttributeId.Value` behind the scenes. Perhaps that's what you want?

mastersteen, 2019-05-24:
hmm correct it works with  AttributeId.Value...

mastersteen, 2019-05-24:
I will close this issue. Thank you.

kevinherron, 2019-05-24:
Please use the mailing list for future questions :)

closed 2019-05-24 by kevinherron