eclipse-milo/milo #487: Shouldn't Identifiers class better be an enum ???

mojadita, 2019-05-28:
Why that class has not been implemented as an `enum`? Should it be possible to enumerate the `values()` method or to be able to get the `name()` for display purposes?

mojadita, 2019-05-28:
Well I can patch it to be an enum if possible.  I know forcing it to be an enum makes it impossible to create `Node`s freely, but probably `Identifier` can include a `Node` reference in it's definition, so the Node is available then.

kevinherron, 2019-05-28:
I don't think there's a compelling reason to make this change and it would break existing code.

(I do break compatibility between releases, since it's not v1.0 yet, but I try to have a good reason)

closed 2019-05-28 by kevinherron

mojadita, 2019-05-31:
My idea was to associate character strings to the enum values, so they can be used as display strings for the user.  Also, being able to iterate all possible supported values would be something useful in several scenarios.  But anyway, an enum can be constructed, incorporating both, the original `Node` value and the display string, so just forget this.