eclipse-milo/milo #502: Creating ByteString node in a namespace

deify123, 2019-06-19:
Hello,

I am trying to create a ByteString node using Milo 0.2.5 and following the examples I managed to create nodes with Byte, String, Boolean and even an custom enum, but the ByteString doesn't behave like I thought it will.

I create the node with dataType 'Identifiers.ByteString' and even initialize it with 
setValue(new DataValue(new Variant(new ByteString(new byte[0])))); (also tried null or byte[1])
so that the server should see that it is definitely a ByteString but it always marks it as String and upon writing a ByteString via a client i get a Bad_TypeMismatch status.
The same happens when I don't use setValue.

What am I missing?
Thanks in advance

kevinherron, 2019-06-19:
What client are you testing with?

The ExampleServer and ExampleNamespace add a ByteString Node and UaExpert sees it correctly:
[image]

deify123, 2019-06-20:
I am also using UaExpert to view the nodes. And the "DataType" in this view is always String for me.

kevinherron, 2019-06-20:
Without seeing your actual code I'm not sure what the cause could be. The ExampleNamespace from 0.2.5 is doing the same thing you said you're doing and UaExpert sees it correctly.

deify123, 2019-06-24:
Below is the code how I create the Node and a snippet of what UaExpert shows to me:

```public UaVariableNode createByteStringNode(String nodeName, DataValue defaultValue) {
        UaVariableNode result = UaVariableNode.builder(this.nodeMap)
                .setNodeId(new NodeId(this.index, nodeName))
                .setBrowseName(new QualifiedName(this.index, nodeName))
                .setAccessLevel(ubyte(AccessLevel.getMask(AccessLevel.READ_WRITE)))
                .setUserAccessLevel(ubyte(AccessLevel.getMask(AccessLevel.READ_WRITE)))
                .setDisplayName(LocalizedText.english(nodeName))
                .setDataType(Identifiers.ByteString)
                .setTypeDefinition(Identifiers.BaseDataVariableType)
                .build();
        result.setValue(defaultValue);
        return result;
    }

//call
commonVariableCreator.createByteStringNode("Name", new DataValue(new Variant(new ByteString(new byte[0]))));

![image](https://user-images.githubusercontent.com/12694855/59993924-4d4eb800-9652-11e9-95d2-6184f34d942a.png)

CPU1993, 2019-06-24:
Weird, I fiddled around with it a bit more and now it works suddenly, I didn't really change something though.

closed 2019-06-24 by CPU1993