eclipse-milo/milo #505: OPC UA CLIENT JAVA APLICATION WRITE PROBLEM

msamgar, 2019-06-27:
Hello there,
I am trying to write data from Client with OPC UA. I'm using Milo 0.3.1. I cannot write a label that I have successfully read. The type of label I'm trying to write in KepServer is 'Long'. The code I'm trying to write is:
`public void WriteOpcBlocks(OpcRegister[] OpcBlock, Device Device) {
		if(Communicator.OpcClient==null){
			Communicator.Connect();
		}
		for(int i=0;i<OpcBlock.length;i++){
			if(OpcBlock[i]!=null){
				//List<NodeId> nodeIds = ImmutableList.of(new NodeId(Communicator.GetComTypeInfo().GetOpcNodeId(),OpcBlock[i].OpcTagName));
				NodeId nodeIds = new NodeId(Communicator.GetComTypeInfo().GetOpcNodeId(),OpcBlock[i].OpcTagName);
		        
				try {
			            Variant v = new Variant(UInteger.valueOf(5));
			            // don't write status or timestamps
			            DataValue dv = new DataValue(v,null,null);
			            StatusCode status = Communicator.OpcClient.writeValue(nodeIds, dv).get();//statusCodes.get(0);

			            if (status.isGood()) {
			                CommunicationStatus=0;
			            }else{
			            	CommunicationStatus=102;
			            }
				} catch (InterruptedException e) {
					// TODO Auto-generated catch block
					CommunicationStatus=1102;
					if (Communicator.GetComTypeInfo().GetAccessPointLogFlag()) {
						LogManager.info("Opc Okuma başarısız. Cihaz: " + Device.DeviceDescription + " Tip: " + RequestType + " Tag: " + OpcBlock[i].OpcTagName);
					}
					e.printStackTrace();
				} catch (ExecutionException e) {
					// TODO Auto-generated catch block
					CommunicationStatus=1102;
					if (Communicator.GetComTypeInfo().GetAccessPointLogFlag()) {
						LogManager.info("Opc Okuma başarısız. Cihaz: " + Device.DeviceDescription + " Tip: " + RequestType + " Tag: " + OpcBlock[i].OpcTagName);
					}
					e.printStackTrace();
				}
			}else{
				break;
			}
		}
	}
[image]
`

msamgar, 2019-06-27:
The type is in my UInteger code, but I also tried Ulong. Again, my write operation failed. Error code :
StatusCode {name = Bad_TypeMismatch, value = 0x80740000, quality = bad}
OpcBlock [i] .OpcTagName: ModBus TCP.Powys 1 Phase Current Transformer Ratio
Communicator.GetComTypeInfo (). GetOpcNodeId (): 2

msamgar, 2019-06-27:
Sorry. OpcBlock [i] .OpcTagName: ModBus TCP.Powys 1 Faz.Akım Trafo Oran

msamgar, 2019-06-27:
This problem has improved with the following code;
`public void WriteOpcBlocks(OpcRegister[] OpcBlock, Device Device) {
		if(Communicator.OpcClient==null || !Communicator.IsConnectOpc()){
			Communicator.Connect();
			if(Communicator.OpcClient==null || !Communicator.IsConnectOpc()){
				CommunicationStatus=1106;
			}
		}
		for(int i=0;i<OpcBlock.length;i++){
			if(OpcBlock[i]!=null){
				//List<NodeId> nodeIds = ImmutableList.of(new NodeId(Communicator.GetComTypeInfo().GetOpcNodeId(),OpcBlock[i].OpcTagName));
				NodeId nodeIds = new NodeId(Communicator.GetComTypeInfo().GetOpcNodeId(),OpcBlock[i].OpcTagName);
		        
				try {
			            Variant v = new Variant(Integer.parseInt(OpcBlock[i].WriteValueString));
			            System.out.println("start write. val : "+OpcBlock[i].WriteValueString);
			            // don't write status or timestamps
			            
			            DataValue dv = new DataValue(v,null,null);
			            StatusCode status = Communicator.OpcClient.writeValue(nodeIds, dv).get();//statusCodes.get(0);
			            System.out.println("status: " + status.isGood());
			            if (status.isGood()) {
			                CommunicationStatus=0;
			            }else{
			            	CommunicationStatus=1102;
			            }
				} catch (InterruptedException e) {
					// TODO Auto-generated catch block
					CommunicationStatus=1102;
					if (Communicator.GetComTypeInfo().GetAccessPointLogFlag()) {
						LogManager.info("Opc Okuma başarısız. Cihaz: " + Device.DeviceDescription + " Tip: " + RequestType + " Tag: " + OpcBlock[i].OpcTagName);
					}
					e.printStackTrace();
				} catch (ExecutionException e) {
					// TODO Auto-generated catch block
					CommunicationStatus=1102;
					if (Communicator.GetComTypeInfo().GetAccessPointLogFlag()) {
						LogManager.info("Opc Okuma başarısız. Cihaz: " + Device.DeviceDescription + " Tip: " + RequestType + " Tag: " + OpcBlock[i].OpcTagName);
					}
					e.printStackTrace();
				}
			}else{
				break;
			}
		}
	}`

msamgar, 2019-06-27:
I haven't tried other data types yet

kevinherron, 2019-06-27:
Long might just map to Int64 in OPC UA, which would be `new Variant(1L)`

kevinherron, 2019-06-27:
If you connected and browsed with something like UaExpert you could verify.

closed 2019-06-28 by kevinherron