eclipse-milo/milo #506: BadIdentityTokenInvalid for anonymous user

stud0709, 2019-07-01:
Hello Kevin,

in ExampleServer.java (v. 0.3.2), if I exclude identityValidator from serverConfig and try to connect to the server using SecurityPolicy: None and Message Security Mode: None, the client gets BadIdentityTokenInvalid error. 

Is this a desired behaviour? I suppose, there should be no identity check in this case...

Thank you!

BR,

stud0709

kevinherron, 2019-07-01:
This is technically the right behavior for what is not really a valid configuration. 

The closest thing to no identity check is to configure an Anonymous access policy.

stud0709, 2019-07-01:
But USER_TOKEN_POLICY_ANONYMOUS is still defined at line 176.

kevinherron, 2019-07-01:
Oops, I should have read this more carefully. You still need a validator even with anonymous access policy. The `UsernameIdentityValidator` also handles anonymous identity validation but there's nothing stopping you from creating your own. It's important that a validator exists even for anonymous so an object representing the anonymous identity can be returned.

stud0709, 2019-07-02:
OK, many thanks :)

closed 2019-07-02 by stud0709