eclipse-milo/milo #507: Unable to build 0_3_3: NioSocketChannel problem?

WimVriend, 2019-07-02:
Hello Kevin (and others),

At last I have some time to work on my OPC UA client code again. My prototype (which was working fine before) now suddenly fails to read a number of DataTypeDefinitions from my S7-1500 PLC. So decoding data received from items with said type(s) now also fails...

Anyway, I noticed the new release on Github and figured it might be a good idea to try that first?

I downloaded the ZIP-file and ran 'mvn clean install' on my development PC. Unfortunately, an error is thrown:

[image]

The build seems to have successfully downloaded a number of jars from 'central'. Any ideas?

kevinherron, 2019-07-02:
Is it possible your $JAVA_HOME environment variable is pointing to an older version of Java or something?

I had no problem building the latest source download (`mvn clean install`):
```
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  01:13 min
[INFO] Finished at: 2019-07-02T05:53:30-07:00
[INFO] ------------------------------------------------------------------------
kevin@kevin-wx ~/D/milo-master> java -version
openjdk version "1.8.0_222"
OpenJDK Runtime Environment (build 1.8.0_222-b05)
OpenJDK 64-Bit Server VM (build 25.222-b05, mixed mode)
```

WimVriend, 2019-07-02:
Hello Kevin,

Fast as ever :)  

Actually, I think you noticed that I ran 'java -version' after I received the error, because I anticipated that question...  My java version is 1.8.0.172. Is that too old?

kevinherron, 2019-07-02:
It shouldn't be too old, but if `JAVA_HOME` is pointing to some other Java version then Maven will use that instead. 

As for your problem reading some datatypes, it sounds like what was fixed in [this commit](https://github.com/eclipse/milo/commit/078bde34cb94fef52b561aa777a7658e09102638#diff-45ee51937f5b5af95fd0de2e056707e3), which means it's available in all the 0.3.x releases.

kevinherron, 2019-07-02:
FWIW, I was able to obtain an Oracle JDK 8u172 and build successfully with that too. I don't have access to a Windows machine, let alone one with a Cygwin environment set up, to test anything else.

WimVriend, 2019-07-03:
Hello Kevin,

I really appreciate all the things you do and I completely understand that there are so many different possible configurations, that you can't possibly test all. 

You previous comments have indeed helped me! Apparently my development PC was a bit messy. JAVA_HOME did not point to the same version used by Cygwin... I fixed that and now it compiles!

Thanks for also linking to the comment about the fix: I will test it as soon as I can.

kevinherron, 2019-07-03:
Okay let me know. That issue was discovered after Siemens made a change in FW 2.6 that caused a BrowseNext to be necessary while reading the DataTypeDictionary.

and FWIW, I pushed [a commit](https://github.com/eclipse/milo/commit/855a47f91e597626cfdf359170585b6fc0bba030) that should resolve whatever weird issue you had yesterday. The 2 lines that wouldn't compile happened to be using deprecated calls to Netty, so I figured no harm in updating them to the non-deprecated calls, which changes the signature the compiler complained about in the process.

closed 2019-07-03 by kevinherron