eclipse-milo/milo #508: Performance improvements on Create VariableNode

gopalakrishnan-subramani, 2019-07-03:
We are performing load test on OPC UA server with 1/2/3 million UA Variable Node creation.

To create a variable, it takes approximately 4 milli-seconds, 1 million variable needs about 66 minutes to complete.  2 million with 120+ minutes. 

        long start = System.currentTimeMillis();
        UaVariableNode node = new UaVariableNode.UaVariableNodeBuilder(getNodeContext())
                .setNodeId(newNodeId("HelloWorld/ScalarTypes/" + name))
                .setAccessLevel(ubyte(AccessLevel.getMask(AccessLevel.READ_WRITE)))
                .setUserAccessLevel(ubyte(AccessLevel.getMask(AccessLevel.READ_WRITE)))
                .setBrowseName(newQualifiedName(name))
                .setDisplayName(LocalizedText.english(name))
                .setDataType(typeId)
                .setTypeDefinition(Identifiers.BaseDataVariableType)
                .build();

        long end = System.currentTimeMillis();

Tried Object Type Instance with pre-created Object Type, this seems similar 4 milli second per variable, for example to create a object with 1000 variable takes about 4000 milli seconds. 

Can we use separate thread to speed up or any other way to increase the performance?

kevinherron, 2019-07-03:
I think you must be measuring something wrong. Creating 1 million variable nodes using a straightforward single-threaded approach takes less than 3 seconds on my MacBook Pro.

This code won't translate because it's from another project, but it's basically doing the same thing:
```kotlin
    val duration = measureTimeMillis {
        for (i in 0 until 1000) {
            val folderNode = addFolderNode(parentNodeId, "%03d".format(i))

            for (j in 0 until 1000) {
                val name = "%03d".format(j)

                val nodeId = NodeId(namespaceIndex, uint(nextId.getAndIncrement()))

                addVariableNode(folderNode.nodeId, name, nodeId)
            }
        }
    }

    println("Creating 1000000 nodes took ${duration}ms")
```

output during startup:
```
Creating 1000000 nodes took 2582ms
```

kevinherron, 2019-07-03:
And if you're trying to extrapolate the time it will take to create 1, 2, or 3 million nodes from the measurement of a single node that will never be accurate.

closed 2019-07-05 by kevinherron