eclipse-milo/milo #511: Help: Increate serialization buffer size

gopalakrishnan-subramani, 2019-07-06:
We have too large subscription, we get error when as our request size goes above fixed like 4194304. We use OPC UA Milo as client, remote OPC UA server is not Milo based. Any recommendation where to increase the size if required? 

```
ERROR UascServerSymmetricHandler:265 - Error encoding response: StatusCode{name=Bad_ResponseTooLarge, value=0x80B90000, quality=bad}
org.eclipse.milo.opcua.stack.core.UaSerializationException: response exceeds remote max message size: 6314606 > 4194304
at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerSymmetricHandler.checkMessageSize(UascServerSymmetricHandler.java:344)
at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerSymmetricHandler.lambda$sendServiceResponse$1(UascServerSymmetricHandler.java:233)
at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$encode$0(SerializationQueue.java:57)
at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
at java.lang.Thread.run(Thread.java:748)
```

kevinherron, 2019-07-06:
The stack trace you've posted is from Milo running as a server...

Either way, when configuring either a client or server you can set your own `MessageLimits` on the config builder.

gopalakrishnan-subramani, 2019-07-07:
Thank you @kevinherron , I realize this is server error, we will set MessageLimits on config builder. Thanks again for awesome library.

gopalakrishnan-subramani, 2019-07-07:
@kevinherron , Can you please help to validate my understanding here.

https://github.com/eclipse/milo/blob/63c09c13eb7c2368c6b7287b992d05dbb1f420b6/opc-ua-stack/stack-server/src/main/java/org/eclipse/milo/opcua/stack/server/transport/uasc/UascServerSymmetricHandler.java

```java
 private void checkMessageSize(ByteBuf messageBuffer) throws UaSerializationException {
        int messageSize = messageBuffer.readableBytes();
        int remoteMaxMessageSize = serializationQueue.getParameters().getRemoteMaxMessageSize();

        if (remoteMaxMessageSize > 0 && messageSize > remoteMaxMessageSize) {
            throw new UaSerializationException(
                StatusCodes.Bad_ResponseTooLarge,
                "response exceeds remote max message size: " +
                    messageSize + " > " + remoteMaxMessageSize);
        }
    }
```

Is the line checking for the Remote Client Buffer size?         int remoteMaxMessageSize = serializationQueue.getParameters().getRemoteMaxMessageSize();

We tried increasing MessageLimits with config setting on Milo server, but we don't see the setting affecting getRemoteMaxMessageSize() parameter, is it really feasible to set size for remote max message settings?

kevinherron, 2019-07-07:
If you change the MessageLimits of a server config, it's a client that would see the larger remote max message size.

Since the snippet you posted is from server code, the remote max message size is referring to the connected client. Each side reports the max message size they can receive.

gopalakrishnan-subramani, 2019-07-07:
Thanks @kevinherron , I am closing this issue,   the issue should be fixed by the remote client side, not related to Milo Server.

closed 2019-07-07 by gopalakrishnan-subramani