eclipse-milo/milo #512: Discussion: Handling 1 Million Monitored Items on Client and Server SDK

gopalakrishnan-subramani, 2019-07-07:
Seeking a best practices while handling 1 Million Monitored Items for large aggregator/OPC UA Client and Server applications.  

1. Is there any best practices/recommendation to be followed while Subscribing 1 Million Monitored Items from 200+ remote OPC UA servers?
2. any best practices to be handled by OPC UA server while handling 1 Million Monitored Items on server while serving 100+ remote OPC UA Clients subscription?

If we need to customize the MonitoredItem behaviour on both server/client, what could be the classes to inherit and implement in UA server application?

kevinherron, 2019-07-07:
I have not done any testing at this scale so I'm not sure I have any advice to offer offhand.

> If we need to customize the MonitoredItem behaviour on both server/client, what could be the classes to inherit and implement in UA server application?

There's no extension points for this, but the behavior is implemented in the `SubscriptionManager` class in the server. 

The actual sampling of monitored items is handled by your `AddressSpace` implementation, the SDK doesn't implement that for you. This seems like it would be the most important place to have efficient behavior at this scale anyway.

gopalakrishnan-subramani, 2019-07-07:
@kevinherron , The AddressSpace you meant to say example like Milo's Namespace? https://github.com/eclipse/milo/blob/master/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/ExampleNamespace.java

We will look into SubscriptionManager if any customization needed. Can you please give a hint for AddressSpace?

kevinherron, 2019-07-07:
`Namespace` is an `AddressSpace`.

Ultimately, you'll end up implementing the MonitoredItem-related methods from [MonitoredItemServices](https://github.com/eclipse/milo/blob/master/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/api/services/MonitoredItemServices.java).

gopalakrishnan-subramani, 2019-07-07:
Thanks @kevinherron . Session Management functions/services handled as part of NameSpace itself? 
In Milo, is sessions are in-memory based, ie if the server restart, it cannot remember existing session details and cannot restore subscriptions? Is any Session Persistency interface available if the server reboot?

kevinherron, 2019-07-07:
No, Sessions are not persistent (nothing is persistent) and the `SessionManager` is not extensible either.

kevinherron, 2019-07-10:
I'm going to close this, feel free to pick this up on the mailing list if you have more questions.

closed 2019-07-10 by kevinherron