eclipse-milo/milo #514: References with custom ReferenceType not published

6d77, 2019-07-22:
I'm creating an OPC UA server with milo and adding a custom ReferenceType in a custom namespace as follows:

```java
    	UaReferenceTypeNode refType = new UaReferenceTypeNode(
    			getNodeContext(), 
    			nodeId, 
    			new QualifiedName(getNamespaceIndex(), browseName), 
    			LocalizedText.english(displayName), 
    			LocalizedText.NULL_VALUE, 
    			UInteger.valueOf(0), 
    			UInteger.valueOf(0), 
    			false, 
    			false, 
    			LocalizedText.english(inverse));
    	
		createLink(parent, Identifiers.HasSubtype, nodeId, true);
        getNodeManager().addNode(refType);
```

The type nodes are visible below "HierarchicalReferences" on the client (UaExpert). However when I use this ReferenceType to create a reference between two objects, this reference is not published and not visible in UaExpert. The reference is created with code like

```java
            node.addReference(new Reference(
                    node.getNodeId(),
                    link,
                    to.expanded(),
                    true
                ));
```

If I -- just for testing -- replace `link` by `Identifiers.Organizes` or any other standard ReferenceType from namespace 0, then I can see these references in UaExpert.

Are there any other points where I have to register my custom ReferenceType in order to make it usable just like the standard ReferenceTypes?

kevinherron, 2019-07-22:
Just to be clear - `createLink` is adding a HasSubtype reference from `Identifiers.HierarchicalReferences` to your `refType`, yeah?

I think the missing piece might be registering a `ReferenceType` with the map from `OpcUaServer#getReferenceTypes`.

If that doesn't work we'll have to keep troubleshooting. I haven't ever tried or tested using a custom reference type.

6d77, 2019-07-23:
> Just to be clear - createLink is adding a HasSubtype reference from Identifiers.HierarchicalReferences to your refType, yeah?

Yes. It basically retrieves the node referenced by the first parameter and then calls `node.addReference`. 

I then tried to register a `ReferenceType` with the map you mentioned. I created a `CustomReferenceType implements ReferenceType` and used it as follows:

```java
        CustomReferenceType referenceType = new CustomReferenceType(nodeId, new QualifiedName(getNamespaceIndex(), browseName), false, false);
        referenceType.setInverseName(inverse);
        referenceType.setSuperTypeId(parent);
        getServer().getReferenceTypes().put(nodeId, referenceType);
```
 And in fact, this did the trick!

Just a idea to help others: Maybe the `NodeManager` could automatically add such a `CustomReferenceType` when a `UaReferenceTypeNode` is added and the `ReferenceType` is not already registered by the fixed set used for initialization.

kevinherron, 2019-07-23:
I'll try to get a custom reference example added to `ExampleNamespace` and take your suggestion into consideration. I'll have to weigh the performance impact of doing that check though, because if you're using NodeFactory to create instances or importing NodeSet XMLs (via the not yet released tools to do this...) the performance hit might add up.

closed 2019-07-23 by kevinherron