eclipse-milo/milo #520: Does milo support opc ua part 14 pub/sub?

SnailGetRed, 2019-08-06:
OPC UA part 14 - PubSub was released last year. It defines new PubSub communication model. Does Milo support or implement it?

kevinherron, 2019-08-06:
No, it doesn't. PubSub is defined in OPC UA 1.04. 

Milo implements OPC UA version 1.03 right now, and will continue to do so while I move towards a version 1.0.

Immediately after version 1.0 is released I plan to make a version 2.0 branch and begin work on OPC UA 1.04, but that's gonna be a while.

SnailGetRed, 2019-08-07:
You mean that version 1.0 will implement all features of OPC UA version 1.03 and version 1.x will implement features of OPC UA version 1.04, right? According to your plan, when will version 1.0 be released? :). Prosys Java OPC UA sdk may not implement pub/sub yet, So I have to implement it myself if i want to use pub/sub.

kevinherron, 2019-08-07:
OPC UA 1.03 is implemented right now AFAIK, it's just that I don't have any plans to work on support 1.04 *before* I release 1.0 because it's going to take significant effort.

I'm going to bump the version all the way to 2.0 when I implement 1.04. Versions 1.x.x will be for bug fixes, API changes/enhancements, and features in the 1.0 series.

kevinherron, 2019-08-07:
Oh, and for the timeline... I'm really hoping to be able to release 1.0 in the next few months. I don't think there's much left blocking me, other than maybe documentation...

I'm working on 0.4.0 right now, which is mostly just re-generated structs and trying to make sure some codegen and nodeset loading scenarios work and won't require significant API changes.

closed 2019-08-11 by kevinherron