eclipse-milo/milo #522: Threading model for OPC UA server

gopalakrishnan-subramani, 2019-08-06:
We get below exceptions, related to client as the file says UascClientMessageHandler, but how to capture this error, should the application reconnect after this exceptions?

2019-08-06 15:36:18 ERROR UascClientMessageHandler:169 - [remote=/192.168.161.22:9090] Exception caught: Connection reset by peer java.io.IOException: Connection reset by peer at sun.nio.ch.FileDispatcherImpl.read0(Native Method) at sun.nio.ch.SocketDispatcher.read(SocketDispatcher.java:39) at sun.nio.ch.IOUtil.readIntoNativeBuffer(IOUtil.java:223) at sun.nio.ch.IOUtil.read(IOUtil.java:192) at sun.nio.ch.SocketChannelImpl.read(SocketChannelImpl.java:380) at io.netty.buffer.PooledUnsafeDirectByteBuf.setBytes(PooledUnsafeDirectByteBuf.java:288) at io.netty.buffer.AbstractByteBuf.writeBytes(AbstractByteBuf.java:1125) at io.netty.channel.socket.nio.NioSocketChannel.doReadBytes(NioSocketChannel.java:347) at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:148) at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:677) at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:612) at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:529) at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:491) at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:905) at java.lang.Thread.run(Thread.java:748)

kevinherron, 2019-08-06:
This error means the server closed the connection for some reason.

It is not bubbled up to the client; the client will automatically reconnect
and resume its old session or establish a new one if necessary. It will
continue to attempt this until you call OpcUaClient#disconnect(). If you
added a SessionActivityListener (OpcUaClient#addSessionActivityListener)
you will be notified when the session goes inactive and when it is
subsequently active again.

gopalakrishnan-subramani, 2019-08-07:
Thank you Kevin. Does Milo client keep trying to connect at this scenarios? I see a thread level exception, curious know the Milo specific behaviour..

kevinherron, 2019-08-07:
Yes, Milo will continue to reconnect in any unexpected error or disconnect scenario. The high level behavior is described in OPC UA Part 4.

closed 2019-08-11 by kevinherron