eclipse-milo/milo #528: Server example with events is missing

achintya-kumar, 2019-08-10:
Hi,

Thanks for the great work you're doing. :) 
I'm working on building a server and I have a requirement to add events to the server. It seems that there aren't any example for that.
Am I missing something?

Regards, 
Achintya Kumar

kevinherron, 2019-08-10:
The example namespace does post an event every 2 seconds:

https://github.com/eclipse/milo/blob/5e1d12054df4a11a9c8806fb9c4418ed0c3dbf63/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/ExampleNamespace.java#L173-L202

Since in this case the Server object is the notifier, the built-in OpcUaNamespace is actually where the event item gets created and hooked up:

https://github.com/eclipse/milo/blob/5e1d12054df4a11a9c8806fb9c4418ed0c3dbf63/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/namespaces/OpcUaNamespace.java#L129-L150

There's not much more to it than that. If you use your own object(s) as notifiers then the namespace they reside in will need to have those event-related monitored item methods implemented. 

It's perfectly acceptable to use the Server object as the only event notifier in the server, though.

achintya-kumar, 2019-08-10:
OMG,

Thank you very much! I wasn't expecting such a prompt and positive response! :D

achintya-kumar, 2019-08-11:
Hey Kevin,

One more question: how do alarms work in Milo? I would like to add alarms on a server built using Milo.

Thank you!

kevinherron, 2019-08-11:
Alarms and Conditions in OPC UA are built in top of the concept of events. The server SDK does not implement any of these for you, however. If you want alarm functionality you have to build it yourself on top of the event support.

achintya-kumar, 2019-08-11:
[image]

[image]

In the screenshots above, I have changed the EventType to AlarmConditionType. After that, I'm setting the activeState property of the instantiated AlarmConditionNode. However, I'm unable to see the property in the event-view tab of UaExpert. Am I using it correctly?

kevinherron, 2019-08-11:
Hmm. I looked at this a bit, and there's a bug in the current set of generated Nodes.

Using ExclusiveLimitAlarmNode as an example, you can see the setActiveState method:
https://github.com/eclipse/milo/blob/5e1d12054df4a11a9c8806fb9c4418ed0c3dbf63/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/model/nodes/objects/ExclusiveLimitAlarmNode.java#L59-L63

Compared to getActiveStateNode method:
https://github.com/eclipse/milo/blob/5e1d12054df4a11a9c8806fb9c4418ed0c3dbf63/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/model/nodes/objects/ExclusiveLimitAlarmNode.java#L47-L51

the `setActiveState` method is missing the namespace qualifier when fetching the component. This is causing the component node not to be found and your call to set the state to do nothing.

As a workaround until it's fixed, try doing this:
```
eventNode.getActiveStateNode().setValue(
    new DataValue(new Variant(LocalizedText.english("Active")))
);
```

achintya-kumar, 2019-08-19:
Hi Kevin! Sincere apologies for reopening this issue for the second time. :)

I'm trying to read this attribute of an event from a Milo client. I keep getting null in the event-consumer.

[image]

[image]

Also, is there a way to bypass EventFilter and read everything in a received event?

kevinherron, 2019-08-19:
Where is this Event coming from?

I don't see anything in the spec that indicates you can do something "select *" on fields, it seems you have to explicitly list the event fields you are interested in. I'm not an expert on Events, though.

kevinherron, 2019-08-19:
Ah, from Part 9:

[image]

You'd probably create your `SimpleAttributeOperand` like this:
```java
new SimpleAttributeOperand(
    Identifiers.ConditionType,
    new QualifiedName[0],
    AttributeId.NodeId.uid(),
    null
)
```

closed 2019-08-21 by kevinherron