eclipse-milo/milo #531: How to addNodes in milo client?

SnailGetRed, 2019-08-14:
Can you give a sample of addNodes service in client? I try to implement it myself, but it does not work.
question 2, does milo server supports to receive addNodes requests from client? When I use milo client to send a addNodes to the milo server, it responses a Bad_ServiceUnsupported status.

kevinherron, 2019-08-14:
The server doesn't implement the Node Management services by default. 

It's possible, by overriding `addNodes`, `removeNodes`, `addReferences`, and `removeReferences` in your `AddressSpace` implementation,  but there's no sensible default implementation for this. You'd just end up with dummy nodes with no behavior or values.

It sounds like you're probably using it right in the client but you just need a server that implements NodeManagement to test against.

SnailGetRed, 2019-08-14:
yes, I use it in the client, but I don't know whether it's correct or not. I don't know how to assign the value of `nodeAttributes`. 
what you mean is that `addNodes`, `removeNodes`, `addReferences`, and `removeReferences` is useless, because nodes are managed by the server, right?

kevinherron, 2019-08-14:
There's a structure `NodeAttributes` with subtypes for each Node type. You would create one of those appropriate for the node you're creating and then encode with `ExtensionObject#encode`.

The calls aren't necessarily useless, it's just that the server must implement the Node Management services and does something useful with Nodes when it creates them. I suppose the example server could just create dummy Nodes in the address space with no logic behind them, though.

SnailGetRed, 2019-08-15:
thank you, kevin. Can you help me to view my code of addNodes
`ExpandedNodeId parentNodeId = new ExpandedNodeId(Identifiers.Server);
        NodeId referenceTypeId = Identifiers.HasComponent;
        ExpandedNodeId requestedNewNodeId = new ExpandedNodeId(new NodeId(0, 9999));
        QualifiedName browseName = new QualifiedName(0, "testObject");
        NodeClass nodeClass = NodeClass.Object;
        ExtensionObject nodeAttributes = ExtensionObject.encode(
                client.getSerializationContext(),
                new ByteString("attr".getBytes()),
                Identifiers.ActivateSessionRequest_Encoding_DefaultBinary,
                OpcUaDefaultBinaryEncoding.getInstance()
        );
        /*ExtensionObject nodeAttributes = new ExtensionObject(
                new ByteString("attr".getBytes()), Identifiers.ActivateSessionRequest_Encoding_DefaultBinary);*/
        ExpandedNodeId typeDefinition = new ExpandedNodeId(Identifiers.BaseObjectType);
        AddNodesItem node = new AddNodesItem(
                parentNodeId, referenceTypeId, requestedNewNodeId,
                browseName, nodeClass, nodeAttributes, typeDefinition);
        List<AddNodesItem> nodesToAdd = new ArrayList<>();
        nodesToAdd.add(node);
        AddNodesResponse response = client.addNodes(nodesToAdd).get();
        log.info("add nodes response: {}", response);`

kevinherron, 2019-08-15:
I think the attributes should look something like this:
```java
        ObjectAttributes attributes = new ObjectAttributes(
            uint(0b11111),
            LocalizedText.english("My Object"),
            LocalizedText.english("Description of My Object"),
            uint(0),
            uint(0),
            ubyte(0)
        );

        ExtensionObject encodedAttributes = ExtensionObject.encode(
            client.getSerializationContext(), 
            attributes
        );
```

SnailGetRed, 2019-08-16:
Thank you kevin. I think it works although I can not verify it :) When milo server or prosys server is used, Bad_ServiceUnsupported is received, and when UaCPPServer is used, Bad_UserAccessDenied is received.

kevinherron, 2019-08-20:
> Another problem confusing me is why only one attribute can be created with the AddNodesItem.

I'm not sure what you mean by this. There is one `AddNodesItem` per Node you are creating, and each of those items has a `NodeAttributes` structure that lets you specify *all* the attributes for the type of Node being created.

SnailGetRed, 2019-08-20:
I'm clear now. Thank you kevin. I think this issue can be closed.

kevinherron, 2019-08-20:
Ok. Tracking an example of Node Management services in #538

closed 2019-08-20 by kevinherron

Rikom123, 2022-05-01:
Have you successfully achieved the increase in nodes?