eclipse-milo/milo #533: Prosys OPC UA client Events reading

AngelThread, 2019-08-18:
Hi,

Thank you for the all the effort, appreciated.

I am new to Milo and I was trying to get familiar with Events by running ExampleServer. What I noticed is that it is possible to read events by UA Expert successfully but it is not the case with Procsys OPC UA client. It is probably related to Procsys client but I had spent some time on this to discover what the case is, before learning it is not the case with UA expert, thanks to @achintya-kumar.
[image]
[image]

kevinherron, 2019-08-18:
Are there any errors in the Prosys client or its logs?

kevinherron, 2019-08-19:
I looked into this a little more - Prosys client isn't working because the `OfType` FilterOperator is not implemented in Milo. 

You can see in Wireshark that Milo server responds with `Bad_FilterOperatorUnsupported` for the corresponding part of the where clause but it seems the Prosys client doesn't report this in the UI in any way.

I'll see if I can get it implemented for for the 0.4.0 release.

kevinherron, 2019-08-20:
Closing this, feature can be tracked in #537

closed 2019-08-20 by kevinherron