eclipse-milo/milo #534: Bad Node Id Unknown error when trying to read values from OPC UA server.

Kbhargav04, 2019-08-19:
Im trying to read values from GE Cimplicity OPC UA server.
Im getting BadNodeIdUnknown  Exception.
using 3 as the node Id to read from all the tags.
But some tags are giving me this error.
Request your help on this.

kevinherron, 2019-08-19:
There's not really anything to troubleshoot on the Milo side of this.

Either you have the wrong NodeId or there's something wrong with the server.

closed 2019-08-21 by kevinherron

Kbhargav04, 2019-08-26:
Hi kevin ,
this is the log i received in the server.
have you face any problems like this?

15:26:20.028Z|4|6020* ==> UaServer::Read [Request=4]
15:26:20.028Z|6|6020* --> UaSession::startingServiceProcessing [ID=-738963900]
15:26:20.028Z|6|6020* <-- UaSession::startingServiceProcessing - activeServiceCount = 1
15:26:20.028Z|6|6020* --> UaTransactionManager::beginRead with 1 nodes
15:26:20.028Z|7|6020* Read Request [Request=4]:
15:26:20.028Z|7|6020*   MaxAge             = 0.000000 ms
15:26:20.028Z|7|6020*   TimestampsToReturn = Both
15:26:20.028Z|7|6020*   NoOfNodesToRead = 1
15:26:20.028Z|7|6020*     Node[0] NS3|String|PTK_PM_606_12_1404.AE.VALUE Attribute=13
15:26:20.028Z|1|6020*     Node[0] NS3|String|PTK_PM_606_12_1404.AE.VALUE Attribute=13 OpcUa_BadNodeIdUnknown

Kbhargav04, 2019-08-26:
other tags i am reading is present in the same project are providing the value for nodeId 3 and i am able to write values to many tags with the same node id in the project

akileswaran, 2020-04-09:
hi,
I have the same problem. I tried using the Ua Expert client and it is able to read the value from the server but the client that I am using is giving the exception.

kevinherron, 2020-04-09:
@akileswaran make sure you are using the same `NodeId`. Provide screenshots from UaExpert and an example of the code you're using.

umairGithub2282, 2022-12-21:
Hi ,

I am facing the same issue in OPC UA communication .

I have checked with the Kepware Client OPC UA and Matrikon OPC UA software.

[BadNodeIdUnknown (0x80340000)

Can You please support

[image]
[image]

Kbhargav04, 2022-12-21:
Hi,

The issue was solved by using the correct data type while reading the
values. you may need to either change data type configured in the server or
the data type you are using to read it.

regards
krishna bhargav