eclipse-milo/milo #539: Created Alarms doesn't disappear on

AngelThread, 2019-08-20:


AngelThread, 2019-08-20:
Hi,
I have been creating my own alarms as bellow, alarms appear on UA expert as expected but they don't disappear after some time as they are supposed to be. What I am doing is reading all events from another source and conveying it to another OPC UA server. Could you tell me if I am missing a point?

                        AlarmConditionNode eventNodeAlarm = (AlarmConditionNode) server.getServer().getEventFactory().createEvent(
                                new NodeId(1, UUID.randomUUID()),
                                Identifiers.AlarmConditionType
                        );
                        eventNodeAlarm.setActiveState(eventDeserialized.getActiveState());
                        eventNodeAlarm.setBranchId(eventDeserialized.getBranchId());
                        eventNodeAlarm.setConditionName(eventDeserialized.getConditionName());
                        eventNodeAlarm.setEventId(eventDeserialized.getEventId());
                        eventNodeAlarm.setEventType(Identifiers.AlarmConditionType);
                        eventNodeAlarm.setMessage(eventDeserialized.getMessage());
                        eventNodeAlarm.setRetain(eventDeserialized.getRetain());
                        eventNodeAlarm.setSeverity(eventDeserialized.getSeverity());
                        eventNodeAlarm.setSourceName(eventDeserialized.getSourceName());
                        eventNodeAlarm.setTime(eventDeserialized.getUtcTime());
                        eventNodeAlarm.getActiveStateNode()
                                .setValue(new DataValue(new Variant(eventDeserialized.getActiveState())));
                        eventNodeAlarm.setAckedState(eventDeserialized.getAckedState());
[image]

kevinherron, 2019-08-20:
I don't know how or when UaExpert decides to remove alarm events from that list. You might need to ask them.

Do make sure you call `delete()` on the event nodes after they've been posted to the event bus, though, or you'll leak memory.

AngelThread, 2019-08-20:
Thank you for the information. I was thinking maybe it is because of the type of alarm that I am using or I am missing setting a field.

closed 2019-08-21 by kevinherron

juangburgos, 2020-04-18:
@UgurYeter did you ever find out what field was missing?