eclipse-milo/milo #545: Stopping server won't disconnect clients

Christiaan676, 2019-08-28:
Calling `shutdown` on the `OpcUaServer` seams to be closing the bind ports correctly and no new connections are accepted, but the connected Opc-UA clients will remain connected. Event if I use the session manager and kill the session it will only generate an error at the client side and reconnect the session.

To be sure is calling shutdown the correct way to completely shut down the server?

Christiaan676, 2019-08-29:
Yes that way the clients will disconnect and won't be able to reconnect.

Thanks for the quick response!

kevinherron, 2019-08-29:
Okay. I'll make closing these channels part of shutdown. In the meantime you can close them yourself as a workaround.

PR #554 (2019-10-05): Explicitly close connected channels during UaStackServer shutdown
fixes #545
commit 597735b2 (2019-10-05): Explicitly close connected channels during UaStackServer shutdown (#554)

fixes #545
closed 2019-10-05 by kevinherron