eclipse-milo/milo #546: Increasing delay in event creations and event posts

achintya-kumar, 2019-08-30:
Hi Kevin,

I am noticing this behavior in Milo where creation of Events as well as posting of events to the EventBus keeps stacking up delays.
The following are the times (in ms) taken in event creation using `server.getEventFactory().createEvent(...) `and event-posting using `eventBus.post(...)`.

- 5961, 148
- 3223, 119
- 3165, 221
- 3319, 336
- 3532, 588
- 6384, 1355
- 4468, 956
- 3558, 1046
- 3509, 1313
- 3880, 1666

As per your recommendation, I am deleting the event created. What could be the reason behind such delays?

kevinherron, 2019-08-30:
How are you measuring these? What are the first and second number on each bullet point in your list? What version are you using?

Does the example server show the same thing when you leave it running?

achintya-kumar, 2019-08-31:
Sincere apologies. It's just the event creation which is taking time. The event posting is fine. My guess for the delay is that there's about 45 namespaces at the moment on my OPC-UA served.

closed 2019-08-31 by achintya-kumar