eclipse-milo/milo #549: Restrict the OPC UA client for communication

DhruvilDhanani, 2019-09-23:
In the server below are the endpoints.
opc.tcp://192.168.XXX.XXX:12686/milo [Basic256/Sign]
opc.tcp://192.168.XXX.XXX:12686/milo/discovery[None/None].

If we are using the above endpoints(OPC milo example server) then client with security policy[None] and security Message Mode[None] also able to do the communication(Able to accessing the node in the namespaces).
Client using the endpoint url is "opc.tcp://192.168.XXX.XXX:12686/milo/discovery".

In third party OPC server there is provision for restricting the client whose security policy is 
not matching with server endpoint url.

For example: If we configure the prosys OPC UA server with endpoint 
[opc.tcp://192.168.XXX.XXX:12686/opcserver [Basic256/Sign]].  Then client whose security policy[Basic256] and security Message Mode[sign] are only allowed for communication.

kevinherron, 2019-09-23:
The discovery endpoint only allows clients to use the discovery services.

If you have this setup correctly a client will only be able to complete a connection and access nodes on the secured endpoint.

DhruvilDhanani, 2019-09-23:
@kevinherron 

Thanks for quick reply.  Actually i have misplaced my question, Sorry for that.

My concern is as per below. 

Suppose i have configure the OPC server as per below Endpoints. 

> opc.tcp://192.168.XXX.XXX:12686/milo [Basic256/Sign]
> opc.tcp://192.168.XXX.XXX:12686/milo/discovery[None/None].

In the client side I am using the Prosys OPC UA browser .

>  If In the browser if i provide the endpoint url as : opc.tcp://192.168.10.126:12686/milo/discovery  [Basic256/Sign]  then able to access the nodes.

[OPC_UA_MILO_SERVER_DISCOVERY_ISSUE.log](https://github.com/eclipse/milo/files/3642212/OPC_UA_MILO_SERVER_DISCOVERY_ISSUE.log)

kevinherron, 2019-09-23:
The log seems to indicate things are working as they should.

Here, the client is calling discovery services against the discovery endpoint (path=/milo/discovery):
```
onServiceRequest(path=/milo/discovery, request=ServiceRequest{request=GetEndpointsRequest})
onServiceRequest(path=/milo/discovery, request=ServiceRequest{request=GetEndpointsRequest})
```

Then subsequent connection and requests are all against the secured endpoint (path=/milo):
```
onServiceRequest(path=/milo, request=ServiceRequest{request=CreateSessionRequest})
onServiceRequest(path=/milo, request=ServiceRequest{request=ActivateSessionRequest})
onServiceRequest(path=/milo, request=ServiceRequest{request=ReadRequest})
```

The only thing we need to verify is that the client was forced to use security when it connected to the second endpoint.

Can you get a Wireshark capture or tell me which version of the Prosys client you're using?

DhruvilDhanani, 2019-09-24:
@kevinherron 

If i provide the endpoint URL in prosys browser as a

>  opc.tcp://192.168.10.126:12686/milo/discovery [None/None] 

then it should connect with OPC milo server. 

Are you agree with me?

Prosys OPC UA Browser Product version :4.0.0-200
Sdk version : 4.1.2-876

Let me know if this is not sufficient , I will capture the Wire shark log if requires.

kevinherron, 2019-09-24:
They aren't forced to use security while using discovery services. I think this is a quirk with the way the Prosys client and it looks like you have to enable security for discovery but what you're really doing is telling it to use security after discovery. 

You can verify this with Wireshark captures.

You can also use UaExpert 1.5.1+ and verify this as well.

kevinherron, 2019-09-24:
I downloaded the new Prosys browser and confirmed things are working as they should. It just presents a little bit strange in the Prosys user interface.

Some clients aren't able to connect at all when you have a server configured with a separate unsecured discovery endpoint and another security-only endpoint, so I guess we should just be happy it even works.