eclipse-milo/milo #550: How to connect client and server

jdhanasekar, 2019-09-30:
Hi Milo,
           I am new to OPC UA and Milo.  My task is to connect client and server.  I am trying to under stand how connection is established. I am unable to understand.

  When I run this command,
```
milo-master/milo-examples/client-examples# mvn exec:java -D"exec.mainClass"="org.eclipse.milo.examples.client.ClientExampleRunner"

```

I am getting below error,

```
[INFO] Scanning for projects...
[WARNING] 
[WARNING] Some problems were encountered while building the effective model for org.eclipse.milo:client-examples:jar:0.3.4-SNAPSHOT
[WARNING] 'build.plugins.plugin.version' for org.apache.maven.plugins:maven-deploy-plugin is missing. @ org.eclipse.milo:client-examples:[unknown-version], /home/user/dhanasekar/BMS/OPC/milo/milo-master/milo-examples/client-examples/pom.xml, line 81, column 21
[WARNING] 
[WARNING] It is highly recommended to fix these problems because they threaten the stability of your build.
[WARNING] 
[WARNING] For this reason, future Maven versions might no longer support building such malformed projects.
[WARNING] 
[INFO] 
[INFO] ------------------< org.eclipse.milo:client-examples >------------------
[INFO] Building client-examples 0.3.4-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- exec-maven-plugin:1.6.0:java (default-cli) @ client-examples ---
[WARNING] 
java.lang.Exception: The specified mainClass doesn't contain a main method with appropriate signature.
    at org.codehaus.mojo.exec.ExecJavaMojo$1.run (ExecJavaMojo.java:286)
    at java.lang.Thread.run (Thread.java:748)
Caused by: java.lang.NoSuchMethodException: org.eclipse.milo.examples.client.ClientExampleRunner.main([Ljava.lang.String;)
    at java.lang.Class.getMethod (Class.java:1786)
    at org.codehaus.mojo.exec.ExecJavaMojo$1.run (ExecJavaMojo.java:270)
    at java.lang.Thread.run (Thread.java:748)
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  1.178 s
[INFO] Finished at: 2019-09-30T11:08:19+05:30
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.codehaus.mojo:exec-maven-plugin:1.6.0:java (default-cli) on project client-examples: An exception occured while executing the Java class. The specified mainClass doesn't contain a main method with appropriate signature.: org.eclipse.milo.examples.client.ClientExampleRunner.main([Ljava.lang.String;) -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

```

Can  some one please me to understand milo and its client/sever connection?.
I dont know pre requisite things for client/server.
I am running ```milo``` in my server and I am having ```movicon``` in my desktop. I am try in to connect movicon with milo
Please help me on this

kevinherron, 2019-09-30:
`ClientExampleRunner` doesn't have a `main` method, so you can't run it directly.

Try running one of the examples instead, such as `org.eclipse.milo.examples.client.ReadExample`.

jdhanasekar, 2019-09-30:
Thanks for the info. I will  try and update you

jdhanasekar, 2019-10-01:
with ```ReadExample```,  jar file is not getting created in ```milo-master/milo-examples/client-examples```

jdhanasekar, 2019-10-01:
When I try to start milo server by running server example ```Exampleserver```, I am gettinb below error,

```
root@quanta:/home/user/dhanasekar/BMS/OPC/milo/milo-master/milo-examples/server-examples#  mvn exec:java -D"exec.mainClass"="org.eclipse.milo.examples.client.ExampleServer"
[INFO] Scanning for projects...
[WARNING] 
[WARNING] Some problems were encountered while building the effective model for org.eclipse.milo:server-examples:jar:0.3.4-SNAPSHOT
[WARNING] 'build.plugins.plugin.version' for org.apache.maven.plugins:maven-deploy-plugin is missing. @ org.eclipse.milo:server-examples:[unknown-version], /home/user/dhanasekar/BMS/OPC/milo/milo-master/milo-examples/server-examples/pom.xml, line 77, column 21
[WARNING] 
[WARNING] It is highly recommended to fix these problems because they threaten the stability of your build.
[WARNING] 
[WARNING] For this reason, future Maven versions might no longer support building such malformed projects.
[WARNING] 
[INFO] 
[INFO] ------------------< org.eclipse.milo:server-examples >------------------
[INFO] Building server-examples 0.3.4-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- exec-maven-plugin:1.6.0:java (default-cli) @ server-examples ---
[WARNING] 
java.lang.ClassNotFoundException: org.eclipse.milo.examples.client.ExampleServer
    at java.net.URLClassLoader.findClass (URLClassLoader.java:382)
    at java.lang.ClassLoader.loadClass (ClassLoader.java:424)
    at java.lang.ClassLoader.loadClass (ClassLoader.java:357)
    at org.codehaus.mojo.exec.ExecJavaMojo$1.run (ExecJavaMojo.java:270)
    at java.lang.Thread.run (Thread.java:748)
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  1.016 s
[INFO] Finished at: 2019-10-01T16:33:59+05:30
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.codehaus.mojo:exec-maven-plugin:1.6.0:java (default-cli) on project server-examples: An exception occured while executing the Java class. org.eclipse.milo.examples.client.ExampleServer -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
root@quanta:/home/user/dhanasekar/BMS/OPC/milo/milo-master/milo-examples/server-examples# 
```

kevinherron, 2019-10-01:
It doesn't look like the classpath is correct when using that plugin.

I don't use the maven exec plugin and none of these examples have been built or tested with it. They're meant to be run from within an IDE that has the Milo project open (and possibly after you've run `mvn clean install` from the root, depending on the IDE).

jdhanasekar, 2019-10-01:
I am running project from command line.  From  your suggestion, Examples in server-examples, client-examples and standalone-examples are should be run from eclipse IDE.

When I run ```mvn clean install``` from the root in command line, ```server-examples.jar``` and ```orginal-server-examples.jar``` are getting created. Is these enough to connect to client?

In eclipse IDE, when I given ```Run as```, I am getting below error,\

```
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-deploy-plugin:2.7:deploy (default-deploy) on project build-tools: Failed to deploy artifacts: Could not transfer artifact org.eclipse.milo:build-tools:jar:0.3.4-20191001.141401-8 from/to ossrh (https://oss.sonatype.org/content/repositories/snapshots): Access denied to https://oss.sonatype.org/content/repositories/snapshots/org/eclipse/milo/build-tools/0.3.4-SNAPSHOT/build-tools-0.3.4-20191001.141401-8.jar. Error code 401, Unauthorized -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
```

kevinherron, 2019-10-01:
It looks like you ran a build target that tried to deploy, which you can't do because you don't have access to the Sonatype repos. 

I think there's a couple posts on the mailing list where people asked for help getting things running in Eclipse. I don't use Eclipse so I'm not much help there.

jdhanasekar, 2019-10-01:
I have set ```Goal``` as ```deploy```. So that eclipse IDE is throwing error.  What can I set for ```Goal``` when I running POM file

kevinherron, 2019-10-01:
Try `install` if that's an option.

jdhanasekar, 2019-10-01:
It is working. I am not seeing any error.
I am seeing ```client-examples-0.3.4-SNAPSHOT.jar```  in client-example. 
When I try run client-examples, I am getting below error. I am debugging, 

```
$ sudo java -jar client-examples-0.3.4-SNAPSHOT.jar 
[sudo] password for user: 
no main manifest attribute, in client-examples-0.3.4-SNAPSHOT.jar
````

kevinherron, 2019-10-01:
Those jars are not meant to be runnable.

As I said, these examples are meant to run from the IDE. There are many classes with a `main` method in the client examples. There is no configuration in the pom files to turn these into runnable jars or specify main methods.

jdhanasekar, 2019-10-01:
Ok.. got it, I will do this

closed 2019-10-03 by kevinherron