eclipse-milo/milo #551: Mapping a OpcUaMonitoredItem Object to corresponding subscriptionId

hogwild65, 2019-09-30:
I have a project where I am connecting to factory OPC Server thru Milo OPC client. I have existing automation application that acts as client to the Milo OPC client setting up data subscriptions and routing data messages back to the subscribing automation clients. The automation client has requirement to send back other information with the data message like collection event ids, SECS/GEM stream and function info. There can be many automation clients subscribing and could be same OPC nodeId but different additional data so I need a way to determine at onSubscriptionValue event the corresponding OpcUaSubscription subscriptionId. 
    I tried to get to the subscriptionId using available getters and getValues but did not work for me. Tried to extend the UaMonitoredItem interface to include a subscriptionId field and a setter, getter. This snowballed on me because other classes  (UaSubscription, OpcUaSubscription, UaSubscriptionManager, OpcUaSubscriptionManager, probably more..) did not recognize my new type and type casting fails at runtime. 
    I found and tested  simple solution:
    Add these two methods to UaMonitoredItem interface
   UInteger getASubscribeId();
   void setASubscribeId(UInteger aSubscribeId);
   Add to the implementing class OpcUaMonitoredItem:
  private UInteger aSubscribeId;
  public UInteger getASubscribeId() {
    	return aSubscribeId;
    }
  public void setASubscribeId(UInteger aSubscribeId) {
    	this.aSubscribeId = aSubscribeId;
    }
The problem with this is it makes upgrading to newer milo versions more difficult because I have to make these changes to new milo every time and rebuild. I have signed up on eclipse and github to be a contributor and studying the guidelines.  I would like to request permission to do a pull request and submit these changes for your approval please?

kevinherron, 2019-09-30:
Does your application not have its own bookkeeping that would let you figure out what monitored items belong to what subscriptions?

The SubscriptionExample can also be trivially modified so that the callback also receives the subscription as a parameter. Maybe you can apply the idea to your application:

```java
public class SubscriptionExample implements ClientExample {

    public static void main(String[] args) throws Exception {
        SubscriptionExample example = new SubscriptionExample();

        new ClientExampleRunner(example).run();
    }

    private final Logger logger = LoggerFactory.getLogger(getClass());

    private final AtomicLong clientHandles = new AtomicLong(1L);

    @Override
    public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
        // synchronous connect
        client.connect().get();

        // create a subscription @ 1000ms
        UaSubscription subscription = client.getSubscriptionManager().createSubscription(1000.0).get();

        // subscribe to the Value attribute of the server's CurrentTime node
        ReadValueId readValueId = new ReadValueId(
            Identifiers.Server_ServerStatus_CurrentTime,
            AttributeId.Value.uid(), null, QualifiedName.NULL_VALUE);

        // important: client handle must be unique per item
        UInteger clientHandle = uint(clientHandles.getAndIncrement());

        MonitoringParameters parameters = new MonitoringParameters(
            clientHandle,
            1000.0,     // sampling interval
            null,       // filter, null means use default
            uint(10),   // queue size
            true        // discard oldest
        );

        MonitoredItemCreateRequest request = new MonitoredItemCreateRequest(
            readValueId, MonitoringMode.Reporting, parameters);

        // when creating items in MonitoringMode.Reporting this callback is where each item needs to have its
        // value/event consumer hooked up. The alternative is to create the item in sampling mode, hook up the
        // consumer after the creation call completes, and then change the mode for all items to reporting.
        BiConsumer<UaMonitoredItem, Integer> onItemCreated =
            (item, id) -> item.setValueConsumer(
                (i, value) -> onSubscriptionValue(subscription, i, value)
            );

        List<UaMonitoredItem> items = subscription.createMonitoredItems(
            TimestampsToReturn.Both,
            newArrayList(request),
            onItemCreated
        ).get();

        for (UaMonitoredItem item : items) {
            if (item.getStatusCode().isGood()) {
                logger.info("item created for nodeId={}", item.getReadValueId().getNodeId());
            } else {
                logger.warn(
                    "failed to create item for nodeId={} (status={})",
                    item.getReadValueId().getNodeId(), item.getStatusCode());
            }
        }

        // let the example run for 5 seconds then terminate
        Thread.sleep(5000);
        future.complete(client);
    }

    private void onSubscriptionValue(UaSubscription subscription, UaMonitoredItem item, DataValue value) {
        logger.info(
            "subscription value received: item={}, value={}",
            item.getReadValueId().getNodeId(), value.getValue());
    }

}
```

hogwild65, 2019-09-30:
I do have bookkeeping in place but was having issues at onSubscriptionValue getting the subscriptionId. 
I did not think of your solution and looks like it will work! I will give it a try. Thanks a lot!!

hogwild65, 2019-09-30:
That worked. Much obliged for the help sir.

closed 2019-09-30 by hogwild65