eclipse-milo/milo #556: Unable to access Milo server from remote client

jdhanasekar, 2019-10-09:
Hi Milo,
           I have installed milo server in my ubuntu machine. I am  using ```Prosys OPC UA client``` to connect to Milo server in same machine and it is working.
           Now, I have installed   ```Prosys OPC UA client``` in window machine  and trying to connect to remote ```Milo server``` which is running in ubuntu machine. But I am getting error ```could not connect to the server```.

In wireshark, I am not seeing any activity for ```opcua``` protocol.

For remote access, do i need to do any extra stuff?.
Can you please help me to solve this issue?

kevinherron, 2019-10-09:
If you're using the example server there's no configuration changes needed.

Check that you have the right IP, that you can even ping the server machine from the client machine, that firewalls on either side aren't blocking the connection, etc...

jdhanasekar, 2019-10-10:
Solved the issue. It is firewall issue. Sorry for minor mistake

kevinherron, 2019-10-10:
Glad you got it working 👍

closed 2019-10-10 by kevinherron

jdhanasekar, 2019-10-10:
Thanks for you support....