eclipse-milo/milo #557: Statuscode Bad_AttributeIdInvalid

ramontijsseclaase, 2019-10-14:
Hi Kevin,

I want to read and write booleans in Beckhoff with OPC communication. Your code gives the following result:

milo backing type: class java.lang.Boolean
trying to write: Success(DataValue{value=Variant{value=false}, status=null})
StatusCode{name=Bad_AttributeIdInvalid, value=0x80350000, quality=bad}
OPCUA - Failed to write to node GVL.R4 - probably wrong datatype, should be: NodeId{ns=0, id=1}

Do you know if this error is something in your code? I also downloade UaExpert, and for all variables the RolePermissions, UserRolePermissions and AccessRestrictions are BadAttributeID (0x80350000), so maybe the problem is in the nodes itself. If so, do you know how to fix this?

Thanks in advance.

Kind regards,

Ramon

kevinherron, 2019-10-14:
Can you share the code that is responsible for the write? Are you writing to the Value attribute? Can the value be written to the same node using UaExpert?

ramontijsseclaase, 2019-10-14:
In the following part the datavalue is determined:

    def toDataValue(value: Any, targetType: NodeId): Try[DataValue] = {
    Try {
      val c = BuiltinDataType.getBackingClass(targetType)
      println("milo backing type: " + c.toString)
      c match {
        case q if q == classOf[java.lang.Integer] => new DataValue(new Variant(value.asInstanceOf[Int]))
        case q if q == classOf[UByte] => new DataValue(new Variant(ubyte(value.asInstanceOf[Byte])))
        case q if q == classOf[UShort] => new DataValue(new Variant(ushort(value.asInstanceOf[Short])))
        case q if q == classOf[java.lang.Short] => new DataValue(new Variant(value.asInstanceOf[Short]))
        case q if q == classOf[String] => new DataValue(new Variant(value.asInstanceOf[String]))
        case q if q == classOf[ByteString] => new DataValue(new Variant(ByteString.of(value.asInstanceOf[String].getBytes())))
        case q if q == classOf[java.lang.Boolean] => new DataValue(new Variant(value.asInstanceOf[Boolean]),null,null,null)
        case q if q == classOf[java.lang.Double] => new DataValue(new Variant(value.asInstanceOf[Double]))
        case _ => println(s"need to add type: ${c}"); new DataValue(new Variant(false))
      }
    }
 

Then the code responsible for the write is shown below:

    def write(nodeIdentifier: String, value: Any): Boolean = {
    availableNodes.get(nodeIdentifier) match {
      case Some(n) =>
        val typeid = n.getDataType.get()
        val dv = toDataValue(value, typeid)
        println("trying to write: " + dv)

        dv match {
          case Failure(exception) => throw new Exception(s"could not convert datavalue $dv, throwing exception $exception")
          case Success(value) =>
            val w = client.writeValue(n.getDataType.get,value).get
            println(w)
            if (w.isGood) {
            println("OPCUA - value written")
            true
          }
          else {
            println(s"OPCUA - Failed to write to node ${nodeIdentifier} - probably wrong datatype, should be: " + typeid)
            false
          }
        }
      case None => println(s"OPCUA No such node ${nodeIdentifier}"); false
    }
  

What do you mean with writing to the value attribute? Changing the value of the node in UaExpert is possible.

kevinherron, 2019-10-14:
This looks suspicious:
```
val w = client.writeValue(n.getDataType.get,value).get
```

The `writeValue` method takes the `NodeId` of the Node to write to and the value to write. It looks like you're passing the `NodeId` of the datatype instead.

ramontijsseclaase, 2019-10-14:
Yes that was the problem. Thanks a lot!

closed 2019-10-14 by kevinherron