eclipse-milo/milo #559: Tags in Milo

jdhanasekar, 2019-10-15:
Hi Milo,
           Is ```Tag``` feature supported in Milo project by default?. If it is not, how to enable the Tag feature. Currently, I have not connected PLC devices with OPC UA Server. Is it possible to share empty Tag values to Client?.

Please help me on this.

Thanks,
Dhanasekar

kevinherron, 2019-10-15:
This is an SDK that can be used to build an OPC UA server, not an OPC UA server implementation with drivers for any PLCs.

jdhanasekar, 2019-10-16:
Ok.. I just want to connect to client with Tags. This is for experimental purpose.

In Milo, Tags will available only When drivers for PLCs are present?. Or Milo have dummy Tags ?

kevinherron, 2019-10-16:
Most demo servers, including the milo example server and the one hosted at milo.digitalpetri.com, have some "dummy" nodes in it that clients can read, write, or subscribe to.

You can also download a "real" OPC server from someone like Kepware and use their 2 hour trial to experiment.

closed 2019-10-16 by kevinherron

jdhanasekar, 2019-10-17:
Thanks for the info.

I want to understand the flow of Milo Project. Where do I start from ( I mean file to start) ?.

kevinherron, 2019-10-17:
[ExampleServer](https://github.com/eclipse/milo/blob/master/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/ExampleServer.java) and [ExampleNamespace](https://github.com/eclipse/milo/blob/master/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/ExampleNamespace.java) are a good place to start for the server.

jdhanasekar, 2019-10-17:
Ok..I will go through that..
Thanks for guidance..