eclipse-milo/milo #562: Connecting Milo Client to other server

jdhanasekar, 2019-10-22:
Hi Milo,
                      I want to connect ```Milo Client``` to Other Server and to read data/Tags from that server. Which ```client example``` should I run to connect to other server?
Can you please help me to do this?

Thanks,
Dhanasekar

kevinherron, 2019-10-22:
Try playing with the [Read Example](https://github.com/eclipse/milo/blob/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ReadExample.java)

Modify it by overriding `getEndpointUrl()` so it points to the endpoint URL of some other server. Then you can start changing the `NodeId` being read or reading additional `NodeId`s.

jdhanasekar, 2019-10-22:
Ok..  I am unable to open ```function/method``` declaration in eclipse IDE. It is throwing error ```Problem opening an editor. Reason: milo-master does not exist```. Which IDE I should use for Milo to understand/flow?

kevinherron, 2019-10-22:
I use IntelliJ IDEA. Others have successfully used Eclipse IDE with Milo. I'm not able to provide you support for either, though. You'll need to figure out your tools on your own.

closed 2019-10-23 by kevinherron

jdhanasekar, 2019-10-24:
I am trying to connect ```UaServer``` with Milo client. So, I have modified ```UnifiedAutomationReadCustomDataTypeExample.jave ``` like this,

```
 @Override
    public String getEndpointUrl() {
        // Change this if UaCPPServer is running somewhere other than localhost.
        //return "opc.tcp://localhost:48010";
        return "opc.tcp://XX.1.XX.XX:48020";
    }
```

Am I doing correct thing?.
After making chages, I need to run "ReadExamples.java" which will create jar file.
If run that ```ReadExamples.jar```, It will connect to my server.

Please correct me, if I am wrong.
I am new to OPC UA and JAVA.  Please help me to understand ...

kevinherron, 2019-10-24:
These examples are all meant to be run from the IDE:

[image]

jdhanasekar, 2019-10-24:
Oh.. I got it.. But Milo client is not able to connect to milo server which is running in same system..

```
20:28:07.963 [milo-netty-event-loop-0] WARN  o.e.m.o.stack.server.UaStackServer - Bind failed for endpoint opc.tcp://quanta:12686/milo
java.net.BindException: Address already in use

````

I like to connect to other servers which is running remote system

kevinherron, 2019-10-24:
Each example starts and shuts down a server, so don't try to start it yourself.

If you're going to override `getEndpointUrl()` to connect to another server, though, you can set the `serverRequired` parameter in the `ClientExampleRunner` constructor to `false` and the example won't try to start a server.

jdhanasekar, 2019-10-24:
After making ```serverRequired``` parameter like below,

```
 public ClientExampleRunner(ClientExample clientExample) throws Exception {
        this(clientExample, false);
    }
```

How will client to know about the server which it is going to connect?.
We need to hard code the server URL in any file

kevinherron, 2019-10-24:
You would override `getEndpointUrl()` in each example you were playing with and hard code the server URL.

kevinherron, 2019-10-25:
Yes, it connected and read the State and CurrentTime variables:
```
13:43:38.005 [milo-shared-thread-pool-3] INFO  o.e.milo.examples.client.ReadExample - State=Running
13:43:38.005 [milo-shared-thread-pool-3] INFO  o.e.milo.examples.client.ReadExample - CurrentTime=DateTime{utcTime=132164648177467536, javaDate=Fri Oct 25 13:43:37 IST 2019}
```

jdhanasekar, 2019-10-25:
Thanks for the confirmation....

AshuSingh07, 2022-07-05:
Can i read the values of nodes for certain period of time (example - 1 hr) of anoter server using eclipse milo sdk client.