eclipse-milo/milo #563: Class file for FieldType, structuredType ,

jdhanasekar, 2019-10-24:


jdhanasekar, 2019-10-24:
Hi Milo,

I am using ```Intellij``` IDE  for Milo project. When I press **Build -> BuildProject**, I am getting below error,

```
Information:java: Errors occurred while compiling module 'bsd-parser-core'
Information:javac 1.8.0_222 was used to compile java sources
Information:10/24/19, 6:26 PM - Build completed with 49 errors and 0 warnings in 1 s 935 ms
/home/user/dhanasekar/BMS/OPC/milo/milo-master-intellij/opc-ua-stack/bsd-parser/bsd-parser-core/src/main/java/org/eclipse/milo/opcua/binaryschema/AbstractCodec.java
Error:(48, 44) java: package org.opcfoundation.opcua.binaryschema does not exist
Error:(49, 44) java: package org.opcfoundation.opcua.binaryschema does not exist
Error:(50, 44) java: package org.opcfoundation.opcua.binaryschema does not exist
Error:(167, 31) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(168, 31) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(170, 19) java: cannot find symbol
  symbol:   class StructuredType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(172, 29) java: cannot find symbol
  symbol:   class StructuredType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(305, 9) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(401, 29) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(423, 36) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(447, 68) java: cannot find symbol
  symbol:   class SwitchOperand
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(468, 42) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(197, 25) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(201, 13) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(281, 25) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(285, 13) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(344, 17) java: cannot find symbol
  symbol:   class FieldType
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(440, 13) java: cannot find symbol
  symbol:   class SwitchOperand
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(441, 44) java: cannot find symbol
  symbol:   variable SwitchOperand
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(449, 18) java: cannot find symbol
  symbol:   variable EQUALS
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(451, 18) java: cannot find symbol
  symbol:   variable NOT_EQUAL
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(453, 18) java: cannot find symbol
  symbol:   variable GREATER_THAN
  location: class org.eclipse.milo.opcua.binaryschema.AbstractCodec<StructureT,MemberT>
Error:(455, 18) java: cannot find symbol
[... 48 lines cut ...]
  location: class org.eclipse.milo.opcua.binaryschema.parser.BsdParser
Error:(58, 17) java: cannot find symbol
  symbol:   class StructuredType
  location: class org.eclipse.milo.opcua.binaryschema.parser.BsdParser
Error:(58, 50) java: cannot find symbol
  symbol:   class StructuredType
  location: class org.eclipse.milo.opcua.binaryschema.parser.BsdParser
/home/user/dhanasekar/BMS/OPC/milo/milo-master-intellij/opc-ua-stack/bsd-parser/bsd-parser-core/src/main/java/org/eclipse/milo/opcua/binaryschema/GenericEnumCodec.java
Error:(18, 44) java: package org.opcfoundation.opcua.binaryschema does not exist
Error:(22, 19) java: cannot find symbol
  symbol:   class EnumeratedType
  location: class org.eclipse.milo.opcua.binaryschema.GenericEnumCodec
Error:(24, 29) java: cannot find symbol
  symbol:   class EnumeratedType
  location: class org.eclipse.milo.opcua.binaryschema.GenericEnumCodec
/home/user/dhanasekar/BMS/OPC/milo/milo-master-intellij/opc-ua-stack/bsd-parser/bsd-parser-core/src/main/java/org/eclipse/milo/opcua/binaryschema/GenericStructCodec.java
Error:(20, 44) java: package org.opcfoundation.opcua.binaryschema does not exist
Error:(24, 31) java: cannot find symbol
  symbol:   class StructuredType
  location: class org.eclipse.milo.opcua.binaryschema.GenericStructCodec
```

It seems class for file FiledType, sructuredType and SwitchOperand are missing.

I need to add those class file in Milo Project ??

jdhanasekar, 2019-10-24:
Where can I get ```bsd-parser-core 0.3.4-SNAPSHOT``` jar....

kevinherron, 2019-10-24:
From the command line, at the root of the project, run `mvn clean install`. The missing classes will be generated and the JARs will get installed into your local maven repository.

jdhanasekar, 2019-10-24:
I have given the path of ```bsd-parser-core.jar``` in ```open module``` , issue gone except ```StrucuturedType```..

I have already run ```mvn clean installl```..

closed 2019-10-24 by kevinherron