eclipse-milo/milo #575: UaException: nonce must be at least 32 bytes

andvasp, 2019-11-25:
I cannot connect on KepServerEx using version 3.6.

I try to upgrade from 0.2.5 to 0.3.6 but to 0.3.4 it works perfectly the connection.

Kepware.KEPServerEX.V5:UA Server
Version - V5.8.109.0

I tried using the ClientExample project to do the test.

default SecurityPolicy getSecurityPolicy() {
        return SecurityPolicy.None;
    }

    default IdentityProvider getIdentityProvider() {
        return new AnonymousProvider();
    }

[image]

org.eclipse.milo.opcua.stack.core.UaException: nonce must be at least 32 bytes
	at org.eclipse.milo.opcua.stack.core.util.NonceUtil.validateNonce(NonceUtil.java:176)
	at org.eclipse.milo.opcua.stack.core.util.NonceUtil.validateNonce(NonceUtil.java:162)
	at org.eclipse.milo.opcua.sdk.client.session.SessionFsmFactory.activateSession(SessionFsmFactory.java:878)
	at org.eclipse.milo.opcua.sdk.client.session.SessionFsmFactory.lambda$configureActivatingState$17(SessionFsmFactory.java:345)
	at com.digitalpetri.strictmachine.dsl.ActionBuilder$PredicatedTransitionAction.execute(ActionBuilder.java:76)
	at com.digitalpetri.strictmachine.StrictMachine$PollAndEvaluate.lambda$run$0(StrictMachine.java:207)
	at java.util.ArrayList.forEach(ArrayList.java:1257)
	at com.digitalpetri.strictmachine.StrictMachine$PollAndEvaluate.run(StrictMachine.java:198)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	... 1 common frames omitted

kevinherron, 2019-11-25:
0.3.6 is now validating the nonce length returned in the CreateSessionResponse and ActivateSessionResponse.

It seems this older version of KEPServerEx is returning nonce that is too short. 

Try using security if you aren't already. If that doesn't work, you'll either have to upgrade KSE or downgrade the Milo version. 

According to the spec Milo is doing the correct thing here, so don't expect any changes.

andvasp, 2019-11-25:
Ok kevin!

I suspected this is not a milo problem but I think would be good to report. This is happen with another server too... Industrial Gateway Server (IGS) but this server is based on KepServerEx.

I will do another tests using security. I don't think there is a configuration to disable this. Am I right? It is not interresting to have it configurable?

Thanks in advance.

kevinherron, 2019-11-25:
Because this is explicitly outlined by the spec and nonce validation is security-related it's not desirable to be able to turn this off. 

Sorry about the inconvenience this is causing.

kevinherron, 2019-11-25:
@andvasp can you let me know if this works when security is in use?

I suspect this old version of KSE is returning either a null or zero-length nonce when SecurityPolicy is None, rather than a nonce that's at least 32 bytes in length.

andvasp, 2019-11-25:
I've already debug the code at this part and it is returning zero-length.

I will do tests using security and report them to you.

andvasp, 2019-11-25:
@kevinherron I did the test using security and it works!

From the point of view of security, could/should not skip this validation if SecurityPolicy is None?

Thanks again!

Follow the test result:
Eclipse Milo OPC UA Stack version: 0.3.6
Eclipse Milo OPC UA Client SDK version: 0.3.6
Skipping validation for certificate: CN=KEPServerEX/UA Server, O=Unknown, C=BR, DC=xxxxx
Skipping trust chain verification for certificate: CN=KEPServerEX/UA Server, O=Unknown, C=BR, DC=xxxxx
Skipping validation for certificate: CN=KEPServerEX/UA Server, O=Unknown, C=BR, DC=xxxxx
Skipping trust chain verification for certificate: CN=KEPServerEX/UA Server, O=Unknown, C=BR, DC=xxxxx
value: DateTime{utcTime=132191940930543297, javaDate=Mon Nov 25 20:21:33 BRST 2019}

kevinherron, 2019-11-25:
The nonce must be non-empty. There is a not-yet-public CVE regarding nonce validation. It is a serial problem if the client allows a defective server to send an empty nonce and it turns out that nonce is used in the encryption of a password in the subsequent ActivateSessionRequest.

closed 2019-11-25 by kevinherron

andvasp, 2019-11-25:
Understood @kevinherron.

Thank you for your support!

coreteam-co, 2019-12-16:
@kevinherron We are facing the same problem with Siemens WinCC. Do you have a pointer to the mentioned CVE? Or is it still non-public? This would help me to convince people to perform an upgrade. Thanks a lot in advance.

kevinherron, 2019-12-16:
@coreteam-co it’s not public yet. Is your connection to WinCC using a username/password or anonymous?

It seems there’s a few cases where I can relax the nonce validation, which I’m going to do for a 0.3.7 release soon here, but I can’t say for sure it will help your case without more details.

coreteam-co, 2019-12-16:
@kevinherron Thanks for your quick reply. In this case the connection is anonymous. It would be great if the 0.3.7 release would help in this this case.

kevinherron, 2019-12-16:
@coreteam-co can you try [this branch](https://github.com/eclipse/milo/tree/relaxed-nonce-validation) and let me know if it works for you?

coreteam-co, 2019-12-16:
@kevinherron I ran a quick connectivity test with this branch and it worked perfectly. Thanks a lot for looking into this so quickly. Would be great if this change could be part of the next release.

kevinherron, 2019-12-16:
@coreteam-co @andvasp tracking the relaxed validations in #579. Should be able to do a 0.3.7 release relatively soon, but need to test more and check back in with the OPC UA security WG and make sure we're all on the same page about the relaxed nonce validations.