eclipse-milo/milo #577: Copying a node from other Server...

jdhanasekar, 2019-11-29:
Hi Milo,

I have created  two dummy  ```variableNodeClass``` node and it is as named ```flip``` and ```flop``` and it is  based on codes in  ```ExampleNameSpace.java```.   I have added code like this,

```
// Adding code for flip
        UaVariableNode flip = UaVariableNode.builder(getNodeContext())
                .setNodeId(newNodeId("ObjectTypes/MyObjectType.Flip"))
                .setAccessLevel(ubyte(AccessLevel.getMask(AccessLevel.READ_WRITE)))
                .setBrowseName(newQualifiedName("Flip"))
                .setDisplayName(LocalizedText.english("Flip"))
                .setDataType(Identifiers.Boolean)
                .setTypeDefinition(Identifiers.BaseDataVariableType)
                .build();

        flip.addReference(new Reference(
                flip.getNodeId(),
                Identifiers.HasModellingRule,
                Identifiers.ModellingRule_Mandatory.expanded(),
                true
        ));

        flip.setValue(new DataValue(new Variant(0)));
        objectTypeNode.addComponent(flip);

        // Adding code for flop

        UaVariableNode flop = UaVariableNode.builder(getNodeContext())
                .setNodeId(newNodeId("ObjectTypes/MyObjectType.Flop"))
                .setAccessLevel(ubyte(AccessLevel.getMask(AccessLevel.READ_WRITE)))
                .setBrowseName(newQualifiedName("Flop"))
                .setDisplayName(LocalizedText.english("Flop"))
                .setDataType(Identifiers.Boolean)
                .setTypeDefinition(Identifiers.BaseDataVariableType)
                .build();

        flop.addReference(new Reference(
                flop.getNodeId(),
                Identifiers.HasModellingRule,
                Identifiers.ModellingRule_Mandatory.expanded(),
                true
        ));

        flop.setValue(new DataValue(new Variant(0)));
        objectTypeNode.addComponent(flop);
```

I want to copy all the details of ```flip``` and ``` flop``` node  from other server and put those values into ```flip``` and ```flop``` dummy node of Milo Server which already created. 

Can you please help me to implement above task?.  
Please share me steps or algorithm for copying nodes from one server to milo- server  and vice versa..

kevinherron, 2019-11-29:
At a high level:

1) create a local Node that represents each Node in the remote server you want to "mirror".
2) subscribe to value changes for each Node on the remote server.
3) when the value change notification arrives, update the value of local Node in your server.

This kind of question is not appropriate for the Issue tracker. You can try the mailing list and hope that somebody is willing to hold your hand in implementing your application. I can provide support with how to use specific aspects of the library but I cannot help you build an application that you don't already have an idea of how to build.

closed 2019-11-29 by kevinherron

jdhanasekar, 2019-12-03:
Thanks for the info and your help. I will ask my doubts to milo-dev mail group.