eclipse-milo/milo #578: Username Provider Used Connection Failed validate Nonce ISSUE

bsm1981, 2019-12-05:
stack-core-0.3.6 used issue

An issue occurs when connecting as shown below.

OpcUaClientConfigBuilder cfg = new OpcUaClientConfigBuilder();
cfg.setEndpoint(endpoints.get(0)) // please do better, and not only pick the first entry
.setIdentityProvider(new UsernameProvider("username", "password"))
.setRequestTimeout(uint(60000));

The occurrence part was confirmed in stack-core-0.3.6 version, and bs.length is set to 0 among validateNonce methods in NonceUtil.java file.
The portion that occurs less than minimumLength. If bs.length> 0 is set temporarily, it will work normally.

kevinherron, 2019-12-05:
I'm not sure I understand you. Are you having an issue connecting to another server with 0.3.6? Do you have more information about this server or any logs?

bsm1981, 2019-12-06:
It's Test Log .

11:21:53.935 [main] INFO  o.e.m.e.client.ClientExampleRunner - security temp dir: C:\Users\desc\AppData\Local\Temp\security
11:21:53.936 [main] INFO  o.e.m.examples.client.KeyStoreLoader - Loading KeyStore at C:\Users\desc\AppData\Local\Temp\security\example-client.pfx
<b>validateNonce bs.length < minimumLength:::::::::1/0</b>
11:21:54.158 [NonceUtilSecureRandom] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
11:21:54.172 [main] INFO  o.e.m.e.client.ClientExampleRunner - Using endpoint: opc.tcp://192.168.0.109:4840 [None/None]
11:21:54.178 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Stack version: 0.3.7-SNAPSHOT
11:21:54.178 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: 0.3.7-SNAPSHOT
<b>validateNonce bs.length < minimumLength:::::::::1/0</b>
<b>validateNonce bs.length < minimumLength:::::::::0/32</b>
<b>validateNonce bs.length < minimumLength:::::::::0/32</b>

kevinherron, 2019-12-06:
Sorry, I still don't understand what you're trying to tell me.

bsm1981, 2019-12-06:
When connecting to a Connection using a UsernameProvider, this is the part that is not connected because an exception occurs in the validateNonce method of NonceUtil.java as above. This part seems to be related to the part checked again after bs.length <minimumLength.

kevinherron, 2019-12-06:
Yes, that's supposed to happen now. It's not allowed for a server to use an empty nonce, even when the SecurityPolicy is None.

bsm1981, 2019-12-06:
Temporarily change the method as shown below to ensure normal operation

public static void validateNonce(ByteString nonce, int minimumLength) throws UaException {
        byte[] bs = nonce.bytesOrEmpty();
       
        //System.out.println("validateNonce bs.length < minimumLength:::::::::"+ bs.length +"/"+ minimumLength);
        
        if (bs.length > 0 && bs.length < minimumLength) {
            throw new UaException(
                StatusCodes.Bad_NonceInvalid,
                "nonce must be at least " + minimumLength + " bytes"
            );
        }
    }

kevinherron, 2019-12-16:
[This branch](https://github.com/eclipse/milo/tree/relaxed-nonce-validation) may help you, if the connection to this server is no security policy on both the channel and the user token policy.

kevinherron, 2019-12-16:
Should be fixed by #579 unless it's a scenario where the nonce must be valid.

closed 2019-12-16 by kevinherron