eclipse-milo/milo #582: HistoryData request date time zone

BadEvil1, 2019-12-20:
Hi,
I have faced the issue while reading history data. I got time with different time zones from a subscription and from a history data request. I have UTC dates on my server and my current time zone is GMT+3.
So, server:
[image]
I get 19:00 time value in GMT+3 from subscription, everything is still fine.
However, when I invoke the history data method:
[image]
I get 16:00 in Moscow time zone (+3) instead of 16:00 UTC (or 19:00 in GMT+3 time zone)
Here is my code:

    public Optional<List<NodeValue>> getHistoryData(NodeId nodeId,
                                                    ZonedDateTime from, ZonedDateTime to) {
        VariableNode node = client.getAddressSpace().createVariableNode(this.createMiloNodeId(nodeId));
        try {
            EnumSet<AccessLevel> accessLevels = AccessLevel.fromMask(node.getAccessLevel().get());
            if (!accessLevels.contains(AccessLevel.HistoryRead)) {
                logger.warn("Node=" + nodeId + "does not support HistoryRead");
            }
        } catch (InterruptedException | ExecutionException e) {
            throw new OpcuaException("Can not get access level of node ns=" + nodeId.getNamespaceIndex() + " id=" +
                    nodeId, e);
        }

        HistoryReadDetails historyReadDetails = new ReadRawModifiedDetails(
                false,
                new DateTime(from.toInstant()),
                new DateTime(to.toInstant()),
                uint(0),
                true
        );

        HistoryReadValueId historyReadValueId = new HistoryReadValueId(
                this.createMiloNodeId(nodeId),
                null,
                QualifiedName.NULL_VALUE,
                ByteString.NULL_VALUE
        );

        List<HistoryReadValueId> nodesToRead = new ArrayList<>();
        nodesToRead.add(historyReadValueId);

        HistoryReadResponse historyReadResponse = null;
        try {
            historyReadResponse = client.historyRead(
                    historyReadDetails,
                    TimestampsToReturn.Both,
                    false,
                    nodesToRead
            ).get();
        } catch (InterruptedException | ExecutionException e) {
            throw new OpcuaException("Error while requesting history data from server", e);
        }

        HistoryReadResult[] historyReadResults = historyReadResponse.getResults();
        if (historyReadResults != null) {
            HistoryReadResult historyReadResult = historyReadResults[0];
            HistoryData historyData = (HistoryData) historyReadResult.getHistoryData()
                    .decode(client.getSerializationContext());
            return Optional.of(l(historyData.getDataValues()).stream()
                    .map(dataValue -> this.createNodeValueFromMilo(node, dataValue))
                    .collect(Collectors.toList()));
        }

        return Optional.empty();
    }

kevinherron, 2019-12-21:
I can only imagine this is an issue with the server. OPC UA does not deal with time zones, everything is in UTC.

kevinherron, 2019-12-21:
Actually I’m not sure I see an issue at all. Both your screenshots show a Java date (which does apply your local TZ) as “Fri Dec 20 16:00:00 MSK 2019”. You seem to be underlining different things in your screenshot.

BadEvil1, 2019-12-21:
I mean my first screenshot shows 16:00 in UTC on the server, but the second one shows 16:00 in +3 time zone. For the same time on the server (16:00 UTC), subscription returns me 19:00 in +3 time zone. 
So I got time shift from subscription and didnt get it from history data request

kevinherron, 2019-12-21:
You're not comparing the same kinds of object. On the left you have a Java Date, and on the right a Java Instant. Look above the underlined values on the righthand side and you'll see a matching Java Date.

BadEvil1, 2019-12-21:
Yeap, on the left – java date 16:00 GMT+3. On the right — 13:00 java instant. And u can see on the top screenshot that the server node contains 16:00 UTC time. That is the issue. I supposed to get 16:00 instant or 19:00 time in UTC like I get it from subscription

kevinherron, 2019-12-21:
These are the same time:

[image]

and `2019-12-20T13:00:00Z` is UTC and equal to `Fri Dec 2016:00:00 MSK 2019`.

So I still don't understand what your issue is.

BadEvil1, 2019-12-21:
Let me rephrase the question. 
This screenshot shows UTC node time:
[image]
From the history request I get this time:
[image]
16:00 UTC and 16:00 MSK is different time, isnt it?

kevinherron, 2019-12-21:
Yes 16:00 MSK and 16:00 UTC are different. But your previous screenshot does not show 16:00 UTC anywhere.

BadEvil1, 2019-12-21:
After my local timezone been applied I suppose to get 19:00 time but I get 16:00 instead

BadEvil1, 2019-12-21:
I tried to reproduce the issue on your public demo server, unfortunately, it doesnt support history read operation.
I discovered another public server with history requests support. I got correct time values for both subscription values and history values. So, I think the problem is not on the Milo library side.
Thanks for your time, Kevin. Sorry for the inconvenience.

closed 2019-12-21 by BadEvil1

kevinherron, 2019-12-22:
Ok, glad you figured that out.

That was my first suspicion but then for some reason I thought you were using Milo server SDK as well.