eclipse-milo/milo #583: Bad_SequenceNumberInvalid Exception

gopalakrishnan-subramani, 2019-12-25:
We have observed below error with Milo.

JRE 1.8
Milo Version: 0.3.0
OS: Windows 10 64 bit.

There are two exceptions Bad_TcpMessageTooLarge and Bad_SequenceNumberInvalid

UaException: status=Bad_SequenceNumberInvalid, message=expected sequence number 7364 but received 7365
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder$AbstractDecoder.decode(ChunkDecoder.java:170)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:84)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:72)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerSymmetricHandler.lambda$onSecureMessage$0(UascServerSymmetricHandler.java:144)
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:61)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
2019-12-24 16:32:40 ERROR UascServerSymmetricHandler:147 - Error decoding symmetric message: expected sequence number 7365 but received 7366
UaException: status=Bad_SequenceNumberInvalid, message=expected sequence number 7365 but received 7366
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder$AbstractDecoder.decode(ChunkDecoder.java:170)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:84)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:72)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerSymmetricHandler.lambda$onSecureMessage$0(UascServerSymmetricHandler.java:144)
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:61)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
2019-12-24 18:02:41 ERROR UascServerSymmetricHandler:147 - Error decoding symmetric message: expected sequence number 674565 but received 674566
UaException: status=Bad_SequenceNumberInvalid, message=expected sequence number 674565 but received 674566
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder$AbstractDecoder.decode(ChunkDecoder.java:170)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:84)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:72)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerSymmetricHandler.lambda$onSecureMessage$0(UascServerSymmetricHandler.java:144)
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:61)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$InlineTask.run(ExecutionQueue.java:154)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:136)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
2019-12-24 18:02:41 ERROR UascServerSymmetricHandler:147 - Error decoding symmetric message: expected sequence number 674566 but received 674567
UaException: status=Bad_SequenceNumberInvalid, message=expected sequence number 674566 but received 674567
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder$AbstractDecoder.decode(ChunkDecoder.java:170)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:84)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:72)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerSymmetricHandler.lambda$onSecureMessage$0(UascServerSymmetricHandler.java:144)
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:61)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$InlineTask.run(ExecutionQueue.java:154)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:136)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
2019-12-24 18:02:41 ERROR UascServerSymmetricHandler:147 - Error decoding symmetric message: expected sequence number 674566 but received 674568
UaException: status=Bad_SequenceNumberInvalid, message=expected sequence number 674566 but received 674568
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder$AbstractDecoder.decode(ChunkDecoder.java:170)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:84)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:72)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerSymmetricHandler.lambda$onSecureMessage$0(UascServerSymmetricHandler.java:144)
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:61)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
2019-12-24 18:02:41 ERROR UascServerSymmetricHandler:147 - Error decoding symmetric message: expected sequence number 674566 but received 674569
UaException: status=Bad_SequenceNumberInvalid, message=expected sequence number 674566 but received 674569
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder$AbstractDecoder.decode(ChunkDecoder.java:170)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:84)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:72)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerSymmetricHandler.lambda$onSecureMessage$0(UascServerSymmetricHandler.java:144)
[... 120 lines cut ...]
2019-12-25 11:18:28 ERROR UascServerHelloHandler:223 - [remote=/10.68.87.228:56678] Exception caught; sent ErrorMessage{error=StatusCode{name=Bad_TcpMessageTooLarge, value=0x80800000, quality=bad}, reason=max message length exceeded (50529027 > 4128)}
io.netty.handler.codec.DecoderException: UaException: status=Bad_TcpMessageTooLarge, message=max message length exceeded (50529027 > 4128)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:472)
	at io.netty.handler.codec.ByteToMessageDecoder.channelInputClosed(ByteToMessageDecoder.java:405)
	at io.netty.handler.codec.ByteToMessageDecoder.channelInputClosed(ByteToMessageDecoder.java:372)
	at io.netty.handler.codec.ByteToMessageDecoder.channelInactive(ByteToMessageDecoder.java:355)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:242)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:228)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelInactive(AbstractChannelHandlerContext.java:221)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelInactive(DefaultChannelPipeline.java:1403)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:242)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:228)
	at io.netty.channel.DefaultChannelPipeline.fireChannelInactive(DefaultChannelPipeline.java:912)
	at io.netty.channel.AbstractChannel$AbstractUnsafe$8.run(AbstractChannel.java:827)
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:163)
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:404)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:495)
	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:905)
	at java.lang.Thread.run(Unknown Source)
Caused by: UaException: status=Bad_TcpMessageTooLarge, message=max message length exceeded (50529027 > 4128)
	at org.eclipse.milo.opcua.stack.core.channel.headers.HeaderDecoder.getMessageLength(HeaderDecoder.java:39)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerHelloHandler.decode(UascServerHelloHandler.java:102)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:502)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:441)
	... 16 more

kevinherron, 2019-12-25:
Sounds like https://github.com/eclipse/milo/issues/483, which was fixed in https://github.com/eclipse/milo/commit/63c09c13eb7c2368c6b7287b992d05dbb1f420b6.

Upgrade to latest and if it still happens you’ll have to capture it happening with Wireshark as well so we can verify if it’s legit or a bug in the server.

gopalakrishnan-subramani, 2019-12-31:
@kevinherron 

Upgraded, we didn't find any specific errors there after, I am closing the thread, will re-open if we observe the issues

closed 2019-12-31 by gopalakrishnan-subramani