eclipse-milo/milo #584: Automatic-Module-Name needs to be unique for each maven module

comtel2000, 2020-01-06:
based on #442 please choose a dedicated automatic module name for each maven module (stack-client, stack-core, sdk-client, etc.).

current 'stack-core' MANIFEST.MF:

```
..
Automatic-Module-Name: org.eclipse.milo.opcua
Bundle-License: https://www.eclipse.org/org/documents/epl-2.0/EPL-2.0.
 html
Bundle-SymbolicName: org.eclipse.milo.stack-core
Implementation-Version: 0.3.6
..
```
needs to be changed into:
```
..
Automatic-Module-Name: org.eclipse.milo.stack.core
..
```

Just like the Bundle-SymbolicName just with '.' instead of '-'

comtel2000, 2020-01-06:
or changed to the root package name (stack-core):
```
..
Automatic-Module-Name: org.eclipse.milo.opcua.stack.core
..
```

kevinherron, 2020-01-06:
I think I'll just use the root package name. Should be able to do this and release it with 0.3.7 this week.

comtel2000, 2020-01-06:
Thanks, that's really quick :)

commit 8ddf3767 (2020-01-07): Add automatic module names

fixes #584
closed 2020-01-07 by kevinherron