eclipse-milo/milo #589: AttributeObserver lost?

stud0709, 2020-01-14:
Hello Kevin, 

there is something I don't understand: in sdk-server 0.2.5 the following is added while constructing the Namespace: 

```
sendControl.addAttributeObserver(new AttributeObserver() {
	@Override
	public void attributeChanged(UaNode node, AttributeId attributeId, Object value) {
		if (attributeId == AttributeId.Value) {
			logger.debug("{}: value has been set to {}", node.getNodeId(), value);
		}
	}
});

System.out.println("this will work");
sendControl.setValue(new DataValue(new Variant(UByte.valueOf(1)))); 

new Thread(() -> {
	try {
		Thread.sleep(30000);
	} catch (InterruptedException e) {
		// TODO Auto-generated catch block
		e.printStackTrace();
	}
	System.out.println("this will NOT work !!!");
	sendControl.setValue(new DataValue(new Variant(UByte.valueOf(0)))); 
}).start();

```

The log output works immediately afterwards as shown, but I don't see this output when changing the value from the program logic somewhat later (see the example within Thread).  

What should I do to make it work?

Thank you!

Regards,

stud0709

stud0709, 2020-01-14:
ok, I got it. See class UaNode:

```
public synchronized void addAttributeObserver(AttributeObserver observer) {
        if (observers == null) {
            observers = new LinkedList<>();
        }

        observers.add(new WeakReference<>(observer));
}
```

Due to WeakReference one should not declare AttributeObserver in-line, otherwise it will be garbage-collected on the next occasion.

closed 2020-01-14 by stud0709

kevinherron, 2020-01-14:
@stud0709 this was changed in the `dev/0.4` branch already (and there's a new mechanism for intercepting attributes called `AttributeFilterChain`/`AttributeFilter`, but `AttributeObserver` has not been removed yet).