eclipse-milo/milo #590: Issues on opcua subscription

NibeshHub, 2020-01-16:
We found Bad_AttributeIdInvalid while subscribe for the monitored items. These below exception coming on subscribe for  nodeIds 

        CompletableFuture<List<UaMonitoredItem>> feature = subscription.createMonitoredItems(
                TimestampsToReturn.Both,
                monitoredItemCreateRequests,
                onItemCreated
        );

        log.info("Getting MonitoredRefs for " + sub.task.name);

        List<UaMonitoredItem> items = feature.get();

 (status={})NodeId{ns=2, id=Channel2.Citect.K.Something} : StatusCode{name=Bad_AttributeIdInvalid, value=0x80350000, quality=bad}
2020-01-13 15:42:04 ERROR HDAServerActor:143 - failed to create item for subscription nodeId={} (status={})NodeId{ns=2, id=Channel2.Citect.K.test2} : StatusCode{name=Bad_AttributeIdInvalid, value=0x80350000, quality=bad}

Some errors coming like :

 ERROR UascServerSymmetricHandler:147 - Error decoding symmetric message: expected sequence number 2074748 but received 2074749
UaException: status=Bad_SequenceNumberInvalid, message=expected sequence number 2074748 but received 2074749
at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder$AbstractDecoder.decode(ChunkDecoder.java:170)
at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:84)
at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:72)
at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerSymmetricHandler.lambda$onSecureMessage$0(UascServerSymmetricHandler.java:144)
at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:61)
at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
at java.lang.Thread.run(Unknown Source)

kevinherron, 2020-01-16:
What do your `monitoredItemCreateRequests` look like? Just getting `Bad_AttributeIdInvalid` isn't necessarily a problem.

As for the sequence number issue - what version of Milo are you using? It's possible this is a bug in Milo and also possible it's a bug with the server.

NibeshHub, 2020-01-17:
Milo version 0.3.6

kevinherron, 2020-01-17:
Ok, without more information about how you create the requests I can’t say if there is anything wrong.

The sequence number error will require a Wireshark capture to see if it’s a problem with the Milo client or the server you’re connected to.

gopalakrishnan-subramani, 2020-01-17:
@kevinherron 

What happen when Milo client is used in multi threaded environment, where as,

1.  Thread T1 send request R1 to server, server start processing R1, assume some IO operation, it take about 5 seconds to complete
2. Before serve respond to client request R1, client send another request R2 through Thread T2 
3. where as R2 request was processed before R1 with 1 second, while R1 still in process
4. Now Milo Server respond to R2 request first while R1 is still not yet completed, 

There are chances that Request/Response Seq number sync issue. This may be easy to simulate, write a opc method, put sleep there, and call  another OPC method which will execute faster.

kevinherron, 2020-01-17:
This error is about the per-chunk sequence number received from the server being out of order. There are protections for multi-threaded environments in place already.

closed 2020-01-21 by kevinherron