eclipse-milo/milo #600: ExampleNamespace: CustomDataType does not reveal its internal structure

stud0709, 2020-02-03:
Hello Kevin, 

if I start ExampleServer, there is a sample of a custom data type. But its internal structure is not shown, only its encoded data is available.

[image]

What can I do to make its internal structure visible? Here is what I need to do:

[image]

Here, a Line consists of m Machines, a Machine consists of n Win... - Elements and each Win...-Element hast its scalar data. 

Thank you!

BR,

stud0709

kevinherron, 2020-02-03:
There's 2 issues you're seeing here.

1. In 0.3.x custom structures are only readable by clients with advanced knowledge of those structures, unless you use a code modeling tool that also generates the dictionary and then manage to import/create all the nodes represented in the NodeSet2 XML.

In the `dev/0.4` branch this situation has changed and [you can now describe a custom structure and the SDK will generate the dictionary and all the other nodes for you](https://github.com/eclipse/milo/blob/2afe5a42b785ed43a6fa4ce7374ff727ef4d8a2c/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/ExampleNamespace.java#L826-L875), but..

2. Having Nodes in the Address Space that mimic the structure would need to be done entirely yourself, there's no tools or APIs to do that.

stud0709, 2020-02-03:
I see... that's beyond my skill level :)

Anyway, thank you for the fast reply!

BR,

stud0709

closed 2020-02-03 by stud0709