eclipse-milo/milo #601: How can I get the NodeManger from the Opc

0DennisLee0, 2020-02-19:


0DennisLee0, 2020-02-19:
I found the new way to set up the Server's Namespace
`
ExampleNamespace exampleNamespace = new ExampleNamespace(server);
        exampleNamespace.startup();
`
is that mean, if I have another namespace to register.
I need to de
`
ExampleNamespace exampleNamespace = new ExampleNamespace(server);
        exampleNamespace.startup();
AnotherNamespace anoterNamespace = new AnotherNamespace(server);
      anoterNamespace.startup();
`
Can I use this way to start a new namespace?

kevinherron, 2020-02-19:
Yes, that's fine for registering and starting a new Namespace instance.

I'm not sure the code in the `pi-server` project is written against a recent version of Milo, so best to stick to the `ExampleServer` and `ExampleNamespace` classes as a reference.

0DennisLee0, 2020-02-19:
> Yes, that's fine for registering and starting a new Namespace instance.
> 
> I'm not sure the code in the `pi-server` project is written against a recent version of Milo, so best to stick to the `ExampleServer` and `ExampleNamespace` classes as a reference.

Thank you very much! It's so kind of you!

closed 2020-02-19 by 0DennisLee0