eclipse-milo/milo #603: RegisterServer Service in Milo

jdhanasekar, 2020-02-21:
Hi Milo,

I want to register Milo Server to Other LDS server. 
Is  ```RegisterServer2``` service supported ?.

How to call RegisterServer service and register Milo server in LDS ?.

Please advise me.

Thanks,
Dhanasekar

kevinherron, 2020-02-21:
Sure, just open a client connection to the server and then use `sendRequest` to send a `RegisterServer2Request` request.

closed 2020-02-24 by kevinherron

jdhanasekar, 2020-02-25:
Hi Kevin,

Can you please advise me for setting values for ```ExtensionObject[] discoveryConfiguration```  while Registering Server2 Request?.
I am facing difficulty.

kevinherron, 2020-02-25:
Check out the RegistrationHelper class in that out of date PR:
https://github.com/eclipse/milo/blob/b3c8fd290c1f76865bb721a57273c95a72bb0cfe/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/util/RegistrationHelper.java#L120-L148

I'd have to turn to the OPC UA spec to know any more details about how
registration works.

jdhanasekar, 2020-02-26:
Hi Kevin,

I have added code inside ```ReadExample.java``` like this,

```
package org.eclipse.milo.examples.client;

import java.util.List;
import java.util.concurrent.CompletableFuture;

import com.google.common.collect.ImmutableList;
import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.api.nodes.VariableNode;
import org.eclipse.milo.opcua.stack.client.UaStackClient;
import org.eclipse.milo.opcua.stack.core.Identifiers;
import org.eclipse.milo.opcua.stack.core.types.builtin.DataValue;
import org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject;
import org.eclipse.milo.opcua.stack.core.types.builtin.LocalizedText;
import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ServerState;
import org.eclipse.milo.opcua.stack.core.types.enumerated.TimestampsToReturn;
import org.eclipse.milo.opcua.stack.core.types.structured.*;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import static ch.qos.logback.core.joran.action.ActionConst.NULL;
import static org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.Unsigned.uint;

public class ReadExample implements ClientExample {

    public static void main(String[] args) throws Exception {

        ReadExample example = new ReadExample();

        new ClientExampleRunner(example, true).run();
    }

    private final Logger logger = LoggerFactory.getLogger(getClass());

    @Override
    public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {

        //UaStackClient myClient = new UaStackClient();

        String ServerURI ="urn:eclipse:milo:examples:client";
        String ProductURI = "https://opcfoundation.org/";
        LocalizedText[] myServerName = new LocalizedText[]{LocalizedText.english("eclipse milo opc-ua client")};
        ApplicationType ServerType =  ApplicationType.Server;
        String myGateWayURI = NULL;
        String[] myDiscoveryURLs =  new String[]{ "opc.tcp://10.1.61.38:12686/milo/discovery"};
        String mySemaphorePath = NULL;
        Boolean IsOnline = Boolean.TRUE;

        MdnsDiscoveryConfiguration[] myDiscoveryConfig = new  MdnsDiscoveryConfiguration[]{null,null} ;

        //ExtensionObject[] discoveryConfig = new ExtensionObject[] {ExtensionObject.BodyType("Eclipse_Milo_OPCUA_Winodws","NA")};

        UaStackClient stackClient = client.getStackClient();

        RequestHeader requestHeader = stackClient.newRequestHeader(
                NodeId.NULL_VALUE,
                stackClient.getConfig().getRequestTimeout()
        );
        RegisteredServer myServer = new RegisteredServer(ServerURI,ProductURI,myServerName,ServerType,NULL,myDiscoveryURLs,NULL,IsOnline);

        RegisterServerRequest myServerRequest = new RegisterServerRequest (requestHeader,myServer);

        stackClient.sendRequest(myServerRequest);

        //RegisterServer2Request myServer2Request = new RegisterServer2Request(requestHeader,myServer,discoveryConfig);

        // synchronous connect
        client.connect().get();

        // synchronous read request via VariableNode
        VariableNode node = client.getAddressSpace().createVariableNode(Identifiers.Server_ServerStatus_StartTime);
        DataValue value = node.readValue().get();

        logger.info("StartTime={}", value.getValue().getValue());

        // asynchronous read request
        readServerStateAndTime(client).thenAccept(values -> {
            DataValue v0 = values.get(0);
            DataValue v1 = values.get(1);

            logger.info("State={}", ServerState.from((Integer) v0.getValue().getValue()));
            logger.info("CurrentTime={}", v1.getValue().getValue());

            future.complete(client);
        });
    }

    private CompletableFuture<List<DataValue>> readServerStateAndTime(OpcUaClient client) {
        List<NodeId> nodeIds = ImmutableList.of(
            Identifiers.Server_ServerStatus_State,
            Identifiers.Server_ServerStatus_CurrentTime);

        return client.readValues(0.0, TimestampsToReturn.Both, nodeIds);
    }

}
```

Is it correct place to added ?
Please advise me....

kevinherron, 2020-02-26:
"correct place" for what? Just to see if it works in the context of this example? If nothing else you should call `connect()` before you try to send the request.

jdhanasekar, 2020-03-02:
I am getting this error in ```response```. 

[image]

jdhanasekar, 2020-03-03:
I have added ```RegistrationHelper.java``` at this location ``` milo-master\opc-ua-sdk\sdk-client\src\main\java\org\eclipse\milo\opcua\sdk\client\util\RegistrationHelper.java```.

After adding the code, I am seeing lot of error because of  ```UaTcpStackClient``` is missing. What is the equivalent for it. 

[image]

kevinherron, 2020-03-03:
`org.eclipse.milo.opcua.stack.client.UaStackClient`

jdhanasekar, 2020-03-04:
I am getting this error in ```RegistrationHelper.java```,

[image]

It is because ```encode()``` function in ```ExtensionObject.java```  does not have one argument. But you are calling with one parameter.   All encode () function has more than one argument in ExtensionObject.java.