eclipse-milo/milo #604: GetEndpoints matching to wrong endpoint

locka99, 2020-02-23:
I'm the author of OPC UA for Rust and I have a reported [issue ](https://github.com/locka99/opcua/issues/45) which I suspect is in milo.

Basically a milo example client wanted to create a session on "opc.tcp://localhost:4855/", called GetEndpoints on a OPC UA for Rust server to get a list of endpoints and then opened a session to the first matching None security profile on a "opc.tcp://localhost:4855/noaccess" when there was an exact matching "opc.tcp://localhost:4855/" later it should have chosen.

kevinherron, 2020-02-23:
@locka99 Thanks for taking the time to report this, but I don’t think it’s an issue in the SDK itself. The client examples in the repo do just choose the first matching endpoint but that’s just because they’re simple examples meant to run against the example server.

Ultimately it’s up to the application using the SDK to call GetEndpoints and choose one by whatever means is appropriate. I can’t tell what application the reporter in the issue you linked to is using - it doesn’t look like they’re just running one of the examples.

kevinherron, 2020-02-23:
I just realized the configuration they were posting in that issue is for your SDK/server, not the client application.

So yes, if they are trying to use one of the examples, it’s just because of the naive endpoint selection used by the examples.

locka99, 2020-02-23:
Yes I modified the sample to point at my server, changing the path from /milo to just /. This was to replicate what the bug reporter was doing, and in so doing discovered the milo client was matching to /noaccess which was a disabled endpoint declared before an endpoint / where both had None security profiles. 

I was just wondering if the code should match the profile and the path.

We can both close our respective issues if we think there is nothing to fix.

kevinherron, 2020-02-24:
Yeah I don’t think there’s anything to fix here. The client examples can already be modified to select a different endpoint by overriding `endpointFilter()`.

closed 2020-02-24 by kevinherron

evanjs, 2020-02-24:
@kevinherron Thanks!
This helps clear up the issues we were having quite a bit.