eclipse-milo/milo #605: How to decoding the data to be subscribed

0DennisLee0, 2020-02-24:
Hello kevin,
I want to subcribe a node which is encoded.Such as the CustomDataTypeNode in the example.And the codec is
```java
public static class Codec extends GenericDataTypeCodec<CustomStructType> {
        @Override
        public Class<CustomStructType> getType() {
            return CustomStructType.class;
        }

        @Override
        public CustomStructType decode(
            SerializationContext context,
            UaDecoder decoder) throws UaSerializationException {

            String foo = decoder.readString("Foo");
            UInteger bar = decoder.readUInt32("Bar");
            boolean baz = decoder.readBoolean("Baz");

            return new CustomStructType(foo, bar, baz);
        }

        @Override
        public void encode(
            SerializationContext context,
            UaEncoder encoder, CustomStructType value) throws UaSerializationException {

            encoder.writeString("Foo", value.foo);
            encoder.writeUInt32("Bar", value.bar);
            encoder.writeBoolean("Baz", value.baz);
        }
    }
```
And the decoding step I have already know
```java
Variant variant = value.getValue();
        ExtensionObject xo = (ExtensionObject) variant.getValue();

        CustomStructType decoded = (CustomStructType) xo.decode(
            client.getSerializationContext()
        );
```

However, if I want to subscribe another node, I have to repeat in this way to match the data decoding.Is there any easier way to implement this?Or is the there any generic decoding utils in milo?

kevinherron, 2020-02-24:
If the server you're connecting to has defined all the custom types in its DataTypeDictionary and they are properly exposed in the address space then yes, the client will automatically read those and you can access them generically.

By default the client is configured with a `GenericBsdParser` instance that reads custom types as a `org.eclipse.milo.opcua.binaryschema.Struct` instance. There's also a `JsonBsdParser` you can configure that will read them as a GSON `JsonObject`.

0DennisLee0, 2020-02-25:
> If the server you're connecting to has defined all the custom types in its DataTypeDictionary and they are properly exposed in the address space then yes, the client will automatically read those and you can access them generically.
> 
> By default the client is configured with a `GenericBsdParser` instance that reads custom types as a `org.eclipse.milo.opcua.binaryschema.Struct` instance. There's also a `JsonBsdParser` you can configure that will read them as a GSON `JsonObject`.
> 
> This won't work against the Milo example server (at least not in 0.3.x... should work in `dev/0.4` branch), but it generally works with demo servers from other vendors.

3Q，I will work on it then

kevinherron, 2020-02-25:
Take a look at `UnifiedAutomationReadCustomDataTypeExample`.

0DennisLee0, 2020-02-26:
> Take a look at `UnifiedAutomationReadCustomDataTypeExample`.

OK，and I want to know if it works well on the open62541 server,too?

kevinherron, 2020-02-26:
It should work with open62541 or any other server as long as they put the information about the datatype into the address space.

closed 2020-02-28 by kevinherron