eclipse-milo/milo #607: Unable to connect to kepserverEX 6 with neither securityPolicy or securityMode enabled.

tanuja5, 2020-02-28:
I am using 0.2.5 version of stack-client, stack-core and sdk-client. While i am able to connect to the Kepserver via OPC quick client, 
 i am unable to connect to the server in my code, when trying to get endpoints:
_EndpointDescription[] endpoints = UaTcpStackClient.getEndpoints(endpointURL).get();_

I get the following exception:

`2020-02-28 00:05:03,223 @S192_168_1_7 @opcuaa -WARN ua-shared-pool-5 org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run (ExecutionQueue.java:109) Uncaught Throwable during execution.
java.lang.NullPointerException
	at com.test.proc.reader.dataChanged(Reader.java:154)
	at com.test.proc.opcua.lib.NodeReader.onSubscription(NodeReader.java:197)
	at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaMonitoredItem.lambda$setValueConsumer$1(OpcUaMonitoredItem.java:138)
	at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaMonitoredItem.onValueArrived(OpcUaMonitoredItem.java:191)
	at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaSubscriptionManager.lambda$null$39(OpcUaSubscriptionManager.java:700)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:107)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)`

When i run the code in debug mode, i get the following stack trace:

`2020-02-28 00:05:01,074 @S192_168_1_7 @opcuaa -DEBUG Start-opcuaa org.eclipse.milo.opcua.stack.client.ClientChannelManager.connect (ClientChannelManager.java:67) connect() while NotConnected
java.lang.Exception
	at org.eclipse.milo.opcua.stack.client.ClientChannelManager.connect(ClientChannelManager.java:67)
	at org.eclipse.milo.opcua.stack.client.UaTcpStackClient.connect(UaTcpStackClient.java:127)
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.connect(OpcUaClient.java:312)
	at com.test.proc.reader.opcua.Utils.getOpcuaClient(Utilss.java:133)
	at com.test.proc.reader.opcua.Property.getClient(Propertey.java:332)
	at com.test.proc.reader.init(Reader.java:75)
	at com.webaction.runtime.components.Source.start(Source.java:344)
	at com.webaction.runtime.components.Flow.startSources(Flow.java:593)
	at com.webaction.runtime.components.Flow$2.run(Flow.java:1688)
	at java.lang.Thread.run(Unknown Source)`

I tried to clone the repo and used the stack-client with increased timeout from a default of 5000 to 60000ms. But it did not resolve the issue.

kevinherron, 2020-02-28:
Sorry, I'll have to see some of your code or something. The error you posted is not happening in Milo, it's happening in you code:

java.lang.NullPointerException at com.test.proc.reader.dataChanged(Reader.java:154)

tanuja5, 2020-03-02:
Code:

`public class TestConnection {

static OpcUaClient client = null;
public static void main(String[] args) throws Exception{
String endpointURL = "opc.tcp://192.168.0.92:49320";
String hostname = "WIN-9GHDO5MENJL";

client =getOpcuaClient(endpointURL, SecurityPolicy.valueOf(SecurityPolicy.None.toString()), MessageSecurityMode.valueOf(MessageSecurityMode.None.toString()), "TestApplication", "urn:eclipse:milo:examples:client",
            null, new AnonymousProvider(), 60000,
            null,
            null);
    NodeId parse;
    try{
        parse= NodeId.parse("ns=2;s=Simulation Examples.Functions.Random1");
    }
    catch(Exception ex){
        throw new IllegalArgumentException("Error parsing nodeId "+" a typical nodeId looks like ns=1;s=Sensor please refer docs for more details");
    }
    OpcuaUtils.getAllVariableNode(client, parse);
}
public static OpcUaClient getOpcuaClient(String endpointURL, SecurityPolicy policy,MessageSecurityMode securityMode, String applicationName,
String appuri, String certificatePath, IdentityProvider provider, long readTimeOut,char[] keystorePassword,String clientAlias) throws Exception {
if (logger.isDebugEnabled()) logger.debug("Obtaining endpoints from " + endpointURL);

	EndpointDescription[] endpoints = UaTcpStackClient.getEndpoints(endpointURL).get();
    if (logger.isDebugEnabled()) logger.debug("Obtained " + endpoints.length + " endpoints from " + endpointURL);

    if (endpoints == null) new AdapterException("No endpoints returned from " + endpointURL);
    
	EndpointDescription endpoint = null;

	for (EndpointDescription ep : endpoints) {
        if (logger.isDebugEnabled()) logger.debug("Checking endpoint = " + ep);

        if (endpoint == null &&
                ep.getSecurityPolicyUri().equals(policy.getSecurityPolicyUri()) &&
                ep.getSecurityMode().toString().equals(securityMode.toString())) {
            endpoint = ep;
            if (logger.isDebugEnabled()) logger.debug("Endpoint = " + ep + " matches security policy " + ep.getSecurityPolicyUri() + " and security mode " + securityMode);
        }
	}

	/*URI uri = new URI(endpoint.getEndpointUrl()).parseServerAuthority();

	String endpointUrl = String.format(
			"%s://%s:%s%s",
			uri.getScheme(),
			"WIN-9GHDO5MENJL",
			uri.getPort(),
			uri.getPath()
	);

	endpoint = new EndpointDescription(
			endpointUrl,
			endpoint.getServer(),
			endpoint.getServerCertificate(),
			endpoint.getSecurityMode(),
			endpoint.getSecurityPolicyUri(),
			endpoint.getUserIdentityTokens(),
			endpoint.getTransportProfileUri(),
			endpoint.getSecurityLevel()
	);
*/
	if (endpoint == null) throw new AdapterException("No desired endpoints returned, only " + Arrays.toString(endpoints));
	
    if (logger.isDebugEnabled()) logger.debug("Building client config for app name = " + LocalizedText.english(applicationName) + " app uri = " + appuri + " endpoint = " + endpoint);
	OpcUaClientConfigBuilder configBuilder = new OpcUaClientConfigBuilder();
	configBuilder.setApplicationName(LocalizedText.english(applicationName));
[... 48 lines cut ...]
Environment: KepServer running on Windows VM of the same same machine. I am able to connect to the Prosys client in my mac. 

I also tried the commented code 
`/*URI uri = new URI(endpoint.getEndpointUrl()).parseServerAuthority();

	String endpointUrl = String.format(
			"%s://%s:%s%s",
			uri.getScheme(),
			"WIN-9GHDO5MENJL",
			uri.getPort(),
			uri.getPath()
	);

	endpoint = new EndpointDescription(
			endpointUrl,
			endpoint.getServer(),
			endpoint.getServerCertificate(),
			endpoint.getSecurityMode(),
			endpoint.getSecurityPolicyUri(),
			endpoint.getUserIdentityTokens(),
			endpoint.getTransportProfileUri(),
			endpoint.getSecurityLevel()
	);
*/`
as suggested in : https://stackoverflow.com/questions/40554198/java-opc-ua-client-eclipse-milo-endpoint-url-changes-to-localhost

kevinherron, 2020-03-02:
Try using the IP address instead of hostname. The error you're getting is from Java:

> java.nio.channels.UnresolvedAddressException

tanuja5, 2020-03-05:
Thanks @kevinherron , I removed the hostname from server endpoint and that exception was resolved.

closed 2020-03-05 by kevinherron