eclipse-milo/milo #609: Error with MdnsDiscoveryConfiguration in ExtensionObject

jdhanasekar, 2020-03-07:
Hi Milo,

I am working on register Milo server with LDS server. I am using ```  RegisterServer2Request``` service for registering. 

```
RegisterServer2Request(RequestHeader requestHeader, RegisteredServer server, ExtensionObject[] discoveryConfiguration)
 ```

For discoveryConfiguration parameter,  I have  made like this,
```
MdnsDiscoveryConfiguration mymDNS = new MdnsDiscoveryConfiguration("LDS-quanta._opcua-tcp._tcp.local.",new String[] {"NA"}); 

```
If try to encode the mdns discovery congiuration, like this,
```
 ExtensionObject[] myDisConfig =  {ExtensionObject.encode(mymDNS)};
```
I am getting this error,

```
 Cannot resolve method 'encode(org.eclipse.milo.opcua.stack.core.types.structured.MdnsDiscoveryConfiguration)'
```

I am seeing ExtensionObject java file is not containing method to encode discoveryconfiguration. 

Please help me to solve this issue.
Please advice me, if I made mistake. 

-Dhanasekar

kevinherron, 2020-03-07:
You want call the `ExtensionObject#encode(SerializationContext, UaStructure)` overload.

A `SerializationContext` can be obtained from `UaStackClient` or `OpcUaClient` via `getSerializationContext()`.

closed 2020-03-07 by kevinherron

jdhanasekar, 2020-03-09:
I would like to configure my mdnsConfiguration , so that I declared like this,
```
MdnsDiscoveryConfiguration mymDNS = new MdnsDiscoveryConfiguration("LDS-quanta._opcua-tcp._tcp.local.",new String[] {"NA"}); 
```

But you have suggested this ```UaStackClient``` or ```OpcUaClient```,  but this is not taking ```mdnsConfiguration``` as parameter.

How can I achieve in sending mdnscofiguration (to LDS) using ```UaStackClient```?.

Providing other way to setup mdnsconfiguration is highly appreciated.

jdhanasekar, 2020-03-11:
Hi Kevin,

As per your suggestion, I have created ExtensionObject like this,

```

MdnsDiscoveryConfiguration mymDNS = new MdnsDiscoveryConfiguration("LDS-quanta._opcua-tcp._tcp.local.",new String[] {"NA"});
SerializationContext mySerial =  stackClient.getSerializationContext();
ExtensionObject[] myDisConfig =  {ExtensionObject.encode(mySerial,mymDNS)};
````

When I run , I am getting ``` Bad_ServiceUnsupported``` error,

```
17:47:12.067 [milo-shared-thread-pool-2] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 5
17:47:12.112 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error running client example: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1908)
	at org.eclipse.milo.examples.client.ReadExample.run(ReadExample.java:99)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:177)
	at org.eclipse.milo.examples.client.ReadExample.main(ReadExample.java:42)
Caused by: org.eclipse.milo.opcua.stack.core.UaServiceFaultException: status=Bad_ServiceUnsupported, description=The server does not support the requested service.
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:279)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
17:47:12.113 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run() 8
17:47:12.113 [ForkJoinPool.commonPool-worker-25] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run() 3
17:47:12.113 [ForkJoinPool.commonPool-worker-25] ERROR o.e.m.e.client.ClientExampleRunner - Error running example: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requesteExtensionObject[] myDisConfig =  {ExtensionObject.encode(mySerial,mymDNS)};ExtensionObject[] myDisConfig =  {ExtensionObject.encode(mySerial,mymDNS)};ExtensionObject[] myDisConfig =  {ExtensionObject.encode(mySerial,mymDNS)};d service.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1908)
	at org.eclipse.milo.examples.client.ReadExample.run(ReadExample.java:99)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:177)
	at org.eclipse.milo.examples.client.ReadExample.main(ReadExample.java:42)
Caused by: org.eclipse.milo.opcua.stack.core.UaServiceFaultException: status=Bad_ServiceUnsupported, description=The server does not support the requested service.
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:279)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent17:47:12.067 [milo-shared-thread-pool-2] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 5
17:47:12.112 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error running client example: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1908)
	at org.eclipse.milo.examples.client.ReadExample.run(ReadExample.java:99)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:177)
	at org.eclipse.milo.examples.client.ReadExample.main(ReadExample.java:42)
Caused by: org.eclipse.milo.opcua.stack.core.UaServiceFaultException: status=Bad_ServiceUnsupported, description=The server does not support the requested service.
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:279)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
17:47:12.113 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run() 8
17:47:12.113 [ForkJoinPool.commonPool-worker-25] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run() 3
17:47:12.113 [ForkJoinPool.commonPool-worker-25] ERROR o.e.m.e.client.ClientExampleRunner - Error running example: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
[... cut ...]
17:47:12.113 [ForkJoinPool.commonPool-worker-25] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run() 4

Process finished with exit code 0
.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
17:47:12.113 [ForkJoinPool.commonPool-worker-25] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run() 4

Process finished with exit code 0

```

Please help  me to solve this issue.

kevinherron, 2020-03-11:
This StatusCode comes from the server. It seems the server you're connected to does not support this service.

jdhanasekar, 2020-03-11:
Thanks for the info.

I am setting below values to access the server. 
Please correct me, If I have programmed wrong values

```
       String ServerURI ="urn:eclipse:milo:examples:client";
        String ProductURI = "https://opcfoundation.org/";
        LocalizedText[] myServerName = new LocalizedText[]{LocalizedText.english("eclipse milo opc-ua client")};
        ApplicationType ServerType =  ApplicationType.Server;
        String myGateWayURI = NULL;        
        String[] myDiscoveryURLs =  new String[]{ "opc.tcp://localhost.localdomain:12686/milo/discovery"};
        String mySemaphorePath = NULL;
        Boolean IsOnline = Boolean.TRUE;

        UaStackClient stackClient = client.getStackClient();

        MdnsDiscoveryConfiguration mymDNS = new MdnsDiscoveryConfiguration("LDS-quanta._opcua-tcp._tcp.local.",new String[] {"NA"});

```

jdhanasekar, 2020-03-12:
Hi Kevin,

```1 )```As I am implementing LDS registration in ```ReadExample.java```.  There  ```client.connect().get();``` will connects to the client whose URL in programmed in ```ClientExample.java``` like this,

```
 default String getEndpointUrl() {    
        return "opc.tcp://localhost:12686/milo"; 
    }
```

If I uses this ```client.connect().get();```  before ```RegisterServer2Request``` register call, 
```
RegisterServerResponse  rsr = (RegisterServerResponse)stackClient.sendRequest(myServerRequest).get();
```
program will try to register in ```opc.tcp://localhost:12686/milo``` instead of  open62541's  LDS ``` opc.tcp://localhost:4840/```.  So that , I am getting ``` Bad_ServiceUnsupported```

Am I correct ?.
To avoid this, I need to create separate ```ClientConfig``` in separate file. 

```2```, If I mention same URL for LDS server and Milo client , like this,
```
default String getEndpointUrl() {        
        return  "opc.tcp://localhost:4840/";        
    }
 ```
I am getting this error, 
```
ClientExampleRunner - Error running client example: UaServiceFaultException: status=Bad_SessionIdInvalid, message=The session id is not valid.
```
Please advise me.

kevinherron, 2020-03-12:
You don't need a separate ClientConfig file, you just override the `getEndpointUrl` method in the example you are modifying.

I don't know why the server returns Bad_SessionIdInvalid, though. Maybe you can capture the interaction with Wireshark for diagnosis.

jdhanasekar, 2020-03-12:
Thanks for the info. I will update you with debug info....

jdhanasekar, 2020-03-13:
I am unable to extract data from Wireshark. 

Following steps, I have followed to enable Wireshark,

1.In Wireshar, ```Edit -> Preference-> Protocols->OpcUa-> TCP port(s) = 4840```
[image]
FYI, I have tried port ```12686``` also.

2. In Filters, I have mentioned ```opcua```
[image]

Do I need to disable any security setting in Milo?

jdhanasekar, 2020-03-13:
Hi Kevin,

I have added override for getendpointurl() in ```ReadExample.java```, like this

```
  @Override
   public String getEndpointUrl() {       
        return  "opc.tcp://localhost:4840/";       
    }
```

I have added breakpoints in IDE,  debugged further and got this error,

```
java.util.concurrent.CompletionException: UaException: status=Bad_ConnectionRejected, message=io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: localhost/127.0.0.1:4840

````

jdhanasekar, 2020-03-18:
Hi Kevin,

I have override the ```getEndpointUrl()``` method in ```ReadExample.java```, like this,
```
 @Override
   public String getEndpointUrl() {
        return "opc.tcp://10.1.60.19:4840";
    }
```
After running, I am getting Bad_ConnectionRejected error. 

```
19:27:48.454 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java ClientExampleRunner constructor: 
19:27:48.456 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run() 1
19:27:48.456 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run()  & Calling createClient() 2
19:27:48.456 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.jafutureva Inside createClient(): 
19:27:48.458 [main] INFO  o.e.m.e.client.ClientExampleRunner - security temp dir: /tmp/security
19:27:48.463 [main] INFO  o.e.m.examples.client.KeyStoreLoader - Loading KeyStore at /tmp/security/example-client.pfx
19:27:49.750 [main] INFO  o.e.m.o.stack.client.UaStackClient - dhanasekar @UaStackClient.java transport=org.eclipse.milo.opcua.stack.client.transport.tcp.OpcTcpTransport@3c41ed1d
19:27:49.927 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java discoveryUrl: opc.tcp://10.1.60.19:4840
19:27:49.927 [main] INFO  o.e.m.e.client.ClientExampleRunner - Trying explicit discovery URL: opc.tcp://10.1.60.19:4840/discovery
19:27:49.929 [main] INFO  o.e.m.o.stack.client.UaStackClient - dhanasekar @UaStackClient.java transport=org.eclipse.milo.opcua.stack.client.transport.tcp.OpcTcpTransport@2f4948e4
19:27:49.935 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error getting client: UaException: status=Bad_ConnectionRejected, message=io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: /10.1.60.19:4840
java.util.concurrent.ExecutionException: UaException: status=Bad_ConnectionRejected, message=io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: /10.1.60.19:4840
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
```
I have attached my ```ReadExample.java```file.
Correct me,If I made any mistake.

```
/*
 * Copyright (c) 2019 the Eclipse Milo Authors
 *
 * This program and the accompanying materials are made
 * available under the terms of the Eclipse Public License 2.0
 * which is available at https://www.eclipse.org/legal/epl-2.0/
 *
 * SPDX-License-Identifier: EPL-2.0
 */

package org.eclipse.milo.examples.client;

import java.util.List;
import java.util.concurrent.CompletableFuture;

import com.google.common.collect.ImmutableList;
import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.api.nodes.VariableNode;
import org.eclipse.milo.opcua.stack.client.UaStackClient;
import org.eclipse.milo.opcua.stack.core.Identifiers;
import org.eclipse.milo.opcua.stack.core.Stack;
import org.eclipse.milo.opcua.stack.core.serialization.SerializationContext;
import org.eclipse.milo.opcua.stack.core.serialization.UaStructure;
import org.eclipse.milo.opcua.stack.core.types.builtin.DataValue;
import org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject;
import org.eclipse.milo.opcua.stack.core.types.builtin.LocalizedText;
import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ServerState;
import org.eclipse.milo.opcua.stack.core.types.enumerated.TimestampsToReturn;
import org.eclipse.milo.opcua.stack.core.types.structured.*;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import static ch.qos.logback.core.joran.action.ActionConst.NULL;

public class ReadExample implements ClientExample {

    public static void main(String[] args) throws Exception {
        ReadExample example = new ReadExample();
[... 99 lines cut ...]
        Logger logger = LoggerFactory.getLogger(ReadExample.class);
        logger.info("dhanasekar @ReadExample nodeIds : {}", nodeIds);

        //dhanasekar - This call defaults to reading the “value” attribute

        return client.readValues(0.0, TimestampsToReturn.Both, nodeIds);
    }

    @Override
   public String getEndpointUrl() {
        //return "opc.tcp://10.1.61.147:4840";//38:48020";
        //return "opc.tcp://localhost:12686/milo";
        //return "opc.tcp://10.1.60.189:4840";
        //return "opc.tcp://10.1.61.38:49320";
        //return  "opc.tcp://quanta:4840/";
        //return  "opc.tcp://localhost:4840/";
        //return "opc.tcp://localhost.localdomain:12686/milo";
        //return NULL;
        return "opc.tcp://10.1.60.19:4840";
        //return "opc.tcp://10.1.60.19:12686/milo";
    }

}

```

kevinherron, 2020-03-18:
It doesn’t look like you’re doing anything wrong. You just can’t connect to a server at that IP/port for some reason. Are you sure it’s running and that the IP/port are correct? If it’s remote are there firewalls stopping the connection?

jdhanasekar, 2020-03-19:
Actually, ```Milo Server``` and ```open62541 LDS``` servers are running in same host..

I have programmed ```ClientExample.java``` , like this,

```
public interface ClientExample {

    default String getEndpointUrl() {
        return "opc.tcp://10.1.60.19:12686/milo";        
    }

```
After re running, LDS server, I am getting below error```(Bad_SessionIdInvalid)```,

```
18:01:17.705 [main] INFO  o.e.m.o.stack.client.UaStackClient - dhanasekar @UaStackClient.java transport=org.eclipse.milo.opcua.stack.client.transport.tcp.OpcTcpTransport@3c41ed1d
18:01:17.939 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java endpoints: [EndpointDescription{EndpointUrl=opc.tcp://10.1.60.19:4840, Server=ApplicationDescription{ApplicationUri=urn:open62541.server.application, ProductUri=http://open62541.org, ApplicationName=LocalizedText{text=open62541-based OPC UA Application, locale=en}, ApplicationType=Server, GatewayServerUri=null, DiscoveryProfileUri=null, DiscoveryUrls=[opc.tcp://10.1.60.19:4840]}, ServerCertificate=ByteString{bytes=null}, SecurityMode=None, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None, UserIdentityTokens=[UserTokenPolicy{PolicyId=open62541-anonymous-policy, TokenType=Anonymous, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=null}, UserTokenPolicy{PolicyId=open62541-username-policy, TokenType=UserName, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None}], TransportProfileUri=http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary, SecurityLevel=1}]
18:01:17.942 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java clientExample.getEndpointUrl().get(): opc.tcp://10.1.60.19:4840
18:01:17.952 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java endpoint: EndpointDescription{EndpointUrl=opc.tcp://10.1.60.19:4840, Server=ApplicationDescription{ApplicationUri=urn:open62541.server.application, ProductUri=http://open62541.org, ApplicationName=LocalizedText{text=open62541-based OPC UA Application, locale=en}, ApplicationType=Server, GatewayServerUri=null, DiscoveryProfileUri=null, DiscoveryUrls=[opc.tcp://10.1.60.19:4840]}, ServerCertificate=ByteString{bytes=null}, SecurityMode=None, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None, UserIdentityTokens=[UserTokenPolicy{PolicyId=open62541-anonymous-policy, TokenType=Anonymous, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=null}, UserTokenPolicy{PolicyId=open62541-username-policy, TokenType=UserName, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None}], TransportProfileUri=http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary, SecurityLevel=1}
18:01:17.952 [main] INFO  o.e.m.e.client.ClientExampleRunner - Using endpoint: opc.tcp://10.1.60.19:4840 [None/None]
18:01:17.953 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar ClientExampleRunner.java clientExample.getIdentityProvider(): AnonymousProvider{}
18:01:17.961 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Stack version: 0.3.4-SNAPSHOT
18:01:17.961 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: 0.3.4-SNAPSHOT
18:01:17.962 [main] INFO  o.e.m.o.stack.client.UaStackClient - dhanasekar @UaStackClient.java transport=org.eclipse.milo.opcua.stack.client.transport.tcp.OpcTcpTransport@62ddbd7e
18:01:17.982 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 1
18:01:18.001 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 6
18:01:18.005 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() addressSpace=org.eclipse.milo.opcua.sdk.client.DefaultAddressSpace@11fc564b 
18:01:18.175 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run()  and calling back run in ReadExample.java 6
18:01:18.177 [main] INFO  o.e.milo.examples.client.ReadExample - Dhanasekar @ReadExample.java  stackClient.getSerializationContext() =org.eclipse.milo.opcua.stack.client.UaStackClient$1@1af2d44a
18:01:18.189 [NonceUtilSecureRandom] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
18:01:18.210 [milo-shared-thread-pool-0] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 2
18:01:18.213 [milo-shared-thread-pool-0] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 3
18:01:18.236 [milo-shared-thread-pool-0] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 5
18:01:18.256 [main] INFO  o.e.milo.examples.client.ReadExample - Dhanasekar @ReadExample.java  client.connect().get() =org.eclipse.milo.opcua.sdk.client.OpcUaClient@543588e6
18:01:18.290 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error running client example: UaServiceFaultException: status=Bad_SessionIdInvalid, message=The session id is not valid.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_SessionIdInvalid, message=The session id is not valid

```

Image of Milo server , 

[image]

Image of open62541 LDS Server, 

[image]

kevinherron, 2020-03-19:
You're sending the register request without a session auth token. Connect first, then build the header like this:

```
       RequestHeader requestHeader = stackClient.newRequestHeader(
            client.getSession().get().getAuthenticationToken()
        );
```

I should probably add some "helper" methods for Discovery services to `OpcUaClient` like the other service sets have.

jdhanasekar, 2020-03-20:
Thanks for providing info. 

After adding ```AuthenticationToken``` and I have  solved Bad_SessionIdInvalid error. 
Now  I am getting ```Bad_SempahoreFileMissing``` error. 
To fix this issue, I have to create a Semaphore file ?. If so, Please provide me some example semaphore files.

kevinherron, 2020-03-20:
I have no idea why you're getting this error. I've never seen it before and it did not happen when I ran your registration code against the open62541 LDS example server.

jdhanasekar, 2020-03-23:
Thanks for your effort for helping me to solve this issue.  
I have attached the error log,
```
17:30:16.767 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java ClientExampleRunner constructor: 
17:30:16.768 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run() 1
17:30:16.769 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run()  & Calling createClient() 2
17:30:16.769 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.jafutureva Inside createClient(): 
17:30:16.770 [main] INFO  o.e.m.e.client.ClientExampleRunner - security temp dir: /tmp/security
17:30:16.774 [main] INFO  o.e.m.examples.client.KeyStoreLoader - Loading KeyStore at /tmp/security/example-client.pfx
17:30:17.932 [main] INFO  o.e.m.o.stack.client.UaStackClient - dhanasekar @UaStackClient.java transport=org.eclipse.milo.opcua.stack.client.transport.tcp.OpcTcpTransport@3c41ed1d
17:30:18.175 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java endpoints: [EndpointDescription{EndpointUrl=opc.tcp://10.1.60.19:4840, Server=ApplicationDescription{ApplicationUri=urn:open62541.server.application, ProductUri=http://open62541.org, ApplicationName=LocalizedText{text=open62541-based OPC UA Application, locale=en}, ApplicationType=Server, GatewayServerUri=null, DiscoveryProfileUri=null, DiscoveryUrls=[opc.tcp://10.1.60.19:4840]}, ServerCertificate=ByteString{bytes=null}, SecurityMode=None, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None, UserIdentityTokens=[UserTokenPolicy{PolicyId=open62541-anonymous-policy, TokenType=Anonymous, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=null}, UserTokenPolicy{PolicyId=open62541-username-policy, TokenType=UserName, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None}], TransportProfileUri=http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary, SecurityLevel=1}]
17:30:18.177 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java clientExample.getEndpointUrl().get(): opc.tcp://10.1.60.19:4840
17:30:18.187 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java endpoint: EndpointDescription{EndpointUrl=opc.tcp://10.1.60.19:4840, Server=ApplicationDescription{ApplicationUri=urn:open62541.server.application, ProductUri=http://open62541.org, ApplicationName=LocalizedText{text=open62541-based OPC UA Application, locale=en}, ApplicationType=Server, GatewayServerUri=null, DiscoveryProfileUri=null, DiscoveryUrls=[opc.tcp://10.1.60.19:4840]}, ServerCertificate=ByteString{bytes=null}, SecurityMode=None, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None, UserIdentityTokens=[UserTokenPolicy{PolicyId=open62541-anonymous-policy, TokenType=Anonymous, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=null}, UserTokenPolicy{PolicyId=open62541-username-policy, TokenType=UserName, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None}], TransportProfileUri=http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary, SecurityLevel=1}
17:30:18.187 [main] INFO  o.e.m.e.client.ClientExampleRunner - Using endpoint: opc.tcp://10.1.60.19:4840 [None/None]
17:30:18.188 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar ClientExampleRunner.java clientExample.getIdentityProvider(): AnonymousProvider{}
17:30:18.195 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Stack version: 0.3.4-SNAPSHOT
17:30:18.195 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: 0.3.4-SNAPSHOT
17:30:18.196 [main] INFO  o.e.m.o.stack.client.UaStackClient - dhanasekar @UaStackClient.java transport=org.eclipse.milo.opcua.stack.client.transport.tcp.OpcTcpTransport@62ddbd7e
17:30:18.216 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 1
17:30:18.234 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 6
17:30:18.239 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() addressSpace=org.eclipse.milo.opcua.sdk.client.DefaultAddressSpace@11fc564b 
17:30:18.403 [main] INFO  o.e.m.e.client.ClientExampleRunner - dhanasekar @ClientExampleRunner.java inside run()  and calling back run in ReadExample.java 6
17:30:18.404 [main] INFO  o.e.milo.examples.client.ReadExample - Dhanasekar @ReadExample.java  stackClient.getSerializationContext() =org.eclipse.milo.opcua.stack.client.UaStackClient$1@1af2d44a
17:30:18.415 [NonceUtilSecureRandom] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
17:30:18.434 [milo-shared-thread-pool-2] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 2
17:30:18.437 [milo-shared-thread-pool-2] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 3
17:30:18.457 [milo-shared-thread-pool-2] INFO  o.e.m.opcua.sdk.client.OpcUaClient - dhaasekar OpcUaClient() 5
17:30:18.469 [main] INFO  o.e.milo.examples.client.ReadExample - Dhanasekar @ReadExample.java  getAuthenticationToken() =NodeId{ns=1, id=786089bd-8fbd-ecdb-97f9-bf8bca5c2532}
17:30:18.472 [main] INFO  o.e.milo.examples.client.ReadExample - Dhanasekar @ReadExample.java  client.connect().get() =org.eclipse.milo.opcua.sdk.client.OpcUaClient@f5acb9d
[... cut ...]
        client.connect().get();  //Added by dhanasekar
        logger.info("Dhanasekar @ReadExample.java  client.connect().get() ={}", client.connect().get());

        //RegisterServerResponse  rsr = (RegisterServerResponse)stackClient.sendRequest(myServerRequest).get();

        RegisterServer2Response  rsr = (RegisterServer2Response)stackClient.sendRequest(myServer2Request).get();

        logger.info("Dhanasekar @ReadExample.java  rsr.getResponseHeader().getServiceResult() ={}", rsr);
   
```

jdhanasekar, 2020-03-28:
Hi Kevin,

Please share me the steps you have followed to Register (with my code)  the Milo server with open62541.
I will be follow the same and try to isolate the error. 

What value you have assigned for ```semaphoreFilePath``` in ```RegisteredServer```?.
Please provide the details.

jdhanasekar, 2020-03-28:
After making change like this  ```String mySemaphorePath = ""; ```  for setting ```semaphoreFilePath```, I am not seeing any error when I run ```ReadExample.java```.

Below, output I have observed from open62451's LDS log, 
```
[2020-03-28 18:15:18.317 (UTC+0550)] info/network	Connection 6 | New connection over TCP from 10.1.60.19
[2020-03-28 18:15:18.320 (UTC+0550)] info/channel	Creating a new SecureChannel
[2020-03-28 18:15:18.326 (UTC+0550)] info/channel	Connection 6 | SecureChannel 14 | Opened SecureChannel
[2020-03-28 18:15:18.336 (UTC+0550)] info/channel	Connection 6 | SecureChannel 14 | Session f0e11da3-9c1a-8494-90f9-9fe9c52c5205 created
[2020-03-28 18:15:18.348 (UTC+0550)] info/session	Connection 6 | SecureChannel 14 | Session ns=1;g=f0e11da3-9c1a-8494-90f9-9fe9c52c5205 | ActivateSession: Session activated
[2020-03-28 18:15:18.415 (UTC+0550)] info/server	Multicast DNS: add record for domain: opc.tcp://localhost:4840-localhost._opcua-tcp._tcp.local.
[2020-03-28 18:15:18.455 (UTC+0550)] info/session	Connection 6 | SecureChannel 14 | Session ns=1;g=f0e11da3-9c1a-8494-90f9-9fe9c52c5205 | CloseSession
[2020-03-28 18:15:18.457 (UTC+0550)] info/channel	Connection 6 | SecureChannel 14 | CloseSecureChannel
[2020-03-28 18:15:18.457 (UTC+0550)] info/network	Connection 6 | Closed
```
[image]

Have I registered Milo server correctly with ```open62541 LDS Server``` ?
Please provide your comments.

kevinherron, 2020-03-28:
It looks like you've successfully sent a registration request and received
a response. I don't know enough about multicast discovery to know if you've
sent all the correct values in your request or not. Maybe you should take a
look at the OPC UA spec, part 12, and learn about this stuff for yourself.

jdhanasekar, 2020-03-28:
OK. I will look at the OPC UA Spec, Part 12. 
If anything, I will update you. 

Thanks for reply.

jdhanasekar, 2020-03-30:
Hi Kevin,

After registering Milo server with open62541 LDS server, I am working on to call ``` FindServersOnNetwork()``` from other system (In my case, I am calling from windows system) to get the details of registered servers. 

In ```ReadExample.java```, I have added code like this, 

```
 UaStackClient stackClient = client.getStackClient();
        String CapFilter[] = new String[]{""};
        UInteger RecordId = UInteger.valueOf(0);
        UInteger RecordsToReturn = UInteger.valueOf(0);

        RequestHeader myRequestHeader = stackClient.newRequestHeader(
                client.getSession().get().getAuthenticationToken()
        );

        FindServersOnNetworkRequest myFSONRequest = new FindServersOnNetworkRequest(myRequestHeader,RecordId, RecordsToReturn,CapFilter);
        client.connect().get();
        FindServersOnNetworkResponse res = (FindServersOnNetworkResponse)stackClient.sendRequest(myFSONRequest).get();

        logger.info("dhanasekar @ReadExample.java FindServerOnNetwork result ={}", res);
```

After running the ReadExample.java, I am getting below error,

```
18:54:43.531 [main] INFO  o.e.m.examples.server.ExampleServer - security temp dir: C:\Users\DHANAS~1.J\AppData\Local\Temp\security
18:54:43.545 [main] INFO  o.e.m.examples.server.KeyStoreLoader - Loading KeyStore at C:\Users\DHANAS~1.J\AppData\Local\Temp\security\example-server.pfx
18:54:43.870 [main] INFO  o.e.m.examples.server.ExampleServer - pki dir: C:\Users\DHANAS~1.J\AppData\Local\Temp\security\pki
18:54:44.919 [main] INFO  o.e.m.opcua.sdk.server.OpcUaServer - Eclipse Milo OPC UA Stack version: dev
18:54:44.919 [main] INFO  o.e.m.opcua.sdk.server.OpcUaServer - Eclipse Milo OPC UA Server SDK version: dev
18:54:46.688 [main] INFO  o.e.m.o.s.s.n.OpcUaNamespace - Loaded nodes in 848ms.
18:54:46.733 [main] INFO  o.e.m.o.s.s.n.ServerNamespace - dhanasekar  inside ManagedAddressSpace.java
18:54:46.928 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://BLRB1DT166:12686/milo to 0.0.0.0:12686 [None/None]
18:54:47.271 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://BLRB1DT166:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
18:54:47.271 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://BLRB1DT166:12686/milo/discovery to 0.0.0.0:12686 [None/None]
18:54:47.272 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://10.1.61.38:12686/milo to 0.0.0.0:12686 [None/None]
18:54:47.272 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://10.1.61.38:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
18:54:47.272 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://10.1.61.38:12686/milo/discovery to 0.0.0.0:12686 [None/None]
18:54:47.272 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://BLRB1DT166.VTPPL.COM:12686/milo to 0.0.0.0:12686 [None/None]
18:54:47.272 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://BLRB1DT166.VTPPL.COM:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
18:54:47.272 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://BLRB1DT166.VTPPL.COM:12686/milo/discovery to 0.0.0.0:12686 [None/None]
18:54:47.272 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://127.0.0.1:12686/milo to 0.0.0.0:12686 [None/None]
18:54:47.272 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://127.0.0.1:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
18:54:47.272 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://127.0.0.1:12686/milo/discovery to 0.0.0.0:12686 [None/None]
18:54:47.273 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://BLRB1DT166:8443/milo to 0.0.0.0:8443 [None/None]
18:54:47.273 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://BLRB1DT166:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
18:54:47.273 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://BLRB1DT166:8443/milo/discovery to 0.0.0.0:8443 [None/None]
18:54:47.273 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://10.1.61.38:8443/milo to 0.0.0.0:8443 [None/None]
18:54:47.273 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://10.1.61.38:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
18:54:47.273 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://10.1.61.38:8443/milo/discovery to 0.0.0.0:8443 [None/None]
18:54:47.273 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://BLRB1DT166.VTPPL.COM:8443/milo to 0.0.0.0:8443 [None/None]
18:54:47.274 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://BLRB1DT166.VTPPL.COM:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
18:54:47.274 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://BLRB1DT166.VTPPL.COM:8443/milo/discovery to 0.0.0.0:8443 [None/None]
18:54:47.274 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://127.0.0.1:8443/milo to 0.0.0.0:8443 [None/None]
18:54:47.274 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://127.0.0.1:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
18:54:47.274 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://127.0.0.1:8443/milo/discovery to 0.0.0.0:8443 [None/None]
[... cut ...]
	at org.eclipse.milo.examples.client.ReadExample.run(ReadExample.java:52)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:146)
	at org.eclipse.milo.examples.client.ReadExample.main(ReadExample.java:37)
Caused by: org.eclipse.milo.opcua.stack.core.UaException: status=Bad_SessionClosed, description=The session was closed by the client.
	at org.eclipse.milo.opcua.sdk.client.session.SessionFsmFactory.lambda$configureInactiveState$1(SessionFsmFactory.java:160)
```

Please correct me, if I am doing wrong coding. 
Please advise me.

kevinherron, 2020-03-30:
You need to connect before you call `client.getSession()`.

jdhanasekar, 2020-03-31:
After understanding purpose, why we need to call ```.getSession()```?. Is this call will open a secure session with server ?. Then only, Server will allow the client to access..

Am I correct ?

jdhanasekar, 2020-04-01:
I have connected the client before calling```client.getSession()```.
After running ```ReadExample.java```, I am getting ```Bad_ServiceUnsupported``` error. 
It means it is calling the server which I mentioned in ```ClientExample.java```,  Instead of LDS running on my network?.

```
16:54:23.755 [main] INFO  o.e.milo.examples.client.ReadExample - dhanasekar @ReadExample.java  stackClient =org.eclipse.milo.opcua.stack.client.UaStackClient@2c42b421
16:54:23.921 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error running client example: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999)
	at org.eclipse.milo.examples.client.ReadExample.run(ReadExample.java:61)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:146)
	at org.eclipse.milo.examples.client.ReadExample.main(ReadExample.java:37)
Caused by: org.eclipse.milo.opcua.stack.core.UaServiceFaultException: status=Bad_ServiceUnsupported, description=The server does not support the requested service.
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:273)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
16:54:23.922 [ForkJoinPool.commonPool-worker-3] ERROR o.e.m.e.client.ClientExampleRunner - Error running example: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
```

AntoninoBonanno, 2020-06-16:
Hello everyone.
I'm trying to register a milo server with an lds written in node.js.
I tried the code found in this issue but I have the same problem:

``` 
15:41:58.821 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error running client example: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
	at java.util.concurrent.CompletableFuture.reportGet(Unknown Source)
	at java.util.concurrent.CompletableFuture.get(Unknown Source)
	at Server.ServerMilo.run(ServerMilo.java:67)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:119)
	at Server.ServerMilo.main(ServerMilo.java:35)
Caused by: org.eclipse.milo.opcua.stack.core.UaServiceFaultException: status=Bad_ServiceUnsupported, description=The server does not support the requested service.
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:275)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
15:41:58.823 [ForkJoinPool.commonPool-worker-1] ERROR o.e.m.e.client.ClientExampleRunner - Error running example: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
	at java.util.concurrent.CompletableFuture.reportGet(Unknown Source)
	at java.util.concurrent.CompletableFuture.get(Unknown Source)
	at Server.ServerMilo.run(ServerMilo.java:67)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:119)
	at Server.ServerMilo.main(ServerMilo.java:35)
Caused by: org.eclipse.milo.opcua.stack.core.UaServiceFaultException: status=Bad_ServiceUnsupported, description=The server does not support the requested service.
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:275)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
```
My Code is:
```
public class ServerMilo implements ClientExample {
	
	public static void main(String[] args) throws Exception {
		ServerMilo example = new ServerMilo();
        new ClientExampleRunner(example, true).run();
    }	
	
	@Override
	public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
		String ServerURI ="urn:eclipse:milo:examples:server";
        String ProductURI = "https://opcfoundation.org/";
        LocalizedText[] myServerName = new LocalizedText[]{LocalizedText.english("eclipse milo opc-ua server")};
        ApplicationType ServerType =  ApplicationType.Server;
        String myGateWayURI = null;
        String[] myDiscoveryURLs =  new String[]{ "opc.tcp://localhost:4334"};
        String mySemaphorePath = "";
        Boolean IsOnline = Boolean.FALSE;  
               
        MdnsDiscoveryConfiguration mymDNS = new MdnsDiscoveryConfiguration("opc.tcp://localhost:4334",new String[] {"NA"});

        UaStackClient stackClient = client.getStackClient();
        SerializationContext mySerial =  stackClient.getSerializationContext();

        ExtensionObject[] myDisConfig =  {ExtensionObject.encode(mySerial,mymDNS)};
        
        client.connect().get();
        RequestHeader requestHeader = stackClient.newRequestHeader(
        		client.getSession().get().getAuthenticationToken()
                
        );        

        RegisteredServer myServer = new RegisteredServer(ServerURI,ProductURI,myServerName,ServerType,null,myDiscoveryURLs,null,IsOnline);
        
        RegisterServer2Request myServer2Request = new RegisterServer2Request(requestHeader,myServer,myDisConfig);
        client.connect().get();  
        
        RegisterServer2Response  rsr = (RegisterServer2Response)stackClient.sendRequest(myServer2Request).get();
        client.connect().get();
        
        future.complete(client);
	}

	/*@Override
	public String getEndpointUrl() {
        return "opc.tcp://localhost:12686/milo";
    }*/
}
```

How did you solve it @jdhanasekar ? Can you help me?

Is there an good example of a Milo server registering for an LDS?

jdhanasekar, 2020-06-16:
I think in RegisteredServer Call, Semaphore  should be empty string, not NULL.
Please changes this line 
```
        RegisteredServer myServer = new RegisteredServer(ServerURI,ProductURI,myServerName,ServerType,null,myDiscoveryURLs,null,IsOnline);
```
to 
```
        RegisteredServer myServer = new RegisteredServer(ServerURI,ProductURI,myServerName,ServerType,NULL,myDiscoveryURLs, mySemaphorePath,IsOnline);

```

It may work.

AntoninoBonanno, 2020-06-16:
I tried but nothing changes.

I also tried using this code:

```
RegisteredServer myServer =  RegisteredServer(ServerURI,ProductURI,myServerName,ServerType,null,myDiscoveryURLs,mySemaphorePath,IsOnline);        
RegisterServerRequest myServerRequest = new RegisterServerRequest(requestHeader,myServer);
client.connect().get();  
        
RegisterServerResponse  rsr = (RegisterServerResponse)stackClient.sendRequest(myServerRequest).get();
client.connect().get();
```

but the error is always the same

kevinherron, 2020-06-16:
@AntoninoBonanno you should probably be looking at your server, not Milo. The response with `Bad_ServiceUnsupported` can only come from the server.

AntoninoBonanno, 2020-06-16:
This is strange,
i am trying different stacks with different languages and i am able to register correctly at LDS.

This is my repository:
https://github.com/AntoninoBonanno/OPCUA-Local-Discovery-Server-Example

kevinherron, 2020-06-16:
Use Wireshark to capture the traffic between the different stacks and your LDS and maybe we can spot a difference then.

AntoninoBonanno, 2020-06-17:
I noticed that calling `client.connect().get()` it tries to send a "CreateSessionRequest" but LDS does not implement this type of requests.

I also saw this [pull request](https://github.com/eclipse/milo/pull/89) but it did not go through.

What I needed to do was implemented by Pro in this [example ](https://github.com/Pro/milo/blob/feature/lds/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/discovery/ExampleRegisterServer.java) but is not currently available in this version of Milo.

kevinherron, 2020-06-17:
> I noticed that calling client.connect().get() it tries to send a "CreateSessionRequest" but LDS does not implement this type of requests.

Okay, so that's probably what it's returning `Bad_ServiceUnsupported` in response to.

You'll have to do a little less copy and paste and a little more of your own work to only create and use a `UaStackClient` for calling the RegisterServer services. Don't create an `OpcUaClient` because it will always create a session first when you call `connect()`.

See https://github.com/eclipse/milo/blob/master/opc-ua-stack/stack-client/src/main/java/org/eclipse/milo/opcua/stack/client/DiscoveryClient.java for an example that does not use `OpcUaClient` or create a session.