eclipse-milo/milo #611: Not able to Connect

rodrigobelli2, 2020-03-09:
I'm trying to connect to a PROSYS OPC server simulator installed in the same machine.
I'm following a tutorial from
Using Maven w the following dependency
<dependency>
        <groupId>org.eclipse.milo</groupId>
        <artifactId>sdk-client</artifactId>
        <version>0.3.8</version>
    </dependency>

I tried to disable the security, sign&Encrypt and all the policies
See below the code and then the Exception I'm getting. 

**CODE:**
import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.api.config.OpcUaClientConfigBuilder;
import org.eclipse.milo.opcua.stack.client.DiscoveryClient;
import org.eclipse.milo.opcua.stack.core.UaException;
import org.eclipse.milo.opcua.stack.core.types.structured.EndpointDescription;
import java.util.List;
import java.util.concurrent.ExecutionException;

public class ClientTest {

    public void createClient(String endPointURL)  {
        List<EndpointDescription> endPoints = null;
        try {
            endPoints = DiscoveryClient.getEndpoints(endPointURL).get();
        } catch (InterruptedException e) {
            e.printStackTrace();
            e.getMessage();
            e.getCause();
        } catch (ExecutionException e) {
            e.printStackTrace();
        }
        OpcUaClientConfigBuilder cfg = new OpcUaClientConfigBuilder();
        cfg.setEndpoint(endPoints.get(0));
        OpcUaClient client = null;
        try {
            client = OpcUaClient.create(cfg.build());
        } catch (UaException e) {
            e.printStackTrace();
        }
        try {
            client.connect().get();
        } catch (InterruptedException e) {
            e.printStackTrace();
        } catch (ExecutionException e) {
            e.printStackTrace();
        }
    }
}

**EXCEPTION**
SLF4J: No SLF4J providers were found.
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#noProviders for further details.
java.util.concurrent.ExecutionException: UaException: status=Bad_ConnectionRejected, message=io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: ubung/127.0.1.1:53530
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999)
	at ClientTest.createClient(ClientTest.java:18)
	at Main.main(Main.java:14)
Caused by: UaException: status=Bad_ConnectionRejected, message=io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: ubung/127.0.1.1:53530
	at org.eclipse.milo.opcua.stack.client.transport.uasc.ClientChannelFsm$ClientChannelActions.lambda$connect$0(ClientChannelFsm.java:130)
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:502)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners0(DefaultPromise.java:495)
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:474)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:415)
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:540)
	at io.netty.util.concurrent.DefaultPromise.setFailure0(DefaultPromise.java:533)
	at io.netty.util.concurrent.DefaultPromise.tryFailure(DefaultPromise.java:114)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.fulfillConnectPromise(AbstractNioChannel.java:327)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:343)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:665)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:612)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:529)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:491)
	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:905)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: ubung/127.0.1.1:53530
	at java.base/sun.nio.ch.SocketChannelImpl.checkConnect(Native Method)
	at java.base/sun.nio.ch.SocketChannelImpl.finishConnect(SocketChannelImpl.java:779)
	at io.netty.channel.socket.nio.NioSocketChannel.doFinishConnect(NioSocketChannel.java:327)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:340)
	... 6 more
Caused by: java.net.ConnectException: Connection refused
	... 10 more
Exception in thread "main" java.lang.NullPointerException
	at ClientTest.createClient(ClientTest.java:29)
	at Main.main(Main.java:14)

Process finished with exit code 1

kevinherron, 2020-03-09:
This is just a plain Java networking error:
>  java.net.ConnectException: Connection refused

Are you sure the server is running, bound to localhost, and on the port you specified?

rodrigobelli2, 2020-03-09:
I was copying and pasting the URL from the OPC server, but for some reason it was connecting to the wrong address.
I overcame it by typing my IP directly, but I have a new exception when connecting the client.
See below the code and the exception

CODE
public void createClient(String endPointURL) {

        List<EndpointDescription> endPoints = null;
        try {
            endPoints = DiscoveryClient.getEndpoints(endPointURL).get();
            System.out.println("----------End Point --------------");
            System.out.println("End Point: " + endPoints.toString());
            System.out.println("URL: " + endPoints.get(0).getEndpointUrl());
        } catch (InterruptedException e) {
            e.printStackTrace();
        } catch (ExecutionException e) {
            e.printStackTrace();
        }

        //Creating the client
        OpcUaClientConfigBuilder cfg = new OpcUaClientConfigBuilder();
        cfg.setEndpoint(endPoints.get(0));
        System.out.println("-------------Cfg-----------");
        System.out.println("Client Config Builder: " + cfg.toString());
        OpcUaClient client = null;
        try {
            client = OpcUaClient.create(cfg.build());
            System.out.println("--------Client create-------------");
            System.out.println("Client Created: " + client.getAddressSpace().toString());
        } catch (UaException e) {
            e.printStackTrace();
            e.getMessage();
        }
        System.out.println("----------_Client-----------");
        System.out.println("Get Address Space: " + client.getAddressSpace());
        System.out.println("Get Config: " + client.getConfig());
        System.out.println("Get Session: " + client.getSession());
        System.out.println("Get Stack Client: " + client.getStackClient());
        try {
            System.out.println("---------------Trying Connection--------------------");
            client.connect().get();
        } catch (InterruptedException e) {
            e.printStackTrace();
        } catch (ExecutionException e) {
            System.out.println("Execution Exception");
            e.getMessage();
            e.getCause();
        }
}

**AND THE OUTPUT**
----------End Point --------------
End Point: [EndpointDescription{EndpointUrl=opc.tcp://ubung:53530/OPCUA/SimulationServer, Server=ApplicationDescription{ApplicationUri=urn:ubung:OPCUA:SimulationServer, ProductUri=urn:prosysopc.com:OPCUA:SimulationServer, ApplicationName=LocalizedText{text=SimulationServer@ubung, locale=}, ApplicationType=Server, GatewayServerUri=null, DiscoveryProfileUri=null, DiscoveryUrls=[opc.tcp://ubung:53530/OPCUA/SimulationServer]}, ServerCertificate=ByteString{bytes=[48, -126, 3, -55, 48, -126, 2, -79, -96, 3, 2, 1, 2, 2, 6, 1, 112, 14, 56, -122, -111, 48, 13, 6, 9, 42, -122, 72, -122, -9, 13, 1, 1, 11, 5, 0, 48, 77, 49, 31, 48, 29, 6, 3, 85, 4, 3, 12, 22, 83, 105, 109, 117, 108, 97, 116, 105, 111, 110, 83, 101, 114, 118, 101, 114, 64, 117, 98, 117, 110, 103, 49, 19, 48, 17, 6, 3, 85, 4, 10, 12, 10, 80, 114, 111, 115, 121, 115, 32, 79, 80, 67, 49, 21, 48, 19, 6, 10, 9, -110, 38, -119, -109, -14, 44, 100, 1, 25, 22, 5, 117, 98, 117, 110, 103, 48, 30, 23, 13, 50, 48, 48, 50, 48, 52, 48, 50, 50, 50, 51, 48, 90, 23, 13, 51, 48, 48, 50, 48, 49, 48, 51, 50, 50, 51, 48, 90, 48, 77, 49, 31, 48, 29, 6, 3, 85, 4, 3, 12, 22, 83, 105, 109, 117, 108, 97, 116, 105, 111, 110, 83, 101, 114, 118, 101, 114, 64, 117, 98, 117, 110, 103, 49, 19, 48, 17, 6, 3, 85, 4, 10, 12, 10, 80, 114, 111, 115, 121, 115, 32, 79, 80, 67, 49, 21, 48, 19, 6, 10, 9, -110, 38, -119, -109, -14, 44, 100, 1, 25, 22, 5, 117, 98, 117, 110, 103, 48, -126, 1, 34, 48, 13, 6, 9, 42, -122, 72, -122, -9, 13, 1, 1, 1, 5, 0, 3, -126, 1, 15, 0, 48, -126, 1, 10, 2, -126, 1, 1, 0, -42, 68, -90, -78, -100, 30, -103, 4, -76, 4, 69, -70, 2, 30, -118, 18, 96, 110, 93, 61, 79, -120, 61, 28, -39, -74, -110, 63, 118, 51, 74, 11, 109, -70, 116, -107, -21, 6, -29, -88, -29, -72, -11, 20, -11, -36, -32, 82, -10, 73, 113, -105, 112, -18, -100, 90, 79, -91, -25, 100, -9, -2, 125, 68, -122, 29, -79, -48, -114, 29, -107, 56, 123, 36, 41, 15, 102, -51, 37, -105, -127, -41, 55, -92, 25, 105, 118, 72, -125, 44, 107, 18, 109, -33, 70, 55, -67, -115, 16, 75, -39, -30, -69, 39, 22, 9, -110, -43, 69, 27, -2, 67, 111, 61, -98, -57, 115, 4, -123, -48, 43, 44, 32, -17, -46, -26, -103, -6, 72, 63, -72, 19, 35, 65, 20, -79, -7, -35, 86, 80, -38, -4, 21, -39, -109, 74, 51, 99, -96, 13, -106, 22, 66, -3, -35, -57, -111, 88, 70, -88, -104, -91, 7, 55, -16, 53, 46, 17, 97, -14, 84, -84, 30, -90, 59, -121, 33, -17, 105, -42, 34, 70, 87, -35, -11, 87, 77, 10, 122, 23, -101, 81, 115, 60, 69, 36, -26, 17, -117, -10, 91, -107, 25, -125, -66, -50, -108, -74, 15, 94, 33, -113, 34, 112, -93, 98, -56, -55, -75, -27, -127, -9, 124, 15, 64, 57, 7, -74, 94, 108, -40, -114, -37, -94, 2, 36, 104, -103, 3, 97, -55, 27, 42, 92, 124, 11, 57, -124, -24, -125, 24, -5, -64, 71, 67, 37, 2, 3, 1, 0, 1, -93, -127, -82, 48, -127, -85, 48, 31, 6, 3, 85, 29, 35, 4, 24, 48, 22, -128, 20, -110, 58, -120, -94, -9, 88, -105, 77, 92, 108, 103, -119, -4, 93, 9, 53, -11, 21, -94, -82, 48, 29, 6, 3, 85, 29, 14, 4, 22, 4, 20, -110, 58, -120, -94, -9, 88, -105, 77, 92, 108, 103, -119, -4, 93, 9, 53, -11, 21, -94, -82, 48, 9, 6, 3, 85, 29, 19, 4, 2, 48, 0, 48, 11, 6, 3, 85, 29, 15, 4, 4, 3, 2, 2, -12, 48, 29, 6, 3, 85, 29, 37, 4, 22, 48, 20, 6, 8, 43, 6, 1, 5, 5, 7, 3, 1, 6, 8, 43, 6, 1, 5, 5, 7, 3, 2, 48, 50, 6, 3, 85, 29, 17, 4, 43 [... cut ...]
Client Created: org.eclipse.milo.opcua.sdk.client.DefaultAddressSpace@71984c3
----------_Client-----------
Get Address Space: org.eclipse.milo.opcua.sdk.client.DefaultAddressSpace@71984c3
Get Config: org.eclipse.milo.opcua.sdk.client.api.config.OpcUaClientConfigBuilder$OpcUaClientConfigImpl@5536379e
Get Session: java.util.concurrent.CompletableFuture@2364305a[Not completed]
Get Stack Client: org.eclipse.milo.opcua.stack.client.UaStackClient@470a696f
---------------Trying Connection--------------------
Execution Exception

rodrigobelli2, 2020-03-09:
The e.printStackTrace() is:
See that again, it is trying to connect to 127.0.1.1. I dont know where java is getting this value and how to change

java.util.concurrent.ExecutionException: UaException: status=Bad_ConnectionRejected, message=io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: ubung/127.0.1.1:53530
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999)
	at ClientTest.createClient(ClientTest.java:47)
	at Main.main(Main.java:13)

kevinherron, 2020-03-10:
Looks like you didn't include any exceptions.

rodrigobelli2, 2020-03-10:
See below the same OUTPUT with the e.PrintStackTrace():

----------End Point --------------
End Point: [EndpointDescription{EndpointUrl=opc.tcp://ubung:53530/OPCUA/SimulationServer, Server=ApplicationDescription{ApplicationUri=urn:ubung:OPCUA:SimulationServer, ProductUri=urn:prosysopc.com:OPCUA:SimulationServer, ApplicationName=LocalizedText{text=SimulationServer@ubung, locale=}, ApplicationType=Server, GatewayServerUri=null, DiscoveryProfileUri=null, DiscoveryUrls=[opc.tcp://ubung:53530/OPCUA/SimulationServer]}, ServerCertificate=ByteString{bytes=[48, -126, 3, -55, 48, -126, 2, -79, -96, 3, 2, 1, 2, 2, 6, 1, 112, 14, 56, -122, -111, 48, 13, 6, 9, 42, -122, 72, -122, -9, 13, 1, 1, 11, 5, 0, 48, 77, 49, 31, 48, 29, 6, 3, 85, 4, 3, 12, 22, 83, 105, 109, 117, 108, 97, 116, 105, 111, 110, 83, 101, 114, 118, 101, 114, 64, 117, 98, 117, 110, 103, 49, 19, 48, 17, 6, 3, 85, 4, 10, 12, 10, 80, 114, 111, 115, 121, 115, 32, 79, 80, 67, 49, 21, 48, 19, 6, 10, 9, -110, 38, -119, -109, -14, 44, 100, 1, 25, 22, 5, 117, 98, 117, 110, 103, 48, 30, 23, 13, 50, 48, 48, 50, 48, 52, 48, 50, 50, 50, 51, 48, 90, 23, 13, 51, 48, 48, 50, 48, 49, 48, 51, 50, 50, 51, 48, 90, 48, 77, 49, 31, 48, 29, 6, 3, 85, 4, 3, 12, 22, 83, 105, 109, 117, 108, 97, 116, 105, 111, 110, 83, 101, 114, 118, 101, 114, 64, 117, 98, 117, 110, 103, 49, 19, 48, 17, 6, 3, 85, 4, 10, 12, 10, 80, 114, 111, 115, 121, 115, 32, 79, 80, 67, 49, 21, 48, 19, 6, 10, 9, -110, 38, -119, -109, -14, 44, 100, 1, 25, 22, 5, 117, 98, 117, 110, 103, 48, -126, 1, 34, 48, 13, 6, 9, 42, -122, 72, -122, -9, 13, 1, 1, 1, 5, 0, 3, -126, 1, 15, 0, 48, -126, 1, 10, 2, -126, 1, 1, 0, -42, 68, -90, -78, -100, 30, -103, 4, -76, 4, 69, -70, 2, 30, -118, 18, 96, 110, 93, 61, 79, -120, 61, 28, -39, -74, -110, 63, 118, 51, 74, 11, 109, -70, 116, -107, -21, 6, -29, -88, -29, -72, -11, 20, -11, -36, -32, 82, -10, 73, 113, -105, 112, -18, -100, 90, 79, -91, -25, 100, -9, -2, 125, 68, -122, 29, -79, -48, -114, 29, -107, 56, 123, 36, 41, 15, 102, -51, 37, -105, -127, -41, 55, -92, 25, 105, 118, 72, -125, 44, 107, 18, 109, -33, 70, 55, -67, -115, 16, 75, -39, -30, -69, 39, 22, 9, -110, -43, 69, 27, -2, 67, 111, 61, -98, -57, 115, 4, -123, -48, 43, 44, 32, -17, -46, -26, -103, -6, 72, 63, -72, 19, 35, 65, 20, -79, -7, -35, 86, 80, -38, -4, 21, -39, -109, 74, 51, 99, -96, 13, -106, 22, 66, -3, -35, -57, -111, 88, 70, -88, -104, -91, 7, 55, -16, 53, 46, 17, 97, -14, 84, -84, 30, -90, 59, -121, 33, -17, 105, -42, 34, 70, 87, -35, -11, 87, 77, 10, 122, 23, -101, 81, 115, 60, 69, 36, -26, 17, -117, -10, 91, -107, 25, -125, -66, -50, -108, -74, 15, 94, 33, -113, 34, 112, -93, 98, -56, -55, -75, -27, -127, -9, 124, 15, 64, 57, 7, -74, 94, 108, -40, -114, -37, -94, 2, 36, 104, -103, 3, 97, -55, 27, 42, 92, 124, 11, 57, -124, -24, -125, 24, -5, -64, 71, 67, 37, 2, 3, 1, 0, 1, -93, -127, -82, 48, -127, -85, 48, 31, 6, 3, 85, 29, 35, 4, 24, 48, 22, -128, 20, -110, 58, -120, -94, -9, 88, -105, 77, 92, 108, 103, -119, -4, 93, 9, 53, -11, 21, -94, -82, 48, 29, 6, 3, 85, 29, 14, 4, 22, 4, 20, -110, 58, -120, -94, -9, 88, -105, 77, 92, 108, 103, -119, -4, 93, 9, 53, -11, 21, -94, -82, 48, 9, 6, 3, 85, 29, 19, 4, 2, 48, 0, 48, 11, 6, 3, 85, 29, 15, 4, 4, 3, 2, 2, -12, 48, 29, 6, 3, 85, 29, 37, 4, 22, 48, 20, 6, 8, 43, 6, 1, 5, 5, 7, 3, 1, 6, 8, 43, 6, 1, 5, 5, 7, 3, 2, 48, 50, 6, 3, 85, 29, 17, 4, 43, 48, 41, -122, 32, 117, 114, 110, 58, 117, 98, 117, 110, 103, 58, 79, 80, 67, 85, 65, 58, 83, 105, 109, 117, 108, 97, 116, 105, 111, 110, 83, 101, 114, 118, 101, 114, -126, 5, 117, 98, 117, 110, 103, 48, 13, 6, 9, 42, -122, 72, -122, -9, 13, 1, 1, 11, 5, 0, 3, -126, 1, 1, 0, 8, -126, 4, 59, 117, 1, -37, -42, 12, 75, 67, -73, 60, 35, 28, -87, -38, 38, -95, -112, 103, 8, 9, 20, 72, -18, -32, -105, -80, -128, -101, 88, 78, -112, 48, 27, -10, -121, 70, -46, 21, 33, 46, 27, 99, -100, 35, 22, 35, -47, 112, -107, 124, 6, -120, 23, -40, 52, 111, 23, 17, 50, -33, -54, 38, 60, 1, -111, 127, 84, -36, -13, 12, 21, -107, -114, -28, 109, -67, -82, 40, -79, 8, 114, 119, -31, -21, 80, -120, -13, 36, -44, -91, 101, -87, -35, 98, -61, -127, -101, -83, 66, -76, 11, 26, -81, -39, -26, 44, -62, 88, -2, 71, -83, 59, 90, 82, 77, -58, 45, -97, 105, 1, -9, 104, 11, -69, 60, 127, -5, -51, -125, -56, 14, 120, 59, 54, 54, -35, -86, 19, -126, -9, 51, -51, 19, -7, -21, 32, 89, 60, 104, 82, 55, -49, -78, -108, -116, -125, 72, -24, 7, 58, -67, -71, -63, 123, 123, 12, -38, -118, 16, -109, 43, -4, -23, 78, -89, -27, 63, 39, 121, 11, -74, -26, 94, -55, 123, 75, 107, -106, 99, 27, -119, 7, 9, -22, -42, -108, 63, 103, -32, 12, -3, 120, -63, -127, -23, -3, -16, 111, 93, 51, 74, 36, 19, 47, -4, -26, -59, 30, -122, -77, 7, 21, 121, -32, -35, 16, -3, 86, -3, 50, -3, 51, -110, 86, -124, -40, -111, -41, 93, 115, 19, 112, -113, 61, 15, -20, -34, -28, 68, -95, -128, -87, -109]}, SecurityMode=None, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None, UserIdentityTokens=[UserTokenPolicy{PolicyId=anonymous, TokenType=Anonymous, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=null}, UserTokenPolicy{PolicyId=certificate_basic128, TokenType=Certificate, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15}, UserTokenPolicy{PolicyId=certificate_basic256, TokenType=Certificate, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#Basic256}, UserTokenPolicy{PolicyId=username_basic128, TokenType=UserName, IssuedT [... cut ...]
Caused by: io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: ubung/127.0.1.1:53530
	at java.base/sun.nio.ch.SocketChannelImpl.checkConnect(Native Method)
	at java.base/sun.nio.ch.SocketChannelImpl.finishConnect(SocketChannelImpl.java:779)
	at io.netty.channel.socket.nio.NioSocketChannel.doFinishConnect(NioSocketChannel.java:327)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:340)
	... 6 more
Caused by: java.net.ConnectException: Connection refused
	... 10 more

Process finished with exit code 0

kevinherron, 2020-03-10:
Well... again:

> java.net.ConnectException: Connection refused

rodrigobelli2, 2020-03-10:
When creating the method below,
public void createClient(String endPointURL) {...}

I type my IP address (192.168.X.X:53530), and it work well when creating the end point,
endPoints = DiscoveryClient.getEndpoints(endPointURL).get();

But, when connecting the client (using the endpoint above), it go back to the 127.0.1.1:5353

kevinherron, 2020-03-10:
Ok, I understand now. 

The server is misconfigured and returning the wrong IP address in its EndpointDescriptions. 

If you can’t fix the server configuration you can create new endpoints with the hostname in the endpoint URL updated. There is a Stack Overflow connection about this with some sample code. You would then pass the modified EndpointDescription to the client when you configure it.

rodrigobelli2, 2020-03-10:
Can't find this stackoverflow... DO you have the link ?

kevinherron, 2020-03-10:
https://stackoverflow.com/questions/40554198/java-opc-ua-client-eclipse-milo-endpoint-url-changes-to-localhost?r=SearchResults

rodrigobelli2, 2020-03-10:
PERFECT!!!!
Thank you very much...

closed 2020-03-10 by rodrigobelli2

istibekesi, 2020-11-11:
I'm running OPC PROSYS OPC server simulator on linux and it seems `127.0.1.1` is coming from the hosts file.
https://serverfault.com/questions/363095/why-does-my-hostname-appear-with-the-address-127-0-1-1-rather-than-127-0-0-1-in

I removed that line from the `/etc/hosts`, restarted the simulation server and it worked.

Is this a bug in the linux version of OPC UA Simulation Server?

kevinherron, 2020-11-11:
@istibekesi I don't know, the Prosys simulation server is closed source and not related to Milo. I don't know how it works or what logic it uses to determine bind addresses. 

I don't think there is anything fundamentally wrong with how Ubuntu (I assume) uses 127.0.0.1 for localhost and 127.0.1.1 for the hostname, it just might require some extra configuration on the server to make sure the right bind addresses are used and the right IPs and hostnames are returned in the endpoint URLs.

iwongf, 2024-10-16:
wow