eclipse-milo/milo #612: how to subscribe until the end of the program.

zerothree3782, 2020-03-10:
How do I keep on subscribing until the end of the program?

Can I delete "future.get(15, TimeUnit.SECONDS);" in 147 line (ClientExample.java) and  Thread.sleep(5000);, future.complete(client); in 102, 103 line (SubscriptionExample.java)?

kevinherron, 2020-03-10:
You can make any changes you need to the "Runner" and examples to get them to keep running. The point of the examples is to show how to interact with various OPC UA services, not to provide a reference architecture or any kind of starting point for your own code. They terminate quickly to keep them quick and self-contained.

closed 2020-03-10 by kevinherron

zerothree3782, 2020-03-11:
I'll ask you one more question.
When i continue to Subscribe with KEPserver, java's memory continue to rise.
Can you find the cause?

kevinherron, 2020-03-11:
I have not seen any rise in memory that does not also fall again once the GC has run. Are you waiting long enough? How are you measuring memory usage?

zerothree3782, 2020-03-11:
i ran subscriptionexample for 40minute.
The memory was 279MB then went up to 432MB.

kevinherron, 2020-03-11:
Are you looking at the memory usage in a program that understands the Java heap or just looking at the Windows task manager or something?

If you use something like VisualVM you'll see the GC run eventually:
[image]

The JVM will use whatever memory it is allocated though, and it never releases it back to the OS, so you may never see memory in task manager decrease.

zerothree3782, 2020-03-12:
Thank you so much.
I saw the GC run by using Jconsole.
I'm sorry, but I'll just ask one more question.
"BiConsumer<UaMonitoredItem, Integer> onItemCreated =
            (item, id) -> item.setValueConsumer(this::onSubscriptionValue);"
I can receive one item at a time in the above source
Can I receive all the items I have subscribed to at once?

kevinherron, 2020-03-12:
You can add a `NotificationListener` to the `UaSubscription` and implement `onDataChangeNotification`.

zerothree3782, 2020-03-12:
wow!! Thank you very much for always good answers.

zerothree3782, 2020-03-16:
Hi kevinherron.  I'm sorry, but can I ask you one?
Is there a way to receive data or listen request publishing event even if the data does not change during the subscription?

kevinherron, 2020-03-16:
When you create your monitored items use a filter that specifies status, value, and timestamp changes:

```java
ExtensionObject filter = ExtensionObject.encode(
    client.getSerializationContext(),
    new DataChangeFilter(
        DataChangeTrigger.StatusValueTimestamp,
        uint(DeadbandType.None.getValue()),
        0.0
    )
);

MonitoringParameters parameters = new MonitoringParameters(
    clientHandle,
    1000.0,
    filter,
    uint(10),
    true
);
```

zerothree3782, 2020-03-19:
Well solved.
Thank you very much for the help and for the project.