eclipse-milo/milo #613: Is the server-sdk 0.4.0-SNAPSHOT available now?

0DennisLee0, 2020-03-15:
Hello kevein, I want to start some new features in version 0.4.0-SNAPSHOT I found the sdk below is available now. However the version of the server is not available. Could you please tell me when the newest version of server will be published?
```xml
<dependency>
            <groupId>org.eclipse.milo</groupId>
            <artifactId>sdk-client</artifactId>
            <version>0.4.0-SNAPSHOT</version>
        </dependency>
```

kevinherron, 2020-03-15:
SNAPSHOTs are periodically published but since they are SNAPSHOTs they are not retained for long. If you're developing against a SNAPSHOT version the best thing to do is check out the `dev/0.4` branch and run `mvn clean install` so you have your own local copy.

0DennisLee0, 2020-03-15:
> SNAPSHOTs are periodically published but since they are SNAPSHOTs they are not retained for long. If you're developing against a SNAPSHOT version the best thing to do is check out the `dev/0.4` branch and run `mvn clean install` so you have your own local copy.

OK 3Q very much~

0DennisLee0, 2020-03-15:
> SNAPSHOTs are periodically published but since they are SNAPSHOTs they are not retained for long. If you're developing against a SNAPSHOT version the best thing to do is check out the `dev/0.4` branch and run `mvn clean install` so you have your own local copy.

OK Thx very much~

closed 2020-03-15 by 0DennisLee0