eclipse-milo/milo #614: Disconnect not completing in case of monitored nodes and disconnection

drFabio, 2020-03-15:
I am experiencing the following behaviour with sdk-client 0.3.8.

If I connect, create a subscription listener, restart a server , reconnecting creating a new subscription listener. It gets hanging on the disconnect.get() .
The jstack log is here [jstack.log](https://github.com/eclipse/milo/files/4329405/jstack.log)

My code is essentially the following 2 classes

```java
package some.opcua.client;

import com.google.common.collect.ImmutableList;
import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.api.config.OpcUaClientConfig;
import org.eclipse.milo.opcua.sdk.client.api.config.OpcUaClientConfigBuilder;
import org.eclipse.milo.opcua.sdk.client.api.identity.AnonymousProvider;
import org.eclipse.milo.opcua.sdk.client.api.identity.UsernameProvider;
import org.eclipse.milo.opcua.sdk.client.api.subscriptions.UaSubscription;
import org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaSubscriptionManager;
import org.eclipse.milo.opcua.stack.client.DiscoveryClient;
import org.eclipse.milo.opcua.stack.core.types.builtin.LocalizedText;
import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
import org.eclipse.milo.opcua.stack.core.types.builtin.StatusCode;
import org.eclipse.milo.opcua.stack.core.types.structured.EndpointDescription;

import java.util.List;
import java.util.concurrent.ExecutionException;
import java.util.logging.Logger;

public class PLCClient {

    private static final String APPLICATION_NAME = "OPC-UA-ADAPTER";
    private boolean isDisconnected = false;
    private String username;
    private String password;
    private String endpointUrl;
    private boolean isAnonymous;
    private NodeId sessionId;
    private OpcUaClient opcUaClient;
    private UaSubscription uaSubscription;
    private OpcUaSubscriptionManager manager;
    private MySubscriptionListener listener;
    private static final Logger logger = Logger.getLogger(PLCClient.class.getName());

    public PLCClient(String username, String password, String host, Integer port, boolean isAnonymous) {
        this.endpointUrl = String.format("opc.tcp://%s:%d", host, port);
        this.username = username;
        this.password = password;
        this.isAnonymous = isAnonymous;
    }
    public synchronized void connect() throws Exception {
        try {
            List<EndpointDescription> endpoints;
            endpoints = DiscoveryClient.getEndpoints(endpointUrl).get();
            EndpointDescription discoveryEndpoint = endpoints.stream()
                    .filter(predicate -> {
                        logger.info(String.format("Predicate %s", predicate.toString()));
                        return true;
                    })
                    .findFirst()
                    .orElseThrow(() -> new Exception("no desired endpoints returned"));
            logger.info(String.format("Using endpoint: %s", discoveryEndpoint.getEndpointUrl()));
            EndpointDescription connectDescription = new EndpointDescription(endpointUrl, discoveryEndpoint.getServer(), discoveryEndpoint.getServerCertificate(), discoveryEndpoint.getSecurityMode(), discoveryEndpoint.getSecurityPolicyUri(), discoveryEndpoint.getUserIdentityTokens(), discoveryEndpoint.getTransportProfileUri(), discoveryEndpoint.getSecurityLevel());
            OpcUaClientConfigBuilder builder = new OpcUaClientConfigBuilder();
            builder.setEndpoint(connectDescription);
            if (this.isAnonymous) {
                builder.setIdentityProvider(new AnonymousProvider());
            } else {
                builder.setIdentityProvider(new UsernameProvider(username, password));
            }
            builder.setApplicationName(new LocalizedText(APPLICATION_NAME));
[... 132 lines cut ...]
    @Override
    public void onPublishFailure(UaException exception) {
        logger.info("Publish failed");
        logger.info(exception.toString());
        StatusCode code = exception.getStatusCode();
        if (code.getValue() == StatusCodes.Bad_NoSubscription) {
            logger.info("Resubscribing ");
             onNoSubscription.accept(code);
        }
    }
    @Override
    public void onNotificationDataLost(UaSubscription subscription) {
        logger.info("Notificiation data lost");
        logger.info(String.format("Subscription %s ", subscription.toString()));
    }

    @Override
    public void onSubscriptionTransferFailed(UaSubscription subscription, StatusCode statusCode) {
        logger.info("Subscription transfer failed");
        logger.info(String.format("Subscription %s", subscription.toString()) );
        logger.info(String.format("Status code %s", statusCode.toString()) );
    }
}

```

kevinherron, 2020-03-15:
Can you turn the logging levels for `org.eclipse.milo.opcua.sdk.client.SessionFsm` and `org.eclipse.milo.opcua.stack.client.ChannelFsm` to TRACE in whatever logging system you're using and then trigger the deadlock and attach the logs? Usually this just means adding an entry in log4j.properties or logback.xml or something like that.

I've fallen ill after traveling for work last week so it might be a little bit before I can really dig into this.

kevinherron, 2020-03-15:
Actually I've identified the issue here. You can reproduce it as simply as this:

```java
client.readValue(0.0, TimestampsToReturn.Both, Identifiers.Server_ServerStatus_CurrentTime).thenRun(() -> {
    System.out.println("Reading 2nd value...");
    try {
        client.readValue(.0, TimestampsToReturn.Both, Identifiers.Server_ServerStatus_CurrentTime).get();
    } catch (InterruptedException | ExecutionException e) {
        e.printStackTrace();
    }
    System.out.println("...read 2nd value");
});
```

The problem is that down in `UaStackClient` responses are delivered asynchronously but on queue to ensure that you can rely on the order you receive them in. What's happening here is that the `onPublishFailed` callback is happening while still on this queue, and when you block indefinitely by calling another service while still on this queue it is impossible for the response to that service call to ever get delivered, because it gets put into the queue for delivery.

I may be able to fix this particular case in `OpcUaSubscriptionManager` by delivering these callbacks asynchronously (I need to see if this will be okay first), but generally you must be careful not to making blocking calls while in the callback for a response being delivered. If you must, then you need to use then async versions of the CompletableFuture callbacks. 

For example the above code that reproduces the deadlock is safe if you use `thenRunAsync`:

```java
client.readValue(0.0, TimestampsToReturn.Both, Identifiers.Server_ServerStatus_CurrentTime).thenRunAsync(() -> {
    System.out.println("Reading 2nd value...");
    try {
        client.readValue(.0, TimestampsToReturn.Both, Identifiers.Server_ServerStatus_CurrentTime).get();
    } catch (InterruptedException | ExecutionException e) {
        e.printStackTrace();
    }
    System.out.println("...read 2nd value");
});
```

drFabio, 2020-03-15:
Hope you get well soon! I have no rush on my part and was more interested in report the problem in case it was an unintended behavior.  

Do you still need other logs?

If this is an intended behavior or a willing consequence it would be nice if we could at least throw some warning. Will make my code non blocking and then proceed will let you know if that does not solve the issue but feel free to close it .

kevinherron, 2020-03-15:
I don't think I need the logs right now.

I'll start looking at whether I can make these callbacks safer. In the meantime, when you're executing inside a callback, whether from the subscription manager or just a CompletableFuture for a response, just make sure not to block waiting for another response on the same thread.

kevinherron, 2020-03-15:
@drFabio I nearly forgot, but you should know that your handling of subscription re-creation is misplaced.

You only need to re-create a subscription if `onSubscriptionTransferFailed` is called. The client SDK already handles attempting to transfer existing subscriptions when it reconnects and only calls this if the transfer fails. This also means you don't need to delete the subscription or items because at this point the server has indicated the subscription no longer exists.

drFabio, 2020-03-16:
Ok now I got it. Not blocking on the resubscribe and reconnectio non transfer failed did the trick.
Thank you very much for the help and for the project.

kevinherron, 2020-03-17:
Thanks. I’ll close this for now, let me know if it’s still broken after you make some changes.

closed 2020-03-17 by kevinherron