eclipse-milo/milo #615: How can I enable user/password connection

rodrigobelli2, 2020-03-21:
I'm able to connect to an OPC server when disable the login, but how can I connect with a login required connection ?

My code is:

List<EndpointDescription> endPoints = null;
        try {
            endPoints = DiscoveryClient.getEndpoints(endPointURL).get();
        } catch (InterruptedException e) {
            e.printStackTrace();
        } catch (ExecutionException e) {
            e.printStackTrace();
        }

        //Update Url
        EndpointDescription s = updateEndpointUrl(endPoints.get(0), "127.0.0.1");

        //Creating the client Config Builder
        OpcUaClientConfigBuilder cfg = new OpcUaClientConfigBuilder();
        cfg.setEndpoint(s);
        //Creating Client
        OpcUaClient client = null;
        try {
            client = OpcUaClient.create(cfg.build());
        } catch (UaException e) {
            e.printStackTrace();
        }

       client.connect().get();

kevinherron, 2020-03-21:
You would configure a `UsernameProvider` when building the `OpcUaClientConfig`.

See https://github.com/eclipse/milo/blob/master/milo-examples/standalone-examples/src/main/java/org/eclipse/milo/examples/client/ClientWithUserNameAndPasswordExample.java and https://github.com/eclipse/milo/blob/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ClientExampleRunner.java

rodrigobelli2, 2020-03-21:
Can't find in the examples where this is applied.
Could you please, be more specific on how to configure the UsernameProvider ?

Thanks for your support

kevinherron, 2020-03-21:
Call `setIdentityProvider` when building the config: 
https://github.com/eclipse/milo/blob/5c6048d54be28449358f0a395bd8b02924127c51/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ClientExampleRunner.java#L105-L113

The example uses a `UsernameProvider`:
https://github.com/eclipse/milo/blob/5c6048d54be28449358f0a395bd8b02924127c51/milo-examples/standalone-examples/src/main/java/org/eclipse/milo/examples/client/ClientWithUserNameAndPasswordExample.java#L74-L77

rodrigobelli2, 2020-03-21:
I changed the code as below, but I have a nullPointer Exception. The client is not being created

//Creating the client Config Builder
        OpcUaClientConfigBuilder cfg = new OpcUaClientConfigBuilder();
        OpcUaClientConfig config = cfg.setIdentityProvider(getIdentityProvider()).build();
        //Creating Client
        OpcUaClient client = null;
        try {
            client = OpcUaClient.create(config);
        } catch (UaException e) {
            e.printStackTrace();
        }

rodrigobelli2, 2020-03-22:
OK, I did!
But now, It is returning an exception No KeyPair
UaException: status=Bad_ConfigurationError, message=no KeyPair configured

I appreciate...

kevinherron, 2020-03-22:
It seems you're connecting to an endpoint that requires security but have no configured a KeyPair or Certificate for your client to use. It's not possible to connect to a secured endpoint without these.

closed 2020-03-22 by kevinherron