eclipse-milo/milo #619: FindServerOnNetwork() throwing Bad_ServiceUnsupported Error.

jdhanasekar, 2020-04-02:
Hi Milo,
           I am working to extract list of server details which are running on my network. I have running open62541 as LDS server. I am calling ```FindServersOnNetwork``` from other system.  

In ReadExample.java, I have added code like the below, but I am getting ```Bad_ServiceUnsupported``` error,

```
UaStackClient stackClient = client.getStackClient();
        String CapFilter[] = new String[]{""};
        UInteger RecordId = UInteger.valueOf(0);
        UInteger RecordsToReturn = UInteger.valueOf(0);

        client.connect().get();
        logger.info("dhanasekar @ReadExample.java  client.connect().get() ={}", client.connect().get());
        logger.info("dhanasekar @ReadExample.java  stackClient ={}", stackClient);

        RequestHeader myRequestHeader = stackClient.newRequestHeader(
                client.getSession().get().getAuthenticationToken()
        );

        FindServersOnNetworkRequest myFSONRequest = new FindServersOnNetworkRequest(myRequestHeader,RecordId, RecordsToReturn,CapFilter);

        FindServersOnNetworkResponse res = (FindServersOnNetworkResponse)stackClient.sendRequest(myFSONRequest).get();

        logger.info("dhanasekar @ReadExample.java FindServerOnNetwork result ={}", res);
```
I am getting below error,
```
16:54:14.131 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://BLRB1DT166.VTPPL.COM:8443/milo/discovery to 0.0.0.0:8443 [None/None]
16:54:14.131 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://127.0.0.1:8443/milo to 0.0.0.0:8443 [None/None]
16:54:14.132 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://127.0.0.1:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
16:54:14.132 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://127.0.0.1:8443/milo/discovery to 0.0.0.0:8443 [None/None]
16:54:14.437 [main] INFO  o.e.m.e.client.ClientExampleRunner - security temp dir: C:\Users\DHANAS~1.J\AppData\Local\Temp\security
16:54:14.456 [main] INFO  o.e.m.examples.client.KeyStoreLoader - Loading KeyStore at C:\Users\DHANAS~1.J\AppData\Local\Temp\security\example-client.pfx
16:54:15.846 [main] INFO  o.e.m.e.client.ClientExampleRunner - Using endpoint: opc.tcp://10.1.60.19:12686/milo [None/None]
16:54:16.129 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Stack version: dev
16:54:16.129 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: dev
16:54:23.557 [NonceUtilSecureRandom] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
16:54:23.755 [main] INFO  o.e.milo.examples.client.ReadExample - dhanasekar @ReadExample.java  client.connect().get() =org.eclipse.milo.opcua.sdk.client.OpcUaClient@2e647e59
16:54:23.755 [main] INFO  o.e.milo.examples.client.ReadExample - dhanasekar @ReadExample.java  stackClient =org.eclipse.milo.opcua.stack.client.UaStackClient@2c42b421
16:54:23.921 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error running client example: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_ServiceUnsupported, message=The server does not support the requested service.
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999)
	at org.eclipse.milo.examples.client.ReadExample.run(ReadExample.java:61)
```

kevinherron, 2020-04-02:
Why don't you stop hacking up the ReadExample and write your own code to do this? You're making basic mistakes (again) like connecting to the wrong server or endpoint URL:

> [main] INFO  o.e.m.e.client.ClientExampleRunner - Using endpoint: opc.tcp://10.1.60.19:12686/milo [None/None]

closed 2020-04-02 by kevinherron

jdhanasekar, 2020-04-02:
Ok. I will take your advise and  I will write it in new file.  
But my concern, if I call ```FindServerOnNetwork()```, it should return list of sever, instead it is calling Server URL mentioned in ClientExample.java.  I didnt understand this. 

If I make ```serverRequired``` parameter to ```TRUE``` in ```ClientExampleRunner()``` constructor, will it give list of server on the network ?

kevinherron, 2020-04-02:
> If I make serverRequired parameter to TRUE in ClientExampleRunner() constructor, will it give list of server on the network ?

No, you're supposed to send this request to the discovery server, not one of the servers registered to it.

jdhanasekar, 2020-04-02:
Ok.. To Know about the discovery server running any my network, I should call ```GetEndpoint()```. From here, I need to build discovery process. 
Am I correct ?

kevinherron, 2020-04-02:
You have to know ahead of time that there is a discovery server running somewhere, but then sure you would connect to it like any other server and then send the FindServerOnNetwork request.

jdhanasekar, 2020-04-02:
In Real World scenario, new user or stack holders don't know about technology using. . 
How to know the endpoint URL ahead of time which is running some where?.
Instead of finding out endpoint manually, I working to find that problematically. 
Like, If I call function or run a Java class, it should return endpoint URL of discovery running.

kevinherron, 2020-04-02:
The only alternative would be to build support for LDS-ME (see Part 12) directly into your client.
Part 12 says this:

A Client has several choices for finding Servers: 
- Out-of-band discovery (i.e. entry into a GUI) of a DiscoveryUrl for a Server;
- Calling FindServers on the LDS installed on the Client host;
- Calling FindServers on a remote LDS, where the HostName for the remote host is manually entered;
- Calling FindServersOnNetwork (see Part 4) on the LDS-ME installed on Client host;
- Supporting the LDS-ME functionality locally in the Client;
- Searching for Servers known to a GlobalDiscoveryServer.

All of these options except "Supporting the LDS-ME functionality locally in the Client" require you to have a discovery/endpoint URL ahead of time.

I can tell you that in the Real World almost every piece of software currently used requires the end user to enter an endpoint URL. Server's document this URL and client UIs expect it. This is Real World.

jdhanasekar, 2020-04-02:
Thanks for detailed info. 
As I am new bee, I am unaware OPC UA Real World Scenario.

jdhanasekar, 2020-04-04:
Hi Kevin,

I have implementing my JAVA class file to find LDS server.  In ```FindServers``` service, we need to provide URL of Discovery Server.  
Before that , do I need to build config for server/client using ```ClientExampleRunner``` ?.

I have implemented code like this, 
```
import org.eclipse.milo.opcua.stack.core.types.structured.RequestHeader;

public class MyDiscovery extends FindServersRequest {
    public static void main(String[] args) throws Exception {

        String DiscoveryEndPointURL = "opc.tcp://qunta:4840/";

        String[] DiscoveryLocalID = new String[]{"LDS"};
        String[] DiscoveryServerURI = new String[] {""};

        RequestHeader reqHead = new RequestHeader();

        FindServersRequest mySerReq = (new FindServersRequest(reqHead,DiscoveryEndPointURL,DiscoveryLocalID,DiscoveryServerURI));

    }
}
```
I am looking for the ways to get response from ```FindServersRequest``` call .

jdhanasekar, 2020-04-06:
I am waiting for your advise. Please provide me..

kevinherron, 2020-04-08:
You need to configure and connect a client if you want to send a request to a server...

jdhanasekar, 2020-04-08:
Ok. I can take ```ReadExample.java``` implementation as a reference for configuring a client. 
Please advise me.

kevinherron, 2020-04-09:
`ClientExampleRunner` has code that creates a new `OpcUaClient` instance.

jdhanasekar, 2020-04-09:
Ok.. I will look into that and create a new OpcUaClient instance.

jdhanasekar, 2020-04-13:
In ClientExample.java, what is functionality of  ```endpointFilter()``` and ```getIdentityProvide()``` ?. Why do you need them ..

jdhanasekar, 2020-04-14:
I have implemented created OpcUaClient like this,
 
```
package org.eclipse.milo.examples.client;

import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.api.config.OpcUaClientConfig;
import org.eclipse.milo.opcua.stack.client.DiscoveryClient;
import org.eclipse.milo.opcua.stack.core.security.SecurityPolicy;
import org.eclipse.milo.opcua.stack.core.types.builtin.LocalizedText;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType;
import org.eclipse.milo.opcua.stack.core.types.structured.*;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import java.nio.file.Files;
import java.nio.file.Path;
import java.nio.file.Paths;
import java.util.List;
import java.util.concurrent.CompletableFuture;

import static org.eclipse.milo.opcua.stack.core.security.SecurityPolicy.None;
import static org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.Unsigned.uint;

public class MyDiscovery extends FindServersRequest {

    private ClientExample clientExample;
    private final Logger logger = LoggerFactory.getLogger(getClass());
    private final CompletableFuture<OpcUaClient> future = new CompletableFuture<>();

    public static void main(String[] args) throws Exception {

        String ServerURI = "";
        String ProductURI = "";
        LocalizedText[] myServerName = new LocalizedText[]{LocalizedText.english("")};
        ApplicationType ServerType = ApplicationType.DiscoveryServer;
        String[] MyDiscoveryURLs = new String[]{""};
        String mySemaphorePath = "";
        Boolean IsOnline = Boolean.TRUE;
        String DiscoveryEndPointURL = "opc.tcp://qunta:4840/";

        String[] DiscoveryLocalID = new String[]{"LDS"};
        String[] DiscoveryServerURI = new String[] {""};

        MyDiscovery md = new MyDiscovery();
        md.ClientCreation();
        RequestHeader reqHead = new RequestHeader();

        FindServersRequest mySerReq = (new FindServersRequest(reqHead,DiscoveryEndPointURL,DiscoveryLocalID,DiscoveryServerURI));

    }

    public OpcUaClient ClientCreation() throws Exception {

        Path securityTempDir = Paths.get(System.getProperty("java.io.tmpdir"), "security");
        Files.createDirectories(securityTempDir);
        if (!Files.exists(securityTempDir)) {
            throw new Exception("unable to create security dir: " + securityTempDir);
        }

        LoggerFactory.getLogger(getClass())
                .info("security temp dir: {}", securityTempDir.toAbsolutePath());

        KeyStoreLoader loader = new KeyStoreLoader().load(securityTempDir);
        SecurityPolicy securityPolicy = None;
        List<EndpointDescription> endpoints;

        try{
            endpoints = DiscoveryClient.getEndpoints("opc.tcp://10.1.60.19:4840").get();
        }catch (Throwable ex) {
[... 45 lines cut ...]
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999)
	at org.eclipse.milo.examples.client.MyDiscovery.ClientCreation(MyDiscovery.java:78)
	at org.eclipse.milo.examples.client.MyDiscovery.main(MyDiscovery.java:43)
Caused by: UaException: status=Bad_Timeout, message=io.netty.channel.ConnectTimeoutException: connection timed out: /10.1.60.19:4840
	at org.eclipse.milo.opcua.stack.client.transport.uasc.ClientChannelFsm$ClientChannelActions.lambda$connect$0(ClientChannelFsm.java:127)
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:502)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners0(DefaultPromise.java:495)
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:474)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:415)
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:540)
	at io.netty.util.concurrent.DefaultPromise.setFailure0(DefaultPromise.java:533)
	at io.netty.util.concurrent.DefaultPromise.tryFailure(DefaultPromise.java:114)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe$1.run(AbstractNioChannel.java:269)
	at io.netty.util.concurrent.PromiseTask$RunnableAdapter.call(PromiseTask.java:38)
	at io.netty.util.concurrent.ScheduledFutureTask.run(ScheduledFutureTask.java:127)
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:163)
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:404)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:495)
	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:905)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: io.netty.channel.ConnectTimeoutException: connection timed out: /10.1.60.19:4840
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe$1.run(AbstractNioChannel.java:267)
	... 7 more

```

jdhanasekar, 2020-04-25:
Kevin,

I have successfully implemented and connected to client. When I try to extract data from  LDS server by calling ```FindServerRequest()```, I am not getting any error but I am getting this message,

```
java.util.concurrent.CompletableFuture@46b61c56[Not completed]
```

My code like this,

```
package org.eclipse.milo.examples.client;

import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.api.config.OpcUaClientConfig;
import org.eclipse.milo.opcua.sdk.client.api.identity.AnonymousProvider;
import org.eclipse.milo.opcua.stack.client.DiscoveryClient;
import org.eclipse.milo.opcua.stack.client.UaStackClient;
import org.eclipse.milo.opcua.stack.core.security.SecurityPolicy;
import org.eclipse.milo.opcua.stack.core.types.builtin.LocalizedText;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType;
import org.eclipse.milo.opcua.stack.core.types.structured.*;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import java.nio.file.Files;
import java.nio.file.Path;
import java.nio.file.Paths;
import java.util.List;
import java.util.concurrent.CompletableFuture;

import static org.eclipse.milo.opcua.stack.core.security.SecurityPolicy.None;
import static org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.Unsigned.uint;

public class MyDiscovery extends FindServersRequest {

    private ClientExample clientExample;
    private final Logger logger = LoggerFactory.getLogger(getClass());
    private final CompletableFuture<OpcUaClient> future = new CompletableFuture<>();

    public static void main(String[] args) throws Exception {

        MyDiscovery md = new MyDiscovery();
        md.readLDS();

    }

    public OpcUaClient ClientCreation() throws Exception {

        Path securityTempDir = Paths.get(System.getProperty("java.io.tmpdir"), "security");
        Files.createDirectories(securityTempDir);
        if (!Files.exists(securityTempDir)) {
            throw new Exception("unable to create security dir: " + securityTempDir);
        }

        LoggerFactory.getLogger(getClass())
                .info("security temp dir: {}", securityTempDir.toAbsolutePath());

        KeyStoreLoader loader = new KeyStoreLoader().load(securityTempDir);
        SecurityPolicy securityPolicy = None;
        List<EndpointDescription> endpoints;

        try{
            endpoints = DiscoveryClient.getEndpoints("opc.tcp://10.1.60.19:4840").get();
        }catch (Throwable ex) {
            // try the explicit discovery endpoint as well
            //String discoveryUrl = clientExample.getEndpointUrl();
            String discoveryUrl = "opc.tcp://10.1.60.19:4840";

            if (!discoveryUrl.endsWith("/")) {
[... 67 lines cut ...]

        logger.info("dhanasekar @MyDiscovery.java : {}",stackClient.sendRequest(mySerReq));

    }
}

```
I am getting below output in IDE,
```
15:48:08.970 [main] INFO  o.e.milo.examples.client.MyDiscovery - security temp dir: C:\Users\DHANAS~1.J\AppData\Local\Temp\security
15:48:09.016 [main] INFO  o.e.m.examples.client.KeyStoreLoader - Loading KeyStore at C:\Users\DHANAS~1.J\AppData\Local\Temp\security\example-client.pfx
15:48:10.531 [main] INFO  o.e.milo.examples.client.MyDiscovery - Using endpoint: opc.tcp://10.1.60.19:4840 [None/None]
15:48:10.539 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Stack version: dev
15:48:10.539 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: dev
15:48:10.778 [main] INFO  o.e.milo.examples.client.MyDiscovery - dhanasekar @MyDiscovery.java stackClient : org.eclipse.milo.opcua.stack.client.UaStackClient@16eccb2e
15:48:10.778 [main] INFO  o.e.milo.examples.client.MyDiscovery - dhanasekar @MyDiscovery.java Before calling FindServersRequest: 
15:48:10.803 [NonceUtilSecureRandom] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
15:48:10.848 [main] INFO  o.e.milo.examples.client.MyDiscovery - dhanasekar @MyDiscovery.java client.connect().get() = org.eclipse.milo.opcua.sdk.client.OpcUaClient@36b4091c
15:48:10.848 [main] INFO  o.e.milo.examples.client.MyDiscovery - dhanasekar @MyDiscovery.java reqHead: RequestHeader{AuthenticationToken=NodeId{ns=1, id=38f0dfcf-8bbe-8d38-22a2-4a4412f12fc2}, Timestamp=DateTime{utcTime=132322834908480000, javaDate=Sat Apr 25 15:48:10 IST 2020}, RequestHandle=4, ReturnDiagnostics=0, AuditEntryId=null, TimeoutHint=5000, AdditionalHeader=null}
15:48:10.850 [main] INFO  o.e.milo.examples.client.MyDiscovery - dhanasekar @MyDiscovery.java : java.util.concurrent.CompletableFuture@46b61c56[Not completed]

Process finished with exit code 0
```

Please provide your advise.

kevinherron, 2020-04-25:
> Please provide your advise.

Call `.get()` on the CompletableFuture returned by `sendRequested`.

jdhanasekar, 2020-04-26:
I have changed code like this,

```
stackClient.sendRequest(mySerReq).get();
```
I am getting below output,
```
10:37:34.413 [main] INFO  o.e.milo.examples.client.MyDiscovery - dhanasekar @MyDiscovery.java : FindServersResponse{ResponseHeader=
ResponseHeader{Timestamp=DateTime{utcTime=132323512544035600, javaDate=Sun Apr 26 10:37:34 IST 2020}, 
RequestHandle=4, 
ServiceResult=StatusCode{name=Good, value=0x00000000, quality=good},
 ServiceDiagnostics=null,
 StringTable=null,
 AdditionalHeader=ExtensionObject{encoded=ByteString{bytes=null}, 
encodingId=NodeId{ns=0, id=0}}}, 
Servers=null}

```
I am seeing ```Servers=null```. 
I do not know this correct output. Please look at this and confirm me.

jdhanasekar, 2020-04-28:
Kevin,

Please provide your input.

kevinherron, 2020-04-28:
The discovery server returned a response that indicates there no registered servers matching your query. Maybe the value you send for the `serverUris` parameter should be an empty array rather than array with 1 element that is empty string.

jdhanasekar, 2020-04-29:
Kevin,

If ```serverUris``` equal  to null string or empty, server will return know lds server. So I made like that

[image]

After changing my code like this, 

```
 String[] DiscoveryServerURI = new String[] {"urn:open62541.example.local_discovery_server"};
```

I am getting below output,
```
10:26:55.003 [main] INFO  o.e.milo.examples.client.MyDiscovery - dhanasekar @MyDiscovery.java : 
FindServersResponse{ResponseHeader=
ResponseHeader{Timestamp=DateTime{utcTime=132326098150393170, javaDate=Wed Apr 29 10:26:55 IST 2020},
               	RequestHandle=4, 
		ServiceResult=StatusCode{name=Good, value=0x00000000, quality=good}, 
		ServiceDiagnostics=null, 
		StringTable=null, 
		AdditionalHeader=ExtensionObject{encoded=ByteString{bytes=null}, 
		encodingId=NodeId{ns=0, id=0}}}, 
Servers=[ApplicationDescription{
		ApplicationUri=urn:open62541.example.local_discovery_server, 
		ProductUri=http://open62541.org, 
		ApplicationName=LocalizedText{text=open62541-based OPC UA Application, locale=en}, 
		ApplicationType=DiscoveryServer, 
		GatewayServerUri=null, 
		DiscoveryProfileUri=null, 
		DiscoveryUrls=[opc.tcp://quanta:4840/]}]}
```

It seems it is working. 
Please confirm from your side.

kevinherron, 2020-05-01:
Seems to be working... you're getting a response at least.

jdhanasekar, 2020-05-02:
Thanks for the confirmation. 

What changes I have to do such that a host should return a known LDS server running within.