eclipse-milo/milo #620: BrowseExample, NPE by bogus events after program termination

coolbeevip, 2020-04-02:
NPE by bogus events after program termination

```
/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/bin/java "-javaagent:/Users/zhanglei/Library/Application Support/JetBrains/Toolbox/apps/IDEA-U/ch-0/193.5233.102/IntelliJ IDEA.app/Contents/lib/idea_rt.jar=64303:/Users/zhanglei/Library/Application Support/JetBrains/Toolbox/apps/IDEA-U/ch-0/193.5233.102/IntelliJ IDEA.app/Contents/bin" -Dfile.encoding=UTF-8 -classpath /Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/charsets.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/deploy.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/cldrdata.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/dnsns.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/jaccess.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/jfxrt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/localedata.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/nashorn.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/sunec.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/sunjce_provider.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/sunpkcs11.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/ext/zipfs.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/javaws.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/jce.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/jfr.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/jfxswt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/jsse.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/management-agent.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/plugin.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/resources.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/jre/lib/rt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/lib/ant-javafx.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/lib/dt.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/lib/javafx-mx.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/lib/jconsole.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/lib/packager.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/lib/sa-jdi.jar:/Library/Java/JavaVirtualMachines/jdk1.8.0_201.jdk/Contents/Home/lib/tools.jar:/Users/zhanglei/github/milo/milo-examples/client-examples/target/classes:/Users/zhanglei/github/milo/opc-ua-sdk/sdk-client/target/classes:/Users/zhanglei/github/milo/opc-ua-stack/stack-client/target/classes:/Users/zhanglei/github/milo/opc-ua-stack/stack-core/target/classes:/Users/zhanglei/.m2/repository/org/bouncycastle/bcprov-jdk15on/1.61/bcprov-jdk15on-1.61.jar:/Users/zhanglei/.m2/repository/org/bouncycastle/bcpkix-jdk15on/1.61/bcpkix-jdk15on-1.61.jar:/Users/zhanglei/.m2/repository/com/google/guava/guava/26.0-jre/guava-26.0-jre.jar:/Users/zhanglei/.m2/repository/io/netty/netty-codec/4.1.45.Final/netty-codec-4.1.45.Final.jar:/Users/zhanglei/.m2/repository/io/netty/netty-handler/4.1.45.Final/netty-handler-4.1.45.Final.jar:/Users/zhanglei/.m2/repository/org/slf4j/slf4j-api/1.7.25/slf4j-api-1.7.25.jar:/Users/zhanglei/.m2/repository/com/digitalpetri/netty/netty-channel-fsm/0.3/netty-channel-fsm-0.3.jar:/Users/zhanglei/.m2/repository/com/digitalpetri/fsm/strict-machine/0.1/strict-machine-0.1.jar:/Users/zhanglei/.m2/repository/io/netty/netty-codec-http/4.1.45.Final/netty-codec-http-4.1.45.Final.jar:/Users/zhanglei/.m2/repository/io/netty/netty-common/4.1.45.Final/netty-common-4.1.45.Final.jar:/Users/zhanglei/.m2/repository/io/netty/netty-buffer/4.1.45.Final/netty-buffer-4.1.45.Final.jar:/Users/zhanglei/.m2/repository/io/netty/netty-transport/4.1.45.Final/netty-transport-4.1.45.Final.jar:/Users/zhanglei/.m2/repository/io/netty/netty-resolver/4.1.45.Final/netty-resolver-4.1.45.Final.jar:/Users/zhanglei/github/milo/opc-ua-sdk/sdk-core/target/classes:/Users/zhanglei/github/milo/opc-ua-sdk/dictionary-reader/target/classes:/Users/zhanglei/github/milo/opc-ua-stack/bsd-parser/target/classes:/Users/zhanglei/github/milo/opc-ua-stack/bsd-core/target/classes:/Users/zhanglei/.m2/repository/com/sun/activation/jakarta.activation/1.2.1/jakarta.activation-1.2.1.jar:/Users/zhanglei/.m2/repository/org/glassfish/jaxb/jaxb-runtime/2.3.2/jaxb-runtime-2.3.2.jar:/Users/zhanglei/.m2/repository/jakarta/xml/bind/jakarta.xml.bind-api/2.3.2/jakarta.xml.bind-api-2.3.2.jar:/Users/zhanglei/.m2/repository/org/glassfish/jaxb/txw2/2.3.2/txw2-2.3.2.jar:/Users/zhanglei/.m2/repository/com/sun/istack/istack-commons-runtime/3.0.8/istack-commons-runtime-3.0.8.jar:/Users/zhanglei/.m2/repository/org/jvnet/staxex/stax-ex/1.8.1/stax-ex-1.8.1.jar:/Users/zhanglei/.m2/repository/com/sun/xml/fastinfoset/FastInfoset/1.2.16/FastInfoset-1.2.16.jar:/Users/zhanglei/.m2/repository/jakarta/activation/jakarta.activation-api/1.2.1/jakarta.activation-api-1.2.1.jar:/Users/zhanglei/github/milo/milo-examples/server-examples/target/classes:/Users/zhanglei/github/milo/opc-ua-sdk/sdk-server/target/classes:/Users/zhanglei/github/milo/opc-ua-stack/stack-server/target/classes:/Users/zhanglei/github/milo/opc-ua-stack/bsd-generator/target/classes:/Users/zhanglei/github/milo/opc-ua-sdk/dictionary-manager/target/classes:/Users/zhanglei/.m2/repository/ch/qos/logback/logback-classic/1.1.7/logback-classic-1.1.7.jar:/Users/zhanglei/.m2/repository/ch/qos/logback/logback-core/1.1.7/logback-core-1.1.7.jar org.eclipse.milo.examples.client.BrowseExample
20:11:31,191 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback.groovy]
20:11:31,191 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback-test.xml]
20:11:31,191 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Found resource [logback.xml] at [file:/Users/zhanglei/github/milo/milo-examples/client-examples/target/classes/logback.xml]
20:11:31,196 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs multiple times on the classpath.
20:11:31,196 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [file:/Users/zhanglei/github/milo/milo-examples/client-examples/target/classes/logback.xml]
20:11:31,196 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [file:/Users/zhanglei/github/milo/milo-examples/server-examples/target/classes/logback.xml]
20:11:31,293 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - debug attribute not set
20:11:31,307 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.ConsoleAppender]
20:11:31,310 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [STDOUT]
20:11:31,404 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - This appender no longer admits a layout as a sub-component, set an encoder instead.
20:11:31,404 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - To ensure compatibility, wrapping your layout in LayoutWrappingEncoder.
20:11:31,404 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - See also http://logback.qos.ch/codes.html#layoutInsteadOfEncoder for details
20:11:31,405 |-INFO in ch.qos.logback.classic.joran.action.RootLoggerAction - Setting level of ROOT logger to INFO
20:11:31,406 |-INFO in ch.qos.logback.core.joran.action.AppenderRefAction - Attaching appender named [STDOUT] to Logger[ROOT]
20:11:31,406 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - End of configuration.
20:11:31,409 |-INFO in ch.qos.logback.classic.joran.JoranConfigurator@22a71081 - Registering current configuration as safe fallback point

20:11:31.840 [main] INFO  o.e.m.examples.server.ExampleServer - security temp dir: /var/folders/fd/gqdh88px2fj66tmtcy6ffr580000gn/T/security
20:11:31.847 [main] INFO  o.e.m.examples.server.KeyStoreLoader - Loading KeyStore at /var/folders/fd/gqdh88px2fj66tmtcy6ffr580000gn/T/security/example-server.pfx
[... cut ...]
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

Process finished with exit code 0
```

kevinherron, 2020-04-02:
Yes I've noticed this as well. It's because ExampleNamespace does not ever cancel its fixed rate event-posting task during `onShutdown`.

It's not a serious issue but I will fix it anyway.

coolbeevip, 2020-04-02:
I solved it temporarily 

```
            // Post a bogus Event every couple seconds
            getServer().getScheduledExecutorService().scheduleAtFixedRate(() -> {
                Optional.ofNullable(getServer()).map(OpcUaServer::getEventFactory).ifPresent(t -> {
                    try {
                        BaseEventTypeNode eventNode = t.createEvent(
                            newNodeId(UUID.randomUUID()),
                            Identifiers.BaseEventType
                        );
                        eventNode.setBrowseName(new QualifiedName(1, "foo"));
                        eventNode.setDisplayName(LocalizedText.english("foo"));
                        eventNode.setEventId(ByteString.of(new byte[]{0, 1, 2, 3}));
                        eventNode.setEventType(Identifiers.BaseEventType);
                        eventNode.setSourceNode(serverNode.getNodeId());
                        eventNode.setSourceName(serverNode.getDisplayName().getText());
                        eventNode.setTime(DateTime.now());
                        eventNode.setReceiveTime(DateTime.NULL_VALUE);
                        eventNode.setMessage(LocalizedText.english("event message!"));
                        eventNode.setSeverity(ushort(2));
                        getServer().getEventBus().post(eventNode);
                        eventNode.delete();
                    } catch (Throwable e) {
                        logger.error("Error creating EventNode: {}", e.getMessage(), e);
                    }
                });
            }, 0, 2, TimeUnit.SECONDS);
```

kevinherron, 2020-04-02:
I've committed a fix, thanks for opening an issue.

commit f5297862 (2020-04-02): Ensure ExampleNamespace is shut down cleanly

fixes #620
closed 2020-04-02 by kevinherron